sinara-hw/sinara #78: eRTM15 - Signal/CLK ditribution board for RF-Backplane

sbouhabib, 2016-11-28:
the eRTM15 board is the board needed for distributing the signals and clocks over the RF-Backplane
there is possibility to distribute 2 different CLKs o 11 slots and 3 RF signals to 9 slots
Issues to be discussed:
-name proposal?
-CLK requirements: do we need/want both clocks to be distributed e.g. for future use?
--one will be the 100 MHz LVPECL
-RF distribution - 
--one is the 100 MHz 
--the other could be the "2.4GHz" - the board can have the possibility to add the PLL module like the saymas and the it is enough to have one instead of populating all of the Sayma boards
--is there a need for the third one to be distributed?
-Front panel connections
--is there a need to have the possibility to distribute every signal independantly, i.e. 1 "SMA" for every distributed signal, 1 for CLK, one for 100 MHz ref. one for 2,4 GHz... if one would have an external generation box

jbqubit, 2016-11-29:
> -name proposal?
In the spirit of naming the boards after Russian places I propose Baikal. Lake Baikal is the deepest freshwater lake in the world at 1642 m. 

Here's my notes from the call this afternoon where we discussed the remaining points raised by Samer. It was proposed to add a socket for a clock mezzanine just like on Sayma RTM. Here's how it would be wired up. 

# SMP connectors on analog mezzanine
- output: DAC_CLK
- output: REF_LO (unused)
- input: MEZ_CLK_IN 

# SMA inputs on panel 
- LF_SMA_IN (eg 100 MHz)
- HF_SMA_IN (eg 2.0 GHz)

# LF OUT MUX
- ADCLK948 
- input: LF_SMA_IN
- output: LF Driver, MEZ_CLK_IN

# HF OUT MUX
- ADCLK948
- inputs: HF_SMA_IN, DAC_CLK
- output: LF Driver 

# HF Driver
- HMC7043 driver (14 outputs)
- logic is LVPECL

# LF Driver
- HMC7043 driver (14 outputs)
- logic is LVPECL

hartytp, 2016-11-29:
@sbouhabib Can we use ADCLK948/ADCLK950 instead of HMC7043, please? We will use the 100MHz to generate our LO on the Sayma, so the propagation delay stability for this signal is crucial. The phase shifters in the HMC7043 significantly degrade its propagation delay stability and should thus not be used unless actually needed. If we need more than 10 outputs, use 2 x ADCLK948.

hartytp, 2016-11-29:
@jbqubit Why do we need LF OUT MUX when there is only one LF input source?

hartytp, 2016-11-29:
@sbouhabib Is it worth routing the mezzanine REF_LO output to an unused backplane line, since this could be used in future for a LO?

jbqubit, 2016-11-30:
Regarding use of ADCLK948/ADCLK950 instead of HMC7043, @hartytp is right. The propagation delay temperature coefficient of HMC7043 is 750 fs/C (Figure 13 [0]) while for ADCLK948/ADCLK950 it is 50 fs/C (Table 2 [1,2]). It's poor design to use HMC7043 as a clock fanout buffer. 

> @jbqubit Why do we need LF OUT MUX when there is only one LF input source?

This is used in case there is no clock mezzanine board installed. It's acting as 1:2 clock fanout buffer. 

>@sbouhabib Is it worth routing the mezzanine REF_LO output to an unused backplane line, since this could be used in future for a LO?

Good idea.  

[0] http://www.analog.com/media/en/technical-documentation/data-sheets/HMC7043.pdf
[1] http://www.analog.com/media/en/technical-documentation/data-sheets/ADCLK950.pdf
[2] http://www.analog.com/media/en/technical-documentation/data-sheets/ADCLK948.pdf

jordens, 2016-11-30:
That tempco is irrelevant. We don't use the analog delay.

hartytp, 2016-11-30:
@jordens 

> That tempco is irrelevant. We don't use the analog delay.

True -- we don't use the analog delay -- but AFAICT (based on reading a lot of clock distribution IC data sheets over the past year) the propagation delay stability for ICs with delays/dividers is generally much worse than straight muxes, even when the delays are disabled. I didn't see a good specification of propagation delay stability for the HMC7043 with the delays disabled. However, as a rule of thumb, propagation delay stability is usually strongly correlated with net propagation delay. The propagation delay for the HMC7043 is 820ps, compared with 210ps ADCLK948. This suggests that the HMC7043 is likely to be worse.

Since we're not using the delays/dividers, there is no strong reason to prefer the HMC7043 anyway. Moreover, since we already use ADCLK948 as a mux, using it for fanout as well allows us to save a BOM line.

If you're worried about the number of output channels, then use ADCLK950 as the mux and connect one output to a second ADCLK950. That gives a total of 19 outputs...

hartytp, 2016-11-30:
@sbouhabib If we're going to route a DAC clock over the RF BP, would it make sense to also route a REF_LO signal? This could be used e.g. for distributing a LO to the AFE mezzanines.

jordens, 2016-11-30:
@hartytp Sure. It's also fewer things to program. That's important. My comment is aimed at superficial readers of datasheets. Also the tempco is common mode and on top of the tempco that we already get from having some ADCLK948 and one HMC7043 in the same clock chain (which are not common mode).

hartytp, 2016-11-30:
@jordens ACK to all your points. N.B. as soon as one has Sayma cards in different racks, none of this is common-mode any more.

In any case, I think we all agree that ADCLK948/ADCLK950 is the way to go for this...

WeiDaZhang, 2016-11-30:
There is a ADCLK954 which does 2:12 LVPECL with similar spec. to ADCLK94x/ADCLK95x.
http://www.analog.com/media/en/technical-documentation/data-sheets/ADCLK954.pdf

sbouhabib, 2016-12-06:
if the ADCLK954 is OK noise-wise then it would simplify things greatly (2 inputs + more outputs), anyway the phase shifters are on the saymas and will be in the chain, a possibility is the HMC987LP5E, it has only 8 outputs though and one input but great noise performance
I would suggest having 3 inputs on the front panel (REF (100), HF_CLK (2.4), LO) they can be mux-ed using the 'ADCLK954's especially the HF_CLK and LO with the ones on the CLK_Mezzanine.
the Ref is distributed as 2 signals: one as the actual REF, the other as a LF_CLK through ERNI connectors so anyway at least 2 chips will be needed. 20 x 100MHz needed
I think the main question is what is most crucial and if the ADCLK954 is enough then it would simplify the design a lot.

hartytp, 2016-12-06:
@sbouhabib 

> if the ADCLK954 is OK noise-wise then it would simplify things greatly

ADCLK954 is fine noise-wise, and should be used for all mux/fanout on eRTM15 to keep the design simple. In more detail (cf #75):

- it has the same noise as the ADCLK948, which we use on the Sayma, and so will not degrade Sayma performance.

- its noise (close-in and broad-band) is lower than the DAC's, so it's fine for the DAC clock.

- for the REF, we only care about close-in noise (as broad-band noise is taken out by the PLL loop-filter). The close in noise of ADCLK954 is better than the best Wenzel oscillator, so this won't be a problem.

- for a LO, we do care about the broad-band noise. However, this is still very good: @ -150dBc/Hz @1GHz it's better than the best Keysight synth, and so won't be a problem for us. If one wants a better LO than that, then it should be distributed directly to the Sayma AFE mezzanines using front-panel SMAs, as previously discussed.

>anyway the phase shifters are on the saymas and will be in the chain

On the Sayma, the phase shifters are only in the DAC clock path, and not on the REF/LO path. For applications where we care most about phase noise/drift, we will upconvert the DAC's output. In this case, the LO matters much more than the DAC since it's at a higher frequency. Since the 100MHz may be used as a reference the LO PLL, it must also have very good stability (better than the DAC ideally).

As a general rule, we should avoid phase shifters anywhere that they're not actually needed!

> a possibility is the HMC987LP5E, it has only 8 outputs though and one input but great noise performance

The main reason I wouldn't go for the HMC987 is that it doesn't have a mux. Obviously, the better phase noise doesn't help if it's used after an ADCLK954 mux. 

> I would suggest having 3 inputs on the front panel (REF (100), HF_CLK (2.4), LO) they can be mux-ed using the 'ADCLK954's especially the HF_CLK and LO with the ones on the CLK_Mezzanine.

That sounds like a good plan to me. 

I'd suggest connecting these inputs the same way we did for the Sayma: all three should use a TCM2-43X+ balun, with isolated input ground. For the 100MHz ref, we should also use a LTC6957-1 for sine-square conversion.

If the ADCLK954s have spare outputs, is it worth having a buffered output SMA for each signal as well? That way, one can daisy-chain the references if one wants. I don't feel strongly about this, but it could be a nice feature.

> the Ref is distributed as 2 signals: one as the actual REF, the other as a LF_CLK through ERNI connectors so anyway at least 2 chips will be needed. 20 x 100MHz needed

Do we want/need to support both of these at the same time, or can this just be done by solder jumpers? Presumably, in the long term, we only want/need to support one option, and we only support both in this version for testing? I don't mind either way, I'm just curious.

sbouhabib, 2016-12-06:
@hartytp only one thing that comes to me about phase noise would be additive jitter, as in a lot of dist chips are on the way till they reach the DACs, I haven't analyzed this but maybe it makes sense to check the whole chain.
As for the outputs they are a good feature I agree and I think there should not be any problem with adding them,
I think adding jumpers to all those clk/ref does not really make sense and would complicate things a lot,
we could not populate one the "ADCLK954s" with all its "surroundings"
one thing just occurred to me that to have this universal and flexible for future designs maybe an additional SMA for the CLKs that go to the ERNIs would be OK, than the analog ref could be then changed from the CLK frequency, maybe I galloped a little far on this but it is a possibility, not sure if there is any use case

hartytp, 2016-12-06:
@sbouhabib 

> only one thing that comes to me about phase noise would be additive jitter, as in a lot of dist chips are on the way till they reach the DACs, I haven't analyzed this but maybe it makes sense to check the whole chain.

Analysing the whole chain would be good. The only slight issue is that, given how many clocking options we've ended up supporting, one would have to simulate a lot of different paths (each ref option with each clock option?).

The most important simulations are done in #75. As you point out, there will be a degradation due to the fact that we use more than one ADCLK948. However, since we only use one type of mux/fan-out IC (ADCLK948 and ADCLK95x are the same noise-wise), this calculation is easy to do -- just multiply the ADCLK948 noise by the number of muxes, which one can do in one's head! 

For the 100MHz reference and DAC clock distribution, the mux/fanout ICs we have a reasonable amount of head-room (again, cf #75), so I don't suspect that mux phase-noise will be a significant issue -- although, the propagation-delay stability could be. 

For the LO, since the ADCLK948 broad-band noise is -150dBc/Hz, I still think that the noise will be acceptable after a few muxes. If not, one is better of using a specialist distribution amp external to the uTCA crate IMO.

hartytp, 2016-12-06:
@sbouhabib 

> I think adding jumpers to all those clk/ref does not really make sense and would complicate things a lot,
we could not populate one the "ADCLK954s" with all its "surroundings"

good point.

hartytp, 2016-12-06:
@sbouhabib 

> one thing just occurred to me that to have this universal and flexible for future designs maybe an additional SMA for the CLKs that go to the ERNIs would be OK, than the analog ref could be then [...]

I don't think I follow you here. Could you describe this in more detail, please? In general, I think we already support too many clock distribution options, and would like not to let things get too much worse. Aside from having a lot of muxes, which can add noise and drift, this is going to take quite a while to test and document for the users.

sbouhabib, 2016-12-06:
@hartytp as i said it was just a thought, in the initial design of the RF backplane, The CLKs on the ERNI, and the REF, LO and CAL on the Radiall are different signals. In our case the CLK and the REF are the same.
The question was: is there any possible future use case where the CLK and REF signals are separate (different frequencies), then it would be possible to add this feature, esp. if we use the ADCLK954 then it is one additional SMA + balun to the ADCLK945 of the ERNI_CLKs,

hartytp, 2016-12-06:
@sbouhabib Okay, I understand, thank you for clarifying.

It's a nice idea, but personally I wouldn't bother: we have no hardware planned that would make use of this. Giving the user more options means we have more things to document.  eRTM15 is such a small, cheap board that we can always add functionality later on. So, I'd rather not overload it with speculative features.

Having said that all, I don't feel at all strongly about this, and am happy to go for whatever you think best.

jbqubit, 2016-12-07:
>Analysing the whole chain would be good. The only slight 
>issue is that, given how many clocking options we've ended up 
>supporting, one would have to simulate a lot of different paths 
>(each ref option with each clock option?).
I advocate to build PCB_baikal and test out the whole frequency chain. The incremental cost of iterating on PCB_baikal is low. Save additional analysis for the circumstance that performance is lower than anticipated. 

>The most important simulations are done in #75. 
>this calculation is easy to do -- just multiply the ADCLK948 
>noise by the number of muxes, which one can do in one's head!
More generally, independent noise sources add in quadrature. 

>eRTM15 is such a small, cheap board that we can always add 
>functionality later on. So, I'd rather not overload it with speculative 
>features.
I agree with @hartytp. The incremental cost of new boards is small. Defer to future.

sbouhabib, 2016-12-08:
one thing just occurred to me, the ADCLK954 has no option to turn outputs on or off, which means ,that either all outputs are on or all are off, if the whole crate is not populated, the rest are left floating which is not so nice,
the options are:
-keep them floating - the simplest - everything is on - no control logic is needed, but may be reflections, mismatches, worse performance of ADCLK in general (maybe somebody can check this with an eval board of any adclk95x) higher cross-talk, unnecessary additional "noise" on the RF-backplane
-a different chip with the possibility to turn outputs on and off (possible worse phase noise, but not necessarily, possible higher complexity of CLK distribution - contol logic required (switches or jumpers or simple logic or uC ))
-Switches or RF-Gates (possible signal degradation, but not necessarily, possible higher complexity of CLK distribution - contol logic required(switches or jumpers or simple logic or uC ))

hartytp, 2016-12-08:
@sbouhabib As this is a prototype round, I'd be inclined to just add pads for 100R termination resistors on each output. For now, we can leave it up to the user to manually terminate all unused slots by soldering in appropriate resistors.

When we have the prototype hardware, we can test whether this is actually a problem (no point complicating the circuit to fix this if it's not a problem). If the reflections do cause problems, my preferred solution would be to add an ADCLK925 (or similar) as an output buffer for each channel after the fanout buffers. These output buffers should provide sufficient isolation between the outputs. 

I don't know of any ultra-low phase noise muxes with output enable lines, and using output buffers seems simpler than adding ultra-low phase noise, high-frequency switchable terminations IMO.

hartytp, 2016-12-08:
@sbouhabib It's a good point though, and we should make sure we add it to the list of things to test once the hardware arrives. @jbqubit @jordens has anyone started compiling a list of hardware tests to be done for this prototype round?

jordens, 2016-12-08:
Nothing centralized that I know of but it sounds like a good idea. Feel free to start a wiki page.
On the digital side of things the rough plan was that WUT tests/verifies individual items with known-good of-the-shelf gateware/software. Then we'd bootstrap and test everything incrementally. I don't know about the analog side.

hartytp, 2016-12-08:
@jordens Okay, I'll start a basic Wiki page when I have time

hartytp, 2016-12-08:
@jordens created a very rough page.

hartytp, 2016-12-08:
@sbouhabib Exactly what issues might you expect to arise from unterminated clock fanout-buffer outputs (so we know what to look for when testing the hardware)? 

- Phase noise?
- Spurs?
- Fluctuations in output power/phase?

jbqubit, 2016-12-08:
@sbouhabib @hartytp I created an issue for discussing testing plans. #117

closed 2016-12-12 by jbqubit