sinara-hw/sinara #79: Sayma "First" Mezzanine review

sbouhabib, 2016-11-28:
Sorry for reposting (first was an answer to the closed issue)

here are some issues for the discussion:

-is the name final, i.e. "Curly"? or some "name-says-it-all"?
-does the design have to fullfill the "curly" req. only or as @hartytp suggested to add an up-converting possibility?or anything else?
-Are there any monitoring needs (usage for Slow ADC channels, or GPIOs) i.e. temperature measurements etc...
-do I add an "identifier" i.e. eeprom, this would help identifing the type of mezz used and turning on LO signals or such
-The LDO discussion seems to just have ended, any more comments on the matter?

DO I add custom (DNP or 0R resistors/caps) i pi/T formation for matching inputs and outputs, i think this could help for differnet applications
-DAC channel:
--RF switch choice finalized?
--the aforementioned up-conversion possibility
-ADC channel:
--in general: Input requirements finalized?
--programmable/fixed attenuation?
--filter block?

jbqubit, 2016-11-28:
From teleconference 11/28... 

>  - is the name final, i.e. "Curly"? or some "name-says-it-all"?

- Let's go with "allaki" #30.
- [ ] change the name of this post to include "allaki"

>  - does the design have to fullfill the "curly" req. only or as @hartytp suggested to add an up-converting possibility?or anything else?

- There will be no up/down-conversion for this version of the analog mezzanine. 

>  - Are there any monitoring needs (usage for Slow ADC channels, or GPIOs) i.e. temperature measurements etc...

- Yes, add temperature sensor proximal to RF ouput amplifier. Use same type as on Sayma RTM.
- [ ] Power could be measured using power detector attached to 50 Ohm termination of the switch's RF1 output. Measure using existing ADC infrastructure on Sayma RTM motherboard. For those on the call today the anticipated application is driving AOM using off-board power amplifiers; off-board components are expected to dominate drift. Consensus was to skip the power diode. However, in reviewing #30 I see that @dhslichter and @dtcallcock say inclusion of the power detector is "necessary" for their application. So let's include it. Discuss choices in #80. 

>  - do I add an "identifier" i.e. eeprom, this would help identifing the type of mezz used and turning on LO signals or such

- [ ] Greg says #30 that 24AA02E48T-I/OT is already in the schematic. However I don't see it in the lastest schematics. 

>  - The LDO discussion seems to just have ended, any more comments on the matter?

- This was discussed in #50. Greg is going with LT3045. Use #50 if this requires further discussion. 

> DO I add custom (DNP or 0R resistors/caps) i pi/T formation for matching inputs and outputs?

- [ ] @sbouhabib says this was helpful in past designs to eg achieve better matching to RF connectors (SMA). Consensus is that this is not likely to be used but it's a low cost risk mitigation measure. @sbouhabib will add the pads. 

>    - RF switch choice finalized?

- [ ] RF switch measurements by @dtcallcock are discussed in #26. Consensus of those on call is that [HMC349ALP4CE](http://www.analog.com/media/en/technical-documentation/data-sheets/hmc349alp4c.pdf) the way to go (DC-4 GHz). 

- Balun (U_Diff2SE) in schematic is [TC1-1-13M+](http://www.minicircuits.com/pdfs/TC1-1-13M+.pdf) which covers 4.5 MHz to 3000 MHz. 

@sbouhabib Please check the checkboxes [ ] when you've updated schematics.

gkasprow, 2016-11-28:
Sorry guys, I didn't commit the latest version of schematics with EEPROM :)

Check it now.

jbqubit, 2016-11-28:
Move discussion of analog input for allaki to #81.

sbouhabib, 2016-12-06:
it was said to use the same temeprature monitor as in the sayma RTM which is mainly incorporates standard LM75s for temperature monitoring, unless i missed it on the schematics.
 in #11 @dhslichter proposes using ADT7420,
I just wanted to close the matter

jbqubit, 2016-12-06:
ADT7420 still looks like a good choice to me.

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

gkasprow, 2016-12-06:
Where exactly would you like to have these sensors?

jbqubit, 2017-01-26:
@sbouhabib This review still hasn't happened. Waiting for your schematics.

jbqubit, 2017-03-03:
I refered to discussion of specification for allaki analog output (#79) when reviewing this schematic.
- C55 is upside down schematic
DAC filters are discussed in Issue #108. 
- Make filter blocks clearer
    - [ ] put box around 11th order filter and define "MWO"; say something like "see eg DAC_Filter_A"
    - [ ] put box around FL2, FL3, FL4 and make clear that the layout accomodates use of MiniCircuits standard package FV1206.   
- Add "+13 dB" to schematic near ERA-4XSM+.
- Add "-13 dB" to schematic near YAT-12+. 
- Add "SPDT switch" to schematic near HMC349LP4C. 
- Add "power detector" to schematic near AD8363.

gkasprow, 2017-07-19:
@sbouhabib use HMC349ALP4CETR instead of HMC349LP4C

sbouhabib, 2017-08-01:
HMC349 changed

closed 2017-08-01 by sbouhabib