sinara-hw/sinara #80: allaki mezzanine: output power detection IC

jbqubit, 2016-11-28:
@dhslichter @dtcallcock What IC should be used for power detector? Discussed in #30 and #79.

jbqubit, 2016-11-30:
@dhslichter @dtcallcock ping

dtcallcock, 2016-11-30:
I suggest the AD8363 - [Datasheet](http://www.analog.com/media/en/technical-documentation/data-sheets/AD8363.pdf)

Schematic would be as Fig. 43 of datasheet with the following changes:

- A potential divider will be needed on output (see fig. 47) as max output is ~3.5V and the DAC voltage range is only 2.048V.  Suggest divide by 2 with R1=R2=1K.

- TCM1 and TCM2 should be held at 0.47V and 1V respectively (see Table 4) using potential dividers connected to VREF (2.3V).  Suggest R1=3.24K and R2=1K for TCM1, and R1=13K and R2=10K for TCM2.

- The onboard temperature sensor could be connected to a slow DAC ch too if we're not using it elsewhere on the mezzanine.  Output is 1.4V @ 25C and sensitivity is 5mV/C.

- The specced output of Allaki is +13dBm and the detector wants >0dBm so we need a 12dB pad before detector.  Suggest MCL GAT-12+ [Datasheet](http://www.minicircuits.com/pdfs/GAT-12+.pdf)

The part is $4.30 at 100+ and $355 at 1000+.  If people don't want it they can not populate it and the 12dB pad will likely be a sufficiently good termination (VSWR 1.14).

dhslichter, 2016-11-30:
Looks good to me.

hartytp, 2016-12-01:
@dtcallcock No objection to this. Out of curiosity though, what's the use case for this power detector? As @jbqubit pointed out, my feeling is that in most cases drift will be dominated by external power amps, etc, so there isn't much point feeding back on the power level in the AFE mezzanine.

jbqubit, 2016-12-01:
Looks good to me. @sbouhabib do you have what you need to add this to the schematic?

dtcallcock, 2016-12-01:
> Out of curiosity though, what's the use case for this power detector? 

- Self test.

- Some low power applications (like driving a microwave line on a surface trap) don't need external components but may need to be amplitude stabilised.

- If you have a stable source you have the option to passively stabilise external components or use simpler local feedback loops around unstable components (eg. controlling the gain of an external power amp).

- It makes it very easy to operate in a mode where you can set a fixed output power and then scan large frequency ranges, like you can with a synth.  

None of these make it a must-have but since we have the ADC, have an obvious place to put the detector and it can be left unpopulated if people wish, it seems worthwhile to me.

jbqubit, 2016-12-02:
@sbouhabib close this issue when you've added to schematic

hartytp, 2016-12-02:
@dtcallcock Sounds like a good plan.

sbouhabib, 2016-12-06:
@jbqubit, i dont have the part in the libraries, so @gkasprow if you can request it directly before i complete the list of other things,
@dtcallcock in general the 50 OHM res is there for matching if someone does not want to populate the power detection part.

dtcallcock, 2016-12-06:
@sbouhabib Is [this](http://www.analog.com/en/design-center/packaging-quality-symbols-footprints/symbols-and-footprints/AD8363.html) what you need?

jbqubit, 2016-12-12:
@sbouhabib This is not yet in the schematic.

closed 2016-12-18 by sbouhabib