sinara-hw/sinara #81: allaki analog mezzanine: input design review

jbqubit, 2016-11-28:
Discussion of input considerations in this Issue for allaki analog mezzanine. Output design review is #79.

jbqubit, 2016-11-28:
From hangout discussion on 11/28 we plan for two use scenarios allaki mezzanine analog input. Use a jumper to choose between two input paths. A single ADC driver is shared by both paths. 

## path 1: PID-type feedback to RF output
- constraint: The usable loop bandwidth is maybe 300 kHz. @jordens gives detailed analysis https://github.com/m-labs/sinara/wiki/Minutes20160921
- PID feedback subsystem for ARTIQ is not yet fully specified #24 
- signal sources
    - amplified photodiode  with 50 Ohm output
- input is DC-coupled, 50 Ohm
- input is differential (SMA pin, shield) 
- input SMA barrel dc-isolated from ground -- break ground loops 
- input ranges are +/- 100 mV, +/- 1V, +/- 10V
- components
    - SMA input
    - 50 Ohm matching 
    - [ ] differential amplifier  
    - [ ] variable gain amplifier
    - filter
    - ADC driver
- [ ] @hartytp update to reflect components you propose

## path 2: ADC sampling at full data rate 
- not intended for in-loop feedback
- see #82 for discussion of how this path might be used in the future 
- input is DC-coupled, 50 Ohm
- input is differential (SMA pin, shield) 
- input SMA barrel dc-isolated from ground -- break ground loops 
- input ranges are +/- 100 mV, +/- 1V, +/- 10V
- AD9656 data rate is 125 MSPS (16-bit) 
- analog signal is DC to 300 MHz (5th nyquist)
- components
    - 50 Ohm matching 
    - ADC driver

## ADC driver
- from path1 or path2
- differential ADC driver [AD4927](http://www.analog.com/media/en/technical-documentation/data-sheets/ADA4927-1_4927-2.pdf)
- matching to Sayma RTM motherboard (50 Ohm)  
- SMP connector pair to Sayma RTM motherboard

hartytp, 2016-11-29:
@jbqubit For the low-frequency input stage, the topology I'd go for would be (listing components in order):

1. **Input SMA**: ground capacitively coupled to circuit ground with ~10pF to avoid low-frequency ground loops
2. **Fixed input termination**: 100k to circuit ground for both SMA signal and ground (ensures there is always a DC path to ground for bias currents). Higher impedances (e.g. 1Mohm) are generally not required, and can lead to noise/offset issues for small input signals due to the InAmp current noise/bias current.
3. **Switchable termination** (not required? see below): 50R to SMA ground for signal, 0R to circuit ground for SMA ground. 
4. **RFI filter**: common-mode and differential-mode, see e.g. [ADI AN-671](http://www.analog.com/media/en/technical-documentation/application-notes/AN-671.pdf). Trade off between analog bandwidth and DC accuracy (RFI rejection). I'd start with 250kHz differential, 2.5MHz common-mode bandwidths. Keep filter resistors <~1k to minimise noise.
5. **ESD protection diodes**: from both inputs to each supply rail. Should be low leakage, low capacitance, such as BAV199. Must be able to take >10mA continuous forwards current (worse case is +-10V input, with current limited by ~1k RFI filter resistors). Depending on choice of RFI filter resistors and InAmp, these may not be needed.
6. **Switchable gain Instrumentation Amp**: see below
7. **Attenuation/matching**: signal out of InAmp is +-10V (TBC, see below). This must be matched to ADC input. (Use ADA4927-1 as described by @dhslichter?)

Comments/notes (written up quickly, need checking!):

**Noise + drift**:
- Aim for RMS noise + offset drifts (assume +-1K ambient temperature stability) to be <~3LSB to avoid degrading ADC performance too much
- For +-100mV (+-10V), 16-bit, 1LSB = 3uV (300uV)
- For white voltage noise with +-100mV (+-10V) input range and 250kHz single-pole response (390kHz noise BW), 3LSB corresponds to 15nV/rtHz (1.5uV/rtHz) RTI. 
- Assuming the input is driven by a low (<<1k) impedance source, the differential noise voltage due to InAmp current noise is only due to the RFI filter resistors. Assuming 1k resistors, the acceptable current noise is 15pA/rtHz (1.5na/rtHz) for +-0.1V (+-10V inputs).

**Switchable termination**:
- Part number TBD
- Must be DC coupled, compatible with +-10V inputs & +-12V supplies, low leakage and low capacitance
- I consider this optional (personally I wouldn't include it) because...
- As this input path is only for low-frequency signals, 50Ohm termination is generally not required (or a good idea if your source is a +-10V photodiode).
- 50Ohm termination could be helpful if the source is the IF of a mixer, which needs termination at RF. However, to do this properly, we would need an RF switch, which will be hard to find given our other constrains. Thus, in practice, it would probably be better to use an external bias-Tee + terminator.
- 50R resistor needs decent power handling (what should max no-damage current be?)
- 50R termination may be beneficial for self-calibration, as it removes offsets associated with InAmp bias current flowing through 100k termination resistors in the absence of a low-impedance source

**InAmp choice**:
- Main selection criteria: BW at higher gain; digitally selectable gain (don't want to have to use a gain-set resistor and low-capacitance analog switch); +-12V supplies
- Potential choices: AD825x (digitally programmable), AD8429 (gain set by resistor)
- My choice would be AD8253, and use gain to map all input ranges to +-10V after InAmp: G=1 (+-10V input range) f3dB=10MHz, G=10 (+-1V input range) f3dB=4MHz, G=100 (+-100mV input range) f3dB= 550kHz
- If we want more bandwidth than that, we could consider mapping input ranges to +-1V after the InAmp: G=1 for +-1V input; G=10 for +-100mV input; and, G=0.1 for +-10V (implemented by switching in an attenuator before the InAmp). Given that loop BW is limited to ~300kHz by digital latencey anyway, I don't think this small increase in bandwidth is worth the extra complexity.
- AD8253 RTI noise @ G>~10: 12nV/rtHz @ 1kHz, 1uVpp (0.1 to 10Hz), 5pA/rtHz @ 1kHz, 60pApp (0.1 to 10Hz). All fine
- AD8253 offsets: (+-1.2+5/G)uV/K (which is ~0.5LSB/K for the +-100mV range); offset current 160pA/K=> 160nV/K assuming 1kOhm filter resistors & low impedance source. Dominant source of drift is voltage offset. For 3LSB stability, temperature must be good to ~6K. 
- Package thermal impedance: 112K/W => 50mW change in load power gives 3LSB change in offset. Bear this in mind when choosing load current.

Calibration data to be stored in EEPROM?

**Decisions to make**:
- Are we happy with choice of topology and InAmp?
- Do we include switchable input terminations?
- Are we happy with using G=100 for +-100mV signals?
- Choice of CM & DM filter poles?

hartytp, 2016-11-29:
Proposed simple InAmp front-end:

[image]

Output from InAmp is +-10V for all input ranges. I'll leave matching this to the ADC input to @dhslichter or someone else who has thought more about the ADC.

sbouhabib, 2016-12-06:
@hartytp is the RFI filter components set/checked/final or do I have to redesign/simulate?

jordens, 2016-12-06:
+1 on the AD825x and that schematic as is

hartytp, 2016-12-06:
@sbouhabib 

> are the RFI filter components set/checked/final or do I have to redesign/simulate?

They're not set in stone, but I would be happy keeping them as they are. Please do feel free to check them, and make changes as you see fit.

The choice of component values was done "on the back of an envelope" by:
- Set common-mode and differential mode pole frequencies to something roughly sensible
  - Want differential BW >~200kHz to avoid limiting the system BW too much
  - common-mode filter capacitance should be roughly 10x differential mode capacitance to avoid compromising CMRR at higher frequencies
  - common-mode filter BW lower (ideally 1/10, but this isn't possible for high gains) than InAmp BW to minimise rectification of RFI/keep CMRR high at high frequencies
- Pick sensible values for resistors and capacitors:
  - Resistor not so large that we have noise/offset issues (see above)
  - Capacitors not so small that we are dominated by parasitics
  - Input resistor large enough to avoid loading source too heavily @ AC, and to limit diode current

Obviously, this leaves a lot of flexibility, and there are other choices I would be equally happy with. In the end, so long as we put something sensible here, users can always adjust later.

closed 2016-12-12 by jbqubit