sinara-hw/sinara #83: Sayma RTM component values missing

jbqubit, 2016-11-28:
11/23/2016 schematic for Sayma RTM motherboard p 2 ADC-JESD204B.
- Schottky diode types are missing from schematic (eg D16). #38 suggests HSMS-286C-TR1G
- Capacitor values are missing for inputs to AD9656 (eg C649)

gkasprow, 2016-11-28:
Diode types are correct, the PN was not enabled.
Did we decide what RC values are to be installed?

jbqubit, 2016-11-29:
@dhslichter @WeiDaZhang What RC values?

WeiDaZhang, 2016-11-30:
The configuration R210 = R211 = 5 Ohm, C641 = 56pF, R29 = R29 = 22 Ohm, C637 = 5.6pF, AD9656 input differential Cap = 7pF gives a low pass bandwidth of 280MHz (according to #81 300MHz 5th Nyquist). However, the Schottky diode should be placed at least one resistor apart from the input pins (between R28 and R210 // between R29 and R211) for two reasons:
1. As the schematic currently is, when input voltage exceeds the voltage range of the ADC pins, both the external diode and the AD9656 internal diode turns on, since there isn't voltage dropping in between. The external diodes won't be effectively protecting the input sillicon in AD9656. If move the diodes to the place i.e. between R210 and R28, when input exceeds 1.8VAVDD, the external diodes turn on, while the voltage drops across R28, and makes sure that the voltage on the input pin is in range.
2. The equivelant capacitor of the diodes (0.18pFx2) is not small enough comparing to the input capacitor of the AD9656, may cause unnecessary bandwidth uncertainty.

jbqubit, 2016-12-02:
@gkasprow Please indicate how you plan to respond to Weida's comments.

gkasprow, 2016-12-02:
These diodes are schottky ones with lower forward voltage than silicon ones in ADC.
So they will bypass the ADC ones and that's why they were inserted after the 49R5 resistors.
If we place them before the resistors, they can be destroyed if too high voltage appears at the input

jbqubit, 2016-12-02:
Also, capacitor values are still missing from schematic.

gkasprow, 2016-12-03:
There are 5Ohm resistors before the diodes, we can do it as WeiDaZhang  suggested.
Moreover the diodes are now in parallel with 56pF capacitors so their capacitance influence on filter bandwidth is negligible.

jbqubit, 2016-12-05:
@WeiDaZhang Are you happy with this solution?

jbqubit, 2016-12-12:
@WeiDaZhang Please weigh in as part of the design review.

jbqubit, 2017-01-26:
@WeiDaZhang Please close this issue if you're happy with the schematic.

hartytp, 2017-02-08:
Timeout. 

Closing issue under the assumption that no one objects to this choice.

closed 2017-02-08 by hartytp