sinara-hw/sinara #84: allaki analog mezzanine: input protection

jbqubit, 2016-11-28:
In #38 it was decided that ESD protection should go on the mezzanine boards. For example ESD Protection Diodes with Transient Voltage Suppressor (TVS). Such ESD protection is missing from #81. Should it be there? If so which component? @jordens @hartytp @dhslichter @sbouhabib

gkasprow, 2016-11-28:
we discussed that TVS cause signal distortion and that's why Schottky diodes were used.
I don't thing somebody is going to replace mezzanines while the AMC board is working so no protection diodes on IO lines were used.

jordens, 2016-11-29:
I am uncertain why distortion is relevant for the DC coupled path that we are shooting for. But if the input impedance is large and upstream ESD diodes should be sufficient.

jbqubit, 2016-11-29:
OK.

closed 2016-11-29 by jbqubit