sinara-hw/sinara #85: Sayma power analysis

jbqubit, 2016-11-28:
What is power budget for Sayma AMC and RTM? Required for design review. Informs choice of power supply for uTCA crate #21.  Who at WUT is working on this?

gkasprow, 2016-11-28:
Michal already did power analysis. He figured out that  4A from 2V supply for DACs and ADCs is not sufficient. That's why ADCs are now supplied from 4V rail. Since we switched to LT3045 regulators, we cannot use 2V anymore for 1.8V due to minimal dropout requirement .
Power consumption of Metlino and Sayma AMC is similar and does not exceed 25W   @michallgaska please confirm.
Sayma RTM power consumption heavily depends on inserted modules and with maximum load on AFE and clock modules we easily exceed 80W per RTM module. Michal is just updating the calculations after we changed ADC supply distribution.

jbqubit, 2016-11-29:
@michallgaska A more detailed power analysis is expected. At a low level I expect you have tabulated the expected current draw of each IC, the maximum available current of each regulator. Is this in git?

Power consumption of Sayma AMC and Sayma RTM with mezzanine/FMC slots unpopulated should be straightforward to calculate. 
- [ ] list voltages and currents supplied by crate

Regarding power consumption for mezzanine/FMC sockets: 
- [ ] for each voltage supplied by AMC/RTM  what is maximum current available to socket (limited by regulators)?
- [ ] what is efficiency factor for generating that maximum I*V by power electronics on AMC/RTM?
- [ ] what is expected current draw of current allaki mezzanine design?
- [ ] what is expected current draw of current clock mezzanine design?

Same for Metlino AMC.

gkasprow, 2016-11-29:
Michal has detailed Exel file with all ICs.

He will publish it ASAP.  <https://github.com/notifications/beacon/AEH-vmyfRMXYUld8duXsTuBdG1h921fXks5rDEfWgaJpZM4K-R44.gif>

jbqubit, 2016-12-02:
@michallgaska ping

jbqubit, 2016-12-06:
@michallgaska Please add the spreadsheet to the git repository.

michallgaska, 2016-12-08:
Done! 
sinara/ARTIQ_EE/power consumption.xlsx

jbqubit, 2016-12-08:
Thank you @michallgaska! Some questions

- [ ] All these potentials listed in the spreadsheet are derived from the AMC +12V supply, right? If so annotate spreadsheet to indicate this. 
- [ ] Sayma_RTM spreadsheet tab is missing clock mezzanine
- [ ] Annotate Sayma_RTM schematic to indicate budgeted current per analog mezzanine 
    - +3.3V, 1000 mA
    - -6.0V, 100 mA
    - +6.0V, 1500 mA
    - +12V, 200 mA
    - -12V, 50 mA
- [ ] Annotate Sayma_RTM schematic to indicate budgeted current for mezzanine 
- Missing from spreadsheet for all modules
    - [ ] power electronics connected to the AMC 3.3V supply
    - [ ] power budget for each module

jbqubit, 2016-12-12:
@michallgaska ping

jbqubit, 2017-01-26:
@gkasprow  Annotation on Sayma_RTM for mezzanine IO/power interfaces is thorough. Thank you. There's some garbled text on "ADC_DAC_AFE_Mezzanine". 

@michallgaska  Missing tab in power spreadsheet for power consumption of current allaki mezzanine.

closed 2017-02-21 by jbqubit