sinara-hw/sinara #88: Sayma RTM: clocking 2nd review

hartytp, 2016-11-29:
@gkasprow Thanks for the updated schematic Greg, that looks good. A few minor comments:

- [ ]  **Rename REF_100M signal on the AFE mezzanines**: this name is confusing because the signal can also be used for a LO at up to 5GHz (limited by clock mux). Can we make it consistent with the REF_LO name used on the clock mezzanine?

- [ ] **Rename SI_CLK_OUTn**: this name isn't very descriptive. Something like REC_CLK_CLEAN or CDR_CLK_CLEAN would be better (make consistent with SI5324 input signal)

- [ ] **Terminate CLKOUT2 of SI5324** (100Ohm differential). Unless there is a reason you'd prefer not to?

- [ ] **Remove REFSYNCIN RF backplane connection**: AFAICT this signal isn't required, cf #42 (although, @sbourdeauducq @jordens correct me if I'm wrong about this)

- [ ] **AC couple 50Ohm terminators**: move R98 to between C666 & TR1, so that there is no DC path to ground on EXT_100M_RF_IN (otherwise ground loops aren't broken). Might as well do the same thing for R237 on uRFB_REF_IN to keep these two paths the same. 

- [ ] **Remove LTC6957 on uRFB_REF_IN**: this signal will be a PECL square-wave with fast rising edges, so it can be fed directly into the ADCLK948 and does not require the LTC6957 (NB we do still want an LTC697 on EXT_100M_RF_IN, as this may be a sinusoid from an external oscillator).

- [ ] **Remove unnecessary PECL bias resistors on DC coupled outputs**: ADCLK948 inputs include termination to (Vcc+1)/2 (2.15V for Vcc=3V3), via centre-tapped 100Ohm differential termination. Thus, the outputs of the driving stage do not need bias resistors. This means we can remove: R59, R60, R91, R92, R242, R243, etc.

- [ ] **HMC830 ref input**: HMC830 input impedance is 100Ohm, so we should add an external 100Ohm termination resistor to match to 50Ohm. This signal contains harmonics up to several GHz, so proper termination is important unless the trace is very short. NB the single-ended output voltage from an ADCLK948 is a bit marginal for driving this input (cf #75) so consider adding a balun for differential-single ended conversion (this is mainly needed if the impedance match is not good, leading to reflections that can slow down the signal rise time).

- [ ] **HMC830 loop filter components** -- use the values posted in the comments to #75

gkasprow, 2016-11-30:
@sbourdeauducq asked me to connect REFSYNCIN RF to the RF backplane for future use
Done. ADCLK948 need external 50Ohm resistor to properly terminate it. Vref cannot be used for DC termination
HMC830 is so close to the mux that 100Ohm should do the job

hartytp, 2016-12-01:
@gkasprow Great, thanks!

> ADCLK948 need external 50Ohm resistor to properly terminate it. Vref cannot be used for DC termination

Yes, ADCLK948 need a 50Ohm resistor from Vt to GND for DC termination, as well as a 10nF cap to Vcc. However, the 200Ohm bias resistors are not needed.

In general, ADCLK948 inputs should be terminated using the schemes shown in figures 20 (DC) and 21 (AC) from the data sheet.

closed 2016-12-03 by gkasprow