sinara-hw/sinara #90: size of user-replacable SMD components

jbqubit, 2016-11-29:
@gkasprow do you have layout rules in mind for user-replaceable components? It would be convenient to specify SMD part sizes and clearances for easy desoldering. This of course doesn't make sense where design constraints require small (eg parasitics, capacitance). What do you have in mind for the following?
- minimum jumper size 
- minimum clearance to adjacent components 

2512 mm is convenient. 1610 is relatively more painful.

gkasprow, 2016-11-29:
I used 0402 components. You can easily replace the with naked eye. Bigger ones cause impedance matching troubles but we can try to mitigate it.
At the moment there are just a few places where such operation may be necessary:
- clocks - single ended/diff from 
- power (i used 2512 jumpers). In case of power switching (local or uRFB) I can add MOSFETs that do such job nicely.
- RC filters in PLL - I uased 0805 capacitors (to enable NP0s)  and 0603 resistors

dhslichter, 2016-11-29:
My preference would be to use 0603 (1608 metric) components where possible/practical, with 0402 reserved for those locations where space constraints and parasitics demand them.  My two cents -- I defer to @gkasprow in selecting what sizes are appropriate in each situation.

sbourdeauducq, 2016-11-29:
Why 0603? 0402 isn't very hard IME.

dhslichter, 2016-11-29:
I agree, 0402 is certainly doable, but 0603 is both easier to solder, and more importantly, offers a larger range of component values and characteristics.  For example, you may desire NP0 capacitors or air-core inductors to give better performance in band-defining filters (lower loss, less nonlinearity).  You can get more capacitance/inductance values, and often better-performing components, in an 0603 package than in 0402.  

I have no problem with 0402, and there are places where it needs to be used for sure.  I am just suggesting that if one could use 0603 instead for some instances, this might be a better option.

jbqubit, 2016-12-02:
@gkasprow It it feasible to use 0603 as the default size for SMD that might reasonably be modified by users (eg jumpers) except if necessitated by operation considerations? Power components and bypass caps can be tiny as users are unlikely to modify.

jbqubit, 2016-12-02:
@gkasprow Please recommend hot air gun with tube and digital control. Do you recommend a particular soldering iron tweezers?

hartytp, 2016-12-02:
@jbqubit https://www.sparkfun.com/products/10706

gkasprow, 2016-12-03:
I have this one (or at least something that looks identical) for over 5 years and works just fine
http://www.ebay.com/itm/858D-Hot-Air-Gun-Rework-Station-SMD-Solder-Heat-Soldering-Digital-3-Nozzles-/322301372916?hash=item4b0aa8a9f4:g:~l0AAOSwImRYH~CA

closed 2016-12-03 by gkasprow