sinara-hw/sinara #92: allaki analog mezzanine: repository name cleanup

jbqubit, 2016-11-29:
- [ ] Why is there both AFE_mezzanine and Sayma_Mezzanine? 
- [ ] Update m-labs/sinara/blob/master/README.rst to indicate high level view of what's in ARTIQ_EE

Refactoring:
- [ ] m-labs/sinara/Sayma_Mezzanine should be moved to m-labs/sinara/ARTIQ_EE/PCB_Sayma_mezzanine_allaki

- [ ] Since clock mezzanine could find application on both Sayma RTM and on RF_backplane distribution board, m-labs/sinara/Clock_Mezzanine should be moved to m-labs/sinara/ARTIQ_EE/PCB_mezzanine_clock

gkasprow, 2016-11-29:
AFE_Mezzanine is simple test board and template for more complex mezzanines.
ArtiQ_EE is set of boards including Metlino. EE means Expdition Enterprise while mezzanines are done in Altium.

jbqubit, 2016-11-30:
I've updated README.md.  https://github.com/m-labs/sinara/commit/766b4147f5ebf12e9818f8aa8a937d3c5b7d622c For consistency how about refactoring as in this commit? It foreshadows other analog mezzanine boards based on the same template.

jbqubit, 2016-12-02:
Thanks for cleaning this up. Looks like there are still some stray files
Clock_Mezzanine/.DXF
AFE_Mezzanine/AFE_mezzanine.*

gkasprow, 2016-12-03:
They have to be removed manually because were added via web interface

gkasprow, 2016-12-03:
done

closed 2016-12-03 by gkasprow