sinara-hw/sinara #93: PCB_mezzanine_clock: schematics for Oxford PLL design

jbqubit, 2016-12-02:
Presently sinara\ARTIQ_ALTIUM\PCB_mezzanine_clock is bare.  @WeiDaZhang send layout to @gkasprow and @sbouhabib.

hartytp, 2016-12-02:
@jbqubit We are still waiting for our proto hardware to arrive. Once we've finished testing that, we'll do a design for the proper mezzanine and post it.

gkasprow, 2016-12-03:
If you need Altium libraries for some components, let me know.

hartytp, 2016-12-03:
Thanks, will do.

WeiDaZhang, 2017-02-20:
We have done some phase noise test on the prototype clock mezzanine board, thanks @hartytp for his great contributions.
The result is close to our previous estimation (within 3-5 dB) at most of the sideband frequencies.
Please find the plot following presents the phase noise measurement of
- the prototype board, along with
- the previous estimation,
- the reference, 
- simulation of some of the noise contributors, as well as
- a Keysight instrument with low phase noise option (extract from datasheet), and
- the phase noise of the DAC - AD9154 itself (estimated based on datasheet).
[image]

We are going to carry out the long-term stability test on the prototype board, as well as optimising the phase noise performance of it. A proper mezzanine board will be designed in parallel with, but not be finished before, the stability test.
Further results, schematic and layout will be posted as soon as we finish them.

P.S. there were people find difficult viewing the figures we plotted last time, apologise for the large files. I've posted a low-resolution version this time. Please let me know if you have any problem on openning it, or want an HD one instead, thanks.

hartytp, 2017-02-20:
@WeiDaZhang thanks for posting that!

I've copied that plot onto the [wiki](https://github.com/m-labs/sinara/wiki/Clock-mezzanines) and added some extra information.

WeiDaZhang, 2017-03-15:
Update:
- The above result is measured with sine wave reference. A 1~2dB better result has been measured with square wave reference converted by ADCLK925 and LTC6957-1. In the real case, the reference should already be converted on the SaymaRTM when reaching the clock mezzanine.
- We are working on the second round of prototype of the clock mezzanine board, which is more close to the final mezzanine.
- The PLL chips are HMC698 - a same series but newer chip to HMC440, with same noise floor and flicker noise, but wider frequency coverage.
- The frequency of the prototype will cover up to 4.8GHz limited by the fanout chip ADCLK948, same as the further stages on SaymaRTM.
- We will finish the noise and stability measurement of the new board, and post here for WUT to take over.
- The time scale will be ~ a month.

hartytp, 2017-03-15:
Thoughts about the WUT version of this board:

- 2 independent PLLs + VCOs sharing a common 100MHz reference input (one DAC clock and one LO)
- PLL locked indicated by LEDs on the PCB, also connected to 2x mezzanine IO TTL
- Output frequency (PLL divider ratio) fixed by component choice, but readable via I2C mezzanine IO
- Suggested default population options: populate both PLL ICs, DAC clock 2GHz, LO VCO + loop filter DNP (will vary between users). 
- External MMCX (or other suitable microwave connector) for an external LO input. Mux allows switching between this and the LO PLL output (e.g. for debugging). Mux controlled by Mezzanine IO TTL. NB No external 100MHz reference or DAC clock inputs needed, since there are options on Sayma/Baikal for this
- Users who want higher LO frequencies could consider putting a frequency multiplier on their up-converting AFE mezzanine (phase noise for this should still be very good)

WeiDaZhang, 2017-07-03:
Following the previous test on the first round prototype of the clock mezzanine board, we have finished the 2nd round prototype and measured phase noise of it.

This round of prototype is aimed to have as many features of the final clock mezzanine board implemented and tested as possible. The prototype is electrically compatible with the Sayma RTM, however, the physical constraints are relatively loose, and it won't mechanically fit the Sayma RTM.

Some of the thoughts @hartytp mentioned are implemented (or close to):
- [x]  _2 independent PLLs + VCOs sharing a common 100MHz reference input (one DAC clock and one LO)_
- The prototype contents two PLL circuits, which can be configured to lock to the reference input at different ratios, namely the DAC clock generator, and the local oscillator.
HMC698 is chosen as the PLL chip, which is the same series but a newer chip to HMC440 - the PLL chip on the previous prototype. The new one covers up to 259x of the reference or 7GHz whichever is lower.
CVCO55CC-2400-2400 is chosen as the VCO for the DAC clock generator. A different VCO in the 0.5"x0.5" common footprint can be chosen for the local oscillator, depends on the frequency requirements.
- [x] _PLL locked indicated by LEDs on the PCB, also connected to 2x mezzanine IO TTL_
- FET isolated lock indicator LED can be easily read as a digital signal.
- [ ] _Output frequency (PLL divider ratio) fixed by component choice, but readable via I2C mezzanine IO_
- DIP switches are implemented for configuring the ratio. Accessing via I2C depends on the IO definition of the mezzanine, and can be added while finalising the mezzanine design.
- [x] _Suggested default population options: populate both PLL ICs, DAC clock 2GHz, LO VCO + loop filter DNP (will vary between users)._
- A suggested bill of material (BOM) is provided for DAC CLK PLL @ 2.4GHz, along with the schematic
- [x] _External MMCX (or other suitable microwave connector) for an external LO input. Mux allows switching between this and the LO PLL output (e.g. for debugging). Mux controlled by Mezzanine IO TTL. NB No external 100MHz reference or DAC clock inputs needed, since there are options on Sayma/Baikal for this_
- The prototype outputs two AC-coupled differential LVPECL signals, each of them can be sourced between one of the PLLs or one of the external LVPECL inputs.
ADCLK948 is chosen as the MUX-fanout chip for its low jitter and temperature coefficient. The operating frequency can be up to 4.8GHz.
- [ ] _Users who want higher LO frequencies could consider putting a frequency multiplier on their up-converting AFE mezzanine (phase noise for this should still be very good)_
- This is not implemented on the prototype, but the output LO will, for sure, support any frequency multiplier further down the chain.

The measured phase noise result is close to our previous estimation (within 3-5 dB) at most of the sideband frequencies, and is almost the same to the first round prototype.
Please find the plot following presents the phase noise measurement of
- the prototype board, along with
- the previous estimation,
- the reference,
- a Keysight instrument with low phase noise option (extract from datasheet), and
- the phase noise of the DAC - AD9154 itself (estimated based on datasheet).

[image]

The schematic of the prototype is attached together with the mentioned BOM.
[PLL_TEST_HMC698.final.pdf](https://github.com/m-labs/sinara/files/1120572/PLL_TEST_HMC698.final.pdf)
[CMP BOM Assembling.xlsx](https://github.com/m-labs/sinara/files/1120573/CMP.BOM.Assembling.xlsx)

We are still going to carry out some long-term stability test, to perform it on the prototype board or the final mezzanine is currently TBD.

hartytp, 2017-07-03:
Thank you for posting that Wieda!

@jbqubit @gkasprow As we're happy with the phase noise achieved in this design, the Oxford part of this project is now essentially complete, so I'm closing this issue. The next stage is for WUT to produce a version of the clock mezzanine that's mechanically compatible with Sayma RTM. Note that the clock mezzanine is a lower priority than Sayma/Kasli/Urukul/Novogorny/Zotino at the moment. 

@gkasprow When you produce your version of the mezzanine, feel free to look over our schematic and suggest changes/simplifications/improvements.

A couple of other comments:

- The mezzanine will probably need quite careful screening to avoid degrading its noise performance when it's mounted on Sayma RTM.
- The fanout buffers are specified to give >0.8V differential output voltage swing up to 4.8GHz. They should be fine at higher frequencies assuming one can live with lower output powers. My guess is that this board will run fine at 6GHz to provide an Yb LO with a doubling stage on the mezzanine (VCO could be something like VCO55CXT-6000-6075). Interested parties feel free to test this!

closed 2017-07-03 by hartytp

gkasprow, 2017-07-04:
@WeiDaZhang Could you send me or publish sources of your design?
Thanks

WeiDaZhang, 2017-07-05:
@gkasprow I'm afraid Cadence organises its files quite chaotically. There is not a ".sch" file can be considered as the source which I'm aware of. I've attached the whole project folder, it includes the "schematic csv" files as well as the pcb files. The paths are:
- Project file
  - \PLL_Test_HMC698\pll_test_hmc698.cpm
- [Schematic folder]
  - \PLL_Test_HMC698\worklib\pll_test_hmc698\sch_1
- [Layout folder]
  - \PLL_Test_HMC698\worklib\pll_test_hmc698\physical
- Layout file
  - [Layout folder] \ PLL_TEST_HMC698_20170515_1.brd
- Gerber files
  - [Layout folder] \ *.art or *.drl

[PLL_Test_HMC698.zip](https://github.com/m-labs/sinara/files/1124353/PLL_Test_HMC698.zip)

Hope it helps.

gkasprow, 2017-07-05:
That's not a problem, I work with CADENCE files.
Thanks @WeiDaZhang

gkasprow, 2017-07-05:
@WeiDaZhang  which Orcad version did you use?

WeiDaZhang, 2017-07-05:
@gkasprow It's Cadence® Allegro® 16.6 -> Design Entry HDL