sinara-hw/sinara #95: Sayma RTM

hartytp, 2016-12-03:
@gkasprow a few minor comments/questions/fixes for the current Sayma RTM schematics:

- [ ] **LDOs**: IC19 (P3V3_CLKDIS reg) should be LT3045. What is IC20? Should it also be LT3045? IC32 (SI5324 LDO) is LT1963. Should we replace this with LT3045 as well? Both ICs would do the job, and seem to cost similar amounts (looking on Farnell quickly), but switching would allow us to eliminate a BOM line.

- [ ] **uRFB_REF_LO**: please can we add an extra RF backplane signal uRFB_REF_LO? This should connect to a balun and pair of connectors, just like uRFB_DAC_CLK_IN. That way, all clock mezzanine functionality can be provided by the RF backplane. (These baluns/connectors can be DNP in the current hardware round if desired)

- [ ] **Name/annotation for J67-69**: what are J67-J69 for (debugging outputs?)? Please can these connectors be named, or can we have an annotation on the schematic indicating their intended function?

- [ ] **SI5324 CLKOUT2 termination**: should the SI5324 100R termination resistor (R57) be AC coupled to the SI5324, per data sheet figures 3-4?

- [ ] **Termination + grounding for isolated SMA inputs**: current termination for SMA inputs, such as uRFB_REF_IN will not work (my fault), since the signal is terminated to the circuit ground, which is only weakly coupled to the SMA ground (10pF@100MHz = 160Ohm). Also, since the balun is a short between signal and ground (pins 1, 2 & 3) at DC, we need to AC couple the balun ground to circuit ground if we're going to break ground loops (connect it directly to input SMA ground?). Is this acceptable from an EMC/ESD perspective?

- [ ] **TCM2-43X+ connections**: shouldn't pin 5 of the TCM2-43X+ baluns be connected to ground? Should we add small capacitors between the secondary outputs and ground to improve high-frequency performance (values to be determined from simulation?)? See data sheet/ref design for details on these points. This may be affected by the previous point.

- [ ] **ADCLK948 package typo**: should _VR0_ be _VT0_?

hartytp, 2016-12-03:
@gkasprow Wow, that was quick! Could you post updated schematics, please?

gkasprow, 2016-12-03:
done

hartytp, 2016-12-03:
@gkasprow Thanks for doing that Greg!

- [ ] Sorry for not picking this up earlier, but I just noticed that R237 (50R termination on uRFB_REF_IN) is not needed, since the ADCLK948 includes on-chip 100R termination. Can we remove this, please? (Obviously, keep R98 termination for EXT_RF_IN since LT6957 does not include termination).

- [ ] IC20 still missing a part number.

Anyway, that's my last look at the schematic before hardware arrives...

gkasprow, 2016-12-03:
> @hartytp
>
I was thinking about this termination.
Maybe it's better to add 100R termination at the IC53 input instead of R98?

hartytp, 2016-12-03:
@gkasprow 

> Maybe it's better to add 100R termination at the IC53 input instead of R98?

Yes, I think that would be better.

jbqubit, 2016-12-05:
@gkasprow schematics don't yet reflect the two last bits Tom added.

gkasprow, 2016-12-06:
Indeed, the pdf for some funny reason is outdated. Source files are OK

jbqubit, 2016-12-08:
Still missing from PDF.

hartytp, 2016-12-11:
@gkasprow One more thought: is it worth adding a 100k resistor between each isolated ground and circuit ground? In cases where the isolated grounds are not connected to a well-defined DC potential, this prevents them from charging up, while still providing a high enough impedance to break ground loops.

gkasprow, 2016-12-11:
Good idea.

hartytp, 2016-12-12:
@gkasprow on the current version of the schematics, R237 is still present. When you have time, please could you remove this, push the new version and close this issue?

closed 2017-01-26 by jbqubit