sinara-hw/sinara #97: prototype 0.1: packing list

jbqubit, 2016-12-04:
Distribution plan for hardware for prototype 0.1 (aka prototype 1) is #21. This Issue collects the individual components that will be shipped from WUT to each lab doing testing.

Item | Total Quantity | Distribution 
------|-----------------|-----------------
PCB_metlino | 4 | 1 for each of m-labs, UMD, Oxford, NIST
PCB_sayma_AMC | 7 | 1 for each of m-labs, UMD, Oxford; 1 for NIST 
PCB_sayma_RTM | 7 | 2 for each of m-labs, UMD, Oxford; 1 for NIST 
PCB_mezzanine_analog_allaki | 28 | 8 for each of m-labs, UMD, Oxford; 4 for NIST
PCB_mezzanine_clock_template | 4 | 1 for each of m-labs, UMD, Oxford, NIST
PCB_baikal | 4 | 1 for each of m-labs, UMD, Oxford, NIST

Items needed for stand-alone testing of Sayma boards without uTCA.4 chassis.

Item | Total Quantity | Distribution 
------|-----------------|-----------------
ATX power supply | 4 | 1 for each of m-labs, UMD, Oxford, NIST [0] 
ATX cable splitter 1:2 | 12 | 3 for each of m-labs, UMD, Oxford, NIST [0]
SATA to SFP | 11 | 3 for each of m-labs, UMD, Oxford; 2 for NIST [1] 

NOTE: PCB_mezzanine_clock is not part of prototype 0.1 (per call on 12/2)
[0] ATX supply assumed to have at least 3 independent outputs; splitter is needed since PCB_Sayma_AMC and PCB_Sayma_RTM are independently powered
[1] one for PCB_Metlino_AMC and one for each Sayma

jmizrahi, 2016-12-05:
For PCB_sayma_AMC, shouldn't this read "2 for each..."?
For PCB_mezzanine_analog_allaki -- there are 4 allaki mezzanines per Sayma? Not 2?

hartytp, 2016-12-15:
@gkasprow Is it too late to increase the number of Sayma + Metlino we want to order in the initial production round?

hartytp, 2016-12-15:
@jbqubit Assuming we don't find any major mistakes in the prototype hardware, when do you think a second hardware order will take place? This will affect how much prototype hardware we buy.

jbqubit, 2016-12-16:
@hartytp New issue for schedule discussion #131.

jbqubit, 2016-12-16:
Updated leading post with front panel delay info.

hartytp, 2016-12-16:
@gkasprow @jbqubit I've updated the top post with the amount of hardware we'd like to buy for this round. Please let me know if there's an issue with this.

gkasprow, 2016-12-16:
I reserved 11 Kintex FPGAs, will try to reserve one more.

gkasprow, 2016-12-16:
Assembly house says that It could be impossible to chance the order but they will try.

hartytp, 2016-12-16:
@gkasprow Thanks, let me know how it goes.

hartytp, 2017-02-01:
@jbqubit What's the plan for ordering the VHDCI carrier, 3U BNC board, etc? Do you want to use this issue to coordinate?

@gkasprow 
> I reserved 10 Kintex FPGAs, will try to reserve one more.

How did this go? How many Sayma boards will get in the first prototype round?

jordens, 2017-02-01:
@hartytp @jbqubit I am collecting quantities for Kasli, DDS, VCO, 3U_BNC etc anyway. I can do that.

hartytp, 2017-02-01:
@jordens Okay, thanks. I'll have a think about how many we want and let you know.

What's the funding situation + timeline for Kasli, DDS & VCO?

gkasprow, 2017-02-01:
We have 12 FPGAs delivered.

hartytp, 2017-02-01:
> We have 12 FPGAs delivered.

Great! So, we're okay for the hardware quantities currently listed in the table at the top of this post, right?

gkasprow, 2017-02-01:
Providing that all produced boards work :)

jordens, 2017-02-01:
@hartytp https://github.com/m-labs/sinara/wiki/KasliQuantities
From the looks of it we can fund Kasli, the DDS module, and a I2C "Hub/switch" module.
We aim to have that first batch out by August.
VCO, DAC would still need funding for development.

hartytp, 2017-02-01:
@jbqubit @gkasprow Is the RJ45 extension module included in the current round of work?

jbqubit, 2017-02-01:
@jordens Please add a table at the top of the Kasli wiki to track the status of Kasli and the 3U peripherals. 
- funded (and by whom if that's public)
- specification completed
- schematic finished
- production
- testing
This will help us all track development. And encourage the community fill in gaps.

gkasprow, 2017-02-01:
We can try, providing that we fully specify what you want on fith connector :)

hartytp, 2017-02-01:
@gkasprow in #129 

> The fifth R45 was meant as I2C link for slow devices, but I don't have strong opinion about it.

@dtcallcock also from #129 
> I'd probably rather leave this off due to I2Cs lack of robustness to added bus capacitance and possible voltage level incompatibilities. SPI will be better for off-board devices. If remote I2C devices are really desired though, we could utilise bus buffers like these from Intersil.

+1 to leaving the 5th connector off -- particularly if a separate I2C hub module is planned.

IMO, having the RJ45 module do I2C as well violates the "cheap boards performing a single function" model we're going for...

----

Other than that, is the specification for this board complete? Also, has this board been updated in light of any relevant feedback from the 3U_BNC board?

dtcallcock, 2017-02-01:
@jordens https://github.com/m-labs/sinara/wiki/KasliQuantities

What is 'SPI' on this table? Would also be good if I2C could get a brief description on the main Kasli wiki.

gkasprow, 2017-02-02:
OK, so I will just use 4x RJ45 and in remaining space place LEDs indicating port directions.
I did it in case of 3U BNC
[image]

In case of RJ45 it's hard to fit 16 standard LEDs so I will use compact ones:
[image]

jbqubit, 2017-02-21:
What's in the packing list is established. Closing Issue as there's been no recent discussion.

closed 2017-02-21 by jbqubit