sinara-hw/sinara #99: PCB_io_19in_sma: specification discussion

jbqubit, 2016-12-04:
ARTIQ hardware plan calls for a 19" rack breakout for digital signals. Proposed name: PCB_io_19in_sma. Motivation is low-latency digital IO, coaxial cable drivers (50 Ohm) and electrical isolation. Proposed specification follows. Informed by email discussion 6/28 with @gkasprow, @sbourdeauducq, @jordens and @dhslichter. 

# front-side connectors
- electrical interface: support for interface with kasli and metlino_hvdci
- aperture for kasli SFP connector
    - kasli board would be mounted horizontally as mezzanine on PCB_io_19in_sma
    - requires that we define the following for PCB_kasli: geometry, header location, header logic 
- HVDCI connector (eg Molex SD-71430-015)
    - 32-differential LVDS
- 32 isolated SMA connectors in two rows

# back-side connector
- AC connector is IEC 320 C13/C14

# power 
- source: 220V at 50 Hz or 110V at 60 Hz
- supply: AC DC converter of some sort

# geometry
- breakout is 19" rack panel, 1 U height 

# digital interface to ARTIQ control hardware
- assume one of the following is supplied by end user
    - hvdci cable from PCB_metlino (LVDS)
    - PCB_kasli mounted as a mezzanine and commanded by SFP

# SMA drivers/receivers  
- direction (input/output) user configurable in groups of 8 
- optional galvanic isolation breaks ground loops independently for each SMA 
    - jumpers permit use (or not) of isolation IC
    - @dhslichter recommends Si8640, 4 channel, 150Mbit,  13 ns propagation delay, ~350 ps jitter; $4/chip
    - to support reconfigurable direction
        - use 4 Si8640 for each bank of 8 SMA
        - Si8640 has pin to toggle output drive on/off, use to select direction
    - SN74BCT25245 is octal, reconfigurable direction, 25-ohm driver
- if bank of 8 is configured as _input_
    - install jumper to choose direction of Si8640 and SN74BCT25245
    - install 1 kohm input termination
- if bank of 8 is configured as _output_
    - install jumper to choose direction of Si8640 and SN74BCT25245
    - install 50 Ohm output termination (series resistor as close to the driver)

It would be convenient for testing of v0.1 Sinara hardware if this PCB is available to ship at same time.

gkasprow, 2016-12-05:
@jbqubit 
Are you sure about 1U?
3U would be more convenient because we could use low cost racks and install many Kasli-type boards with several IO options.
3U Eurocard is standard with defined dimensions, connectivity, panels, cooling. You can have 19" rack or bench version of the enclosure.

jordens, 2016-12-05:
@jbqubit I would strongly advise against building yet another full size rack for Kasli. This is not what we intended it to look like and not how we would see it being used. If you conclude that you need a break-out rack for Metlino/Sayma because they are so high density, then build one. But Kasli should not be mixed in there. For Kasli, we have a few groups here in Germany who are interested and willing to do detailed testing work and give input on the design.
Since you pushed it back on your list of preferences and funding, why don't you let others take the lead here?

* I'd like Kasli to be small enough that you could place it on the optical table right at the trap and connect 40-100 few-kHz DAC channels or SPI for in-vacuum DACs or miscellaneous front-end electronics to the apparatus.
* I'd like it to be (intended to be) used with just some random wall wart with barrel connector to the back.
* That wall wart power supply provides galvanic isolation.
* SFP for DRTIO uplink and whatever outputs/inputs to the (its!) front panel.
* I want to see it inside one of [these frame enclosures](http://schroff.pentair.com/en/schroff/plug-in-units---frame-type---kits/3074457345616861429)
* If one wants to pack multiple of them, those can still go into a subrack be it [small](http://schroff.pentair.com/en/schroff/Ratiopac-PRO/3074457345616862510) or 
[large 19 inch](http://schroff.pentair.com/en/schroff/Products/3074457345616862554)
* 3U, vertical PCBs, preferrably 100x160mm eurocards.
* No fans moving parts inside.
* If they are placed in a rack, they can power from the backplane (same single-supply voltage) break out to the backplane and then people can do what they want with those signals (e.g. several SPI and several digital lines).
* Don't shoot for ridiculously high density. Neither on the front panel nor on the PCB.
* Don't use VHDCI connectors since that high density stuff needs to be broken out in different hardware again. Just do SMA, BNC, or D-sub.
* Really make this simple and cheap, not a monster. Requiring extensions to extensions to extensions seems unwise. Make it so simple and cheap that it is immediately preferable to use two Kaslis instead of one Kasli with an "extended" mixture of functionalities.
* Make this a simple platform. I.e. Kasli would be a carrier (FPGA, SFP, power supply, DRAM, SI5324, oscilltors, USB/JTAG) plus one or two stacked mezzanines (16 TTL drivers *or* 40 DAC channels *or* CameraLink connector).

dhslichter, 2016-12-05:
I tend to agree with @jordens here, @jbqubit.  Simple, cheap, small, common connectors, few moving parts, and build up complexity with multiple Kasli boards.  Eurocard seems like a reasonable form factor to me, although we'd want passive cooling only, so that will limit density.

dtcallcock, 2016-12-05:
Leaving Kasli aside, do we still want this PCB_io_19in_sma purely as a breakout for Metlino's VHDCIs?

If so:

- It would be nice to have the option (via jumpers) to send 1 or 2 SPI busses out on RJ45 connectors (that each carry 4xLVDS).  This is how we do it with our current hardware and I much prefer it to stringing SMAs.  If people like this idea we could also think about designing a simple receiver board that hangs off a panel mount RJ45. People can use this to retrofit existing SPI devices or quickly box up eval boards.  We already have something that does this but it could be simplified quite a bit.

- Each VHDCI also has an I2C bus wired to it.  This could also be broken out to an RJ45.  One way of doing this robustly would be to convert to a CAN bus [like this](http://mayhewlabs.com/products/i2c-power-extender).  We are about to test one of these as a way to break the I2C bus out of our current ARTIQ crate.

- The Metlino has 2x VHDCIs.  Is there any reason not to break them both out on the same board?  Obviously it'll be a bit more expensive but not twice as much as there'll be only one box/PSU/etc.

- I there the possibility of making an FMC-VHDCI board for the Sayma_AMC that would also work with this breakout?  This assumes there's no other plan for getting reasonable digital IO from Sayma if required.

gkasprow, 2016-12-05:
The idea was to build Kasli replacement as Metlino extension. Just to use
Kasli extension boards with DACs, ADCs, IO, isolated IOs, etc. That's why
I'd make it in EURO form-factor with identical headers as in Kasli case.
Connectivity would be limited to 32 or 64 channels. The board would have
just LVDS-TTL converters
We already have FMC-VHDCI board and Metlino has same pinout
http://www.ohwr.org/projects/fmc-dio-32chlvdsa/wiki/Wiki

>
>

jbqubit, 2016-12-07:
> why don't you let others take the lead here?

I don't intend to take the lead on Kasli. I support others taking a leadership role on the Kasli. @jordens please consider creating a Sinara Issue where people can comment on the Kasli specification on the [wiki](https://github.com/m-labs/sinara/wiki/Kasli). What follows depends on some decisions as to the location and electrical of the Kasli 100 mil header.  

>For Kasli, we have a few groups here in Germany who are interested 
>and willing to do detailed testing work and give input on the design.

Splendid! Is there a timeline for this? Agreed that Kasli is a good approach for many-channel slow DACs. 

> I would strongly advise against building yet another full size rack for Kasli.
> This is not what we intended it to look like and not how we would see it
> being used. If you conclude that you need a [...]

My intent is for PCB_io_19in_sma to act as breakout for Metlino VHDCI IO. As a forward-looking measure I anticipate that it could also serve as a breakout option for Kasli IO. 

>...Are you sure about 1U?
>...RJ45 connectors (that each carry 4xLVDS)
>...Each VHDCI also has an I2C bus wired to it.

I'd be fine with a 3U solution as well. Same intent of driving the SMA breakout board with either VHDCI or Kasli Carrier. Here's a refactoring in light of the comments. 

# Euro Kasli Carrier
Suppose that the Kasli [spec](https://github.com/m-labs/sinara/wiki/Kasli) is refined to specify
- [x] PCB is 3U height
- [ ] front panel width is 4 HP
- [ ] near leading edge of PCB is a left-facing 100 mil header 
    - [ ] 2x34 pins for LVDS, I2C and +12V at 100 mA

# Euro breakout for 68-pin VHDCI 
- geometry and header shadows Euro Kasli Carrier
- front panel is 68-pin VHDCI connector

# Euro SMA/RJ45 breakout
- PCB is 3U height, width is 20 HP
- PCB is mounted behind and parallel to front panel
- front panel width is 20 HP
- front panel is populated with
    - 16 SMA
    - 4 RJ-45 with 4 LVDS each, +12V 20 mA
    - 1 RJ-45 with 1 I2C, +12V 20 mA
- 100 mil header at right-facing end of PCB plugs into Euro Kasli or Euro VHDCI breakout 
- SMA drivers/receivers as above

jbqubit, 2016-12-09:
@gkasprow Please expound on your idea of using ribbon cables.

gkasprow, 2016-12-09:
I've created some drawings of Metlino IO breakout / Kasli ecosystem
It is the Metlino VHDCI base board. 8 IDC connectors are used to connect the board with extension IO modules. Each connector routes the signal for 8 channels.
[image]

Kasli has the same form-factor:
[image]

Extension modules are connected using IDC ribbon cables like this one.
They work well with high frequency digital lsignals. We will use diff signalling if necessary. They are widely used for PCIe extensions [(link)](http://www.dx.com/p/pci-e-16x-male-to-female-extension-cable-20cm-127599#.WEsbK1w1rZ4)
[image]

And here are the extensions.
They can be stacked on the baseboards like this isolated 8-channel BNC, 8 HP
[image]

Isolated 8-channel SMA, 4HP
[image]

LVDS RJ45
[image]

Another option is to attach modules to the panel directly. This is a case of 28-channel DAC
[image]

The modules can be combined in standard 3U crates or stand-alone boxes like this
[image]

Inside there are just the IDC ribbon cables
[image]

gkasprow, 2016-12-09:
Look here
https://github.com/m-labs/sinara/issues/99

dtcallcock, 2016-12-09:
Looks great.  Few minor thoughts:

- Given the modules don't sit in the backplane, should they get power over the IDC/ribbon as well as the 8 LVDS pairs?  This would allow for powering of things like isolators?

- By adding two more pins to the IDC/ribbon and adding multiplexer chips to the Metlino VHDCI base board, you could send I2C to each module.  This could be used for module identification and temp sensing like in the Sayma analog mezzanines.  Also could be used for setting TTL directions (like we do in current NIST ARTIQ hardware) and similar future extensions.

- You can fit 4xLVDS on an RJ45 so that board would want 2xIDC headers

jbqubit, 2016-12-09:
Please consider adding two additional through-PCB holes to the Kasli near the back. Use case is standoffs when attaching a daughter card securely to Kasli in the absence of a chassis.

gkasprow, 2016-12-09:
These are not complete designs but just random components placed on PCB to realize the idea. The IDC connector pin count will be different.
The supply is delivered only to the baseboards, then is distributed using IDC cables.
Usage of I2C for module identification is good idea, in the same way as we have on Sayma mezzanines.
Please specify in one place all possible modules you want to have.
In case of Kasli and other modules we will add holes to enable operation outside of chassis, without the front panels (low cost version)
Maybe it's worth to start another project for Kasli / Metlino VHDCI ecosystem?

gkasprow, 2016-12-10:
Is it really necessary to have all SMA/BNC isolated individually?
It would be much cheaper to have group of 4 SMA/BNCs isolated from the baseboard.
Direction of all group would be also common

hartytp, 2016-12-10:
IMO individual isolation of each channel is not necessary. Having groups of 4 (or 8 if it helps) isolated should be fine.

jordens, 2016-12-11:
The idea with the ribbon cables is a bad one. It implies using a crate. Kasli would be much smaller and simpler if (as explained above) it could be run just in the frame enclosure without extension board dangling around.
There is also no plan from to support an exponentially large set of combinations of eight of those driver modules from the gateware and runtime side.
Why not just put the the driver board into the Kasli frame enclosure. That would reduce the minimum size of Kasli to a single ~12 HP by ~3 U by ~160 mm box with full functionality.
It's absurd to invent another crate-backplane ecosystem. Just observe how the complexity of that ribbon cable is exploding already. It's not what we would like (or buy) for the projects here.

gkasprow, 2016-12-11:
@jordens 
No, you don't have to use crate. It is one of many options. You can use boards stacked one on another with distances, without front panels and let it stay on your desk.
You can also use little enclosures which holds just Kasli/Metlino and one or a few IO boards.
There are plenty of such enclosures, some of them work with panels, some of them require you to design custom panel
Examples:
https://www.amazon.com/1455N-Silver-Aluminum-Sized-Guide/dp/B004RIKZ6A
http://pixustechnologies.com/products/custom-chassis-solutions-4/instrumentation-cases/versatile-series/
http://uk.farnell.com/vero/114-52008l/case-eurocard-3u-6hp/dp/1277439
http://www.srs-products.co.uk/netcase.htm
http://uk.rs-online.com/web/p/rack-panels/1866766/

IDC ribbon cables are vely low cost and popular.
Such 3U approach gives many mechanical options.
From firmware point of view it is very simple - IO boards are always multiples of 8 ports, each board has EEPROM with ID so you can easily identify them.
For Metlino I will make simple VHDCI to IDC adapter to quicky test its functionality.
Kasli will follow this approach.

jordens, 2016-12-11:
Sure. The Front panel is not an issue. And I have nothing against IDC cables and connectors per se. Similarly wire wrap posts are also very low cost and popular ;)

AFAICT the other enclosures you cite would have different and (w.r.t. those Schroff front-panels) mechanically incompatible ways to secure the PCBs to the front panel. How does that work? And do they cope with the DIN 41612 connector protruding beyond the PCB in the back? If those are not issues, I am happy with your ribbon solution if I can easily fit e.g. a single Kasli and a X channel slow DAC into a box and run them like that.

The problem with the amount of possible configurations is that we can't make a gateware that fits them all (SPI/I2C/TTL/CameraLink on every driver board). Even with multiple layers of runtime-configurable muxes (which is complex and slow) this would not be much fun. Identifying a driver board is not the big challenge.

I see no real reason to make the ribbon link differential. This is inside the box, with uncontrolled impedance, no danger ground loops and insensitive to noise.
But I do see the desire to provide a good array of power supply rails to the driver boards.
To me it seems that the interface we have for the analog mezzanines on Sayma (minus the ADC channels and VCM) would be a good one: 16 I/O, plenty of power supplies, I2C.

Implementing the galvanic isolation on groups-of-four (or in the case of the slow SPI DAC, the entire array) is also fine by me.

jordens, 2016-12-11:
or 8 I/O single ended plus 5 power supplies plus ground plus I2C: 20 pins.

gkasprow, 2016-12-11:
The only reason I made ribbon cable differential is that  LVDS over RJ45 was specified.
Since it does not make big difference where we put LVDS to LVTTL translation, I decided to put it on the IO module. Translation  from LVDS->LVTTL->LVDS is not optimal.
 Moreover it gives more possibilities in case of fast interfaces, i.e. one can use Analog Devices  LVDS isolators.
Nobody said that you have to support all possible configurations of IO modules. I just want to maximize IO module reuse.
In case of the enclosures, I have 12 or 16HP boxes in my lab so I can check. They work with panels and DIN connector well.
Hammond boxes would need dedicated, common front panel and DIN would have to be not assembled. Anyway, we don't have any special use case for them so can be left DNP as default.

jordens, 2016-12-11:
IMHO the IO standard over the ribbon cable should remain opaque. The TTL/SMA/BNC driver module would use bidirectional level converters to whatever the desired voltage is, the RJ45 would either use LVTTL-LVDS converters or just use the FPGA outputs directly.
From Metlino there is the usual problem that a each bidirectional signal requires three pins if it is supposed to be configurable at runtime.

gkasprow, 2016-12-11:
Metlino outputs LVDS links that can be used as 2 single-ended for low speed operation. The same applies to Kasli.  So if we define ribbon cable signalling as differential, in some cases can double the number of IOs.
We can use I2C extender to configure direction of pins. Do you need to configure direction of every IO or in groups of 4 would be sufficient ?

jordens, 2016-12-11:
Four is sufficient. But do we expect every translator on the way to be able and willing to switch direction, and the entire chain to do so in the right sequence?

gkasprow, 2016-12-11:
This I don't know.
I have LVDS bidir transceivers and can change direction individually or in groups of 4.
At the moment direction is defined manually with jumper per 4 IOs. I can add I2C extender that would do it remotely if needed. Some LEDs can be added to indicate selected direction.
Here is proposed schematic of the BNC IO module.
[PCB_3U_BNC.pdf](https://github.com/m-labs/sinara/files/644701/PCB_3U_BNC.pdf)

gkasprow, 2017-01-30:
I want to produce prototypes of Metlino VHDCI breakout, 3U BNC adapter, Sayma AFE mezzanine template, Sayma clock mezzanine template, Sayma RTM loopback adapter, Mordovia.
Are there any comments or changes?

jordens, 2017-01-30:
@gkasprow Thanks, especially for providing all the PDF schematics and layouts. Gerbers would be awesome if you want us to properly look at the layout. I didn't do a thorough schematics or PCB review.

* Could you put silkscreen info on all the PCBs where possible ([#123](https://github.com/m-labs/sinara/issues/123#issuecomment-266282945))? Most still have that information missing.
* 3U_BNC: ok by me. I assume you are convinced that the DCDC converter will be "low noise enough"? I know that Traco has some lower noise ones but you would have the experience to decide that.
* VHDCI breakout: ok by me.
* Sayma AFE and clock mezzanine: ok by me. I guess the DIP switches can do 0.75A.
* Mordovia: is that tongue connector there really just a dummy? Can it be omitted then?
* RTM loopback: I can't seem to find that in the repository or in any of the issues.

jordens, 2017-01-30:
Oh. And for all the jumpers and DIP switches please add silkscreen info about what the jumpers do and which position does what.

hartytp, 2017-01-30:
@gkasprow The headers on the VHDCI board look pretty closely spaced, which might make connecting cables a bit of a pain. But, maybe there isn't much you can do about that given the space constraints?

Other than that, all looks good to me!

gkasprow, 2017-01-30:
VHDCI is a nightmare. The only way to extend is to use SCSI cables with bigger MDR-like connectors on another side

gkasprow, 2017-01-30:
> @gkasprow <https://github.com/gkasprow> The headers on the VHDCI board
> look pretty closely spaced, which might make connecting cables a bit of a
> pain. But, maybe there isn't much you can do about that given the space
> constraints?
>
> they are perfectly fine.
VHDCI connectors are not symmetrical and you insert them back-to-back.

gkasprow, 2017-01-30:
> I don't want to mess up the repo with production files.
>
I'll add silkscreen info.
In case of Mordovia, the tongue connector is just a dummy - it adds
distance between tongue 1 and tongue 3. can be replaced by something else
but it needs some work.
I'll add RTM loopback design.

>
>

hartytp, 2017-01-30:
@gkasprow I meant the 100-mil pin headers for the extension boards, not the VHDCI connectors themselves. It looks like everything will fit together fine, but it'll be quite a fiddle if ribbon cables need to be changed while the board is in a rack.

gkasprow, 2017-01-30:
The idea is to pull out the VHDCI/Kasli board and disconnect the cables.
The 3U cards are attached to the chassis using two screws like these
http://uk.rs-online.com/web/p/rack-mounting-hardware/7212957/
so no tools are needed.

hartytp, 2017-01-31:
@gkasprow Okay, sounds good.

closed 2017-02-16 by jbqubit