sinara-hw/sinara #101: sayma_RTM missing temperature sensor

jbqubit, 2016-12-06:
Where are temperature sensors on Sayma_RTM? Sayma_AMC schematic includes a Thermometers block. Maybe I've missed it. In #11 @dhslichter proposed using ADT7420.

jbqubit, 2016-12-06:
I see. It's LM75AIMM, a low performance chip (9-bit, +/-1.5 C, no long-term drift spec). We discussed on today's call. 
- [ ] do we have need for higher performance sensor on RTM (eg ADT7420)?
- [ ] what is performance of the temperatuer sensors integrated into the ADCs and DACs

gkasprow, 2016-12-06:
they are on p.15
https://github.com/m-labs/sinara/blob/master/ARTIQ_EE/Sayma_RTM.pdf

jbqubit, 2016-12-07:
ADT7420 proximal to ADCs and DACs. Strong thermal coupling to PCB.

gkasprow, 2016-12-07:
done, added 2 sensors, connected to the I2C switch.

closed 2016-12-07 by gkasprow