sinara-hw/sinara #103: Allaki output DC blocks

dtcallcock, 2016-12-06:
100pF is used throughout.  Will give high-pass at 32MHz into 50 Ohm.  The component on the board that limits the low end of the frequency range right now is the transformer which is specced down to 4.5MHz.  The DC blocks should work down to that frequency at least.  Suggest 10nF.  [ATC 520L](http://www.atceramics.com/series.aspx?sid=37&series=520L-Series) are nice but might be overkill at $0.80ea.

sbouhabib, 2016-12-19:
I replaced with a 10 nF, I had in the lib, the question do you want to go with the 520L or just "standard" cap or other?

dtcallcock, 2016-12-19:
I think a cheaper part will be fine.  I like Murata as they [publish s-parameters](http://www.murata.com/en-us/tool/sparameter/mlcc?intcid5=com_xxx_xxx_cmn_nv_xxx).  How about GRM155R71C103KA01?  It's 10000pF ±10%, 16V, X7R 0402 (1005 Metric) and costs around a cent.  From the attached S-parameters it is <0.05dB loss from 1.3MHz to 3.4GHz.

[GRM155R71C103KA01.s2p.zip](https://github.com/m-labs/sinara/files/662409/GRM155R71C103KA01.s2p.zip)

dtcallcock, 2016-12-19:
Should also note that C34 and C35 (see #109) are 100pF.  This is less of an issue if the boards are stuffed with zero ohms here (wasn't made clear what the default config would be was in #109), as then the user can pick whatever value they like if they decide to install.

jbqubit, 2016-12-21:
From call today... specify properties of capacitors
- [ ] dielectric material: eg X7R
- [ ] tolerance
- [ ] voltage rating

dtcallcock, 2016-12-21:
> If this is a problem in the future

Given the other components in the path it'll probably be hard to tell if the cap they use has weird loss/parasitics without a test fixture.  Something like the pictured eval board but having paths with and without the cap would work. Obviously would need to use same stackup and CPW dimensions as actual mezzanine.  Using the same SMAs as the mezzanine would also allow an independent check of those.  

[image]

TBH though, this is pretty far down the list of things to be worrying about and optimizing at this stage, especially at these low frequencies.

closed 2017-01-26 by sbouhabib