sinara-hw/sinara #105: sayma_RTM: ADC VREF

jbqubit, 2016-12-06:
In email discussion on 7/28 @dhslichter said. "The AD9656 should have a 1.4V reference, common mode voltage of 0.9V, and can thus take 2.8 Vpp differential swings at its input.  This corresponds to VIN+=1.6, VIN-=0.2 and the reverse (i.e. each individual line can swing +/- 0.7 V around the common mode level).  Absolute maximum ratings are -0.3V and +2V for the device."

- [ ] VREF generated for AD9656 by IC2 is +1.2V, right? (p.1 of schematic). Did I miss the discussion where we moved away from 1.4V? @dhslichter 
- [ ] state the voltage of VREF on the schematic ADC_JESD204B and near IC2 (p1)
- [ ] use temperature grade A for LM4140ACM-1.2 to get 3 ppm/C

gkasprow, 2016-12-07:
@dhslichter , @jbqubit do you know good single digit ppm reference that generates 1.4V?

hartytp, 2016-12-07:
Is there any need for the reference to be that good? IIRC, it's just used for the ADC. 1LSB is 15ppm for 16 bits, and the board temp should be stable to ~1C with active fan control.

gkasprow, 2016-12-07:
That’s true, we don’t need that good one. But  it’s hard to find 1.4V reference. One can always use adjustable one or some dividers but I’d like to avoid it.

hartytp, 2016-12-07:
@gkasprow True, it's an odd voltage. Do we need to use 1.4V, or can we get away with 1.2V or something else a bit more standard? Or, use the internal reference?

gkasprow, 2016-12-08:
We can always use internal reference – the SENSE pin defines reference source. I added eternal reference to have same value for both ADCs.

We can also use internal reference of one ADC by another ADC – just remove reference source and resistor at SENSE pin of one of ADCs.

dtcallcock, 2016-12-08:
Internal reference looks to be about 45ppm/C.

If you went to a 1.25V external ref there are good options (eg MAX6070B -> 8ppm/C, 3.6uVpp noise, 0.08% accuracy, SOT23, <$0.93).  Looking at the AD9656 datasheet, it doesn't look like this would effect performance much (they spec at Vref = 1.0V and 1.4V - some things get a little better, some a little worse).  It does change the input range from 2.8Vpp to 2.5Vpp though.

> We can also use internal reference of one ADC by another ADC – just remove reference source and resistor at SENSE pin of one of ADCs.

If you do this then note the comment in the datasheet that says "If the internal reference of the AD9656 is used to drive multiple converters to improve gain matching, the loading of the reference by the other converters must be considered. Figure 52 and Figure 53 show how the internal reference voltage is affected by loading."

hartytp, 2016-12-08:
agreed, 45ppm/C for the internal reference isn't really good enough.

+1 for using a 1.25V (or other convenient voltage) external reference.

jbqubit, 2016-12-08:
@dhslichter What are implications of 1.2V vs 1.4V ?

jbqubit, 2016-12-12:
@dhslichter ping

dhslichter, 2016-12-15:
Sorry, have been on travel, very distracted.  See above from @dtcallcock -- you get smaller dynamic range for input signals, thus you degrade your SNR/ENOB/etc by this amount because the ADC input noise is in absolute voltage, not fraction of full scale.

jbqubit, 2016-12-15:
How about using +2.0 V version (LM4140ACM-2.0) and voltage divider to get +1.4V?

gkasprow, 2016-12-15:
This is what I proposed. In addition to the voltage divider, a buffer is needed. And here opamp with very low input offset must be used.

I.e. ADA4528 should do the job.

For division purposes, matched resistor pairs could be used, i.e. Vishay Y1365V0548QT9R

dtcallcock, 2016-12-15:
Comparing pages 5 and 6 of the datasheet shows that going from 1.4 to 1V gives something like a 0.1-0.2 reduction in ENOB (and actually an improvement in SFDR and harmonics).  Given we're only talking about going from 1.4 to 1.25V it doesn't seem worth adding BOM lines for.

jbqubit, 2016-12-15:
That's not much of a win. I concur with @dtcallcock. Not worth the additional materials.

gkasprow, 2017-01-12:
we leave it as it is.

closed 2017-01-12 by gkasprow

jbqubit, 2017-01-24:
@gkasprow Please confirm the following from top of this Issue.
- [ ] state the voltage of VREF on the schematic near IC2 (p1)
- [ ] use temperature grade A for LM4140ACM-1.2 to get 3 ppm/C (note on schematic)

gkasprow, 2017-01-29:
Done.
We use grade A from the very beginning.