sinara-hw/sinara #108: allaki: DAC_Filter block

jbqubit, 2016-12-06:
FL2, FL3, FL4 (on DAC_Filter.SchDoc) are intended to be band defining for different end user applications. Outside of these filters the design bandwidth for Allaki is 10 MHz to 3.5 GHz. 

# Two Approaches
This schematic element was discussed in today's call. Two approaches

1. Use MiniCircuits in standard package [FV1206](javascript:ProductWindow('/MCLStore/ModelSearch?model=fv1206&search_type=casestyle');). 
Pro: this is a commercial item that's easy to source and install by end users. 
- Con: Mini-Circuits doesn't support Bessel filters. 
- Con: No Mini-Circuits FV1206 component may exist for some user applications.

2. In layout include pads for 9-pole filter. WUT supplies Microwave Office model of sample filter that is informed by package parasitic values for a standard range of passive components. 
- Pro: End-user gets the filter they want. 
- Con: In practice filters need to be installed by PCB stuffing house (pick-and-place tool). 
- Con: longer lead time to try different filter. 
- Con: uses more space on PCB. 

# Filter Requests
Here's my stab at filter specifications based on today's conversation. 

Filter Name |Proponent | Performance 
-------------|-------------|---------------
allaki_dac_filter_A | @jmizrahi  | pass band 175M to 225M (<3 dB insertion loss, <3 dB ripple, <1.5 VSWR), stop band <150M and >250M (> 20 dB loss),  (compare with MiniCircuits RBP-204+)
allaki_dac_filter_B | @hartytp, @cjbe  | pass band 2800M to 3600M (< 3 dB insertion loss, <3 dB ripple, <1.5 VSWR), stop at 2400M > 30 dB (compare with MiniCircuits BFCN-3085A+)
allaki_dac_filter_C | @dtcallcock, @dhslichter | One of these days ... one of these days ... Pow! Right in the kisser!

hartytp, 2016-12-06:
@jbqubit 

> Con: In practice filters need to be installed by PCB stuffing house (pick-and-place tool).

For small numbers of PCBs/prototyping, hand-soldering 0402 components is absolutely fine. 

FWIW, we'll need a few other configurations on top of the one you've listed there (e.g. 200MHz for AOMs).

jmizrahi, 2016-12-06:
@jbqubit I never advocated for the filter you describe (it wouldn't work for my application). The MiniCircuits part definitely would not work. Before we had talked about a 3rd order Butterworth low pass with 300 MHz corner. That has not changed.
I see no reason not to make the filter on the board from 0402 components. Regarding your concern about users replacing the filter, I would say (a) I don't think hand soldering a few 0402 components is that big a deal, and (b) this is not something I really anticipate doing anyway.

jbqubit, 2016-12-06:
If we take the non-MiniCircuits route @sbouhabib will design, say, two filter variations. I've updated allaki_dac_filter_A as @jmizrahi requested. @hartytp tweaks to allaki_dac_filter_B?

gkasprow, 2016-12-07:
Why cannot we use both versions at the time?

On can use Minicircuit filter pads as 0402 component pads and build discrete filter on the side of Minicircuit one.

jbqubit, 2016-12-07:
>Why cannot we use both versions at the time?
If there's room for it and it doesn't degrade performance that's fine by
me.

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

jbqubit, 2017-01-26:
@sbouhabib 
- [ ] post simulation of filters in table above
- [ ] add to repository filter designs... you're doing this in Microwave Office, right?  
Please confirm that there's space on the PCB for this approach.

sbouhabib, 2017-03-01:
[image]

sbouhabib, 2017-03-01:
[image]

sbouhabib, 2017-03-01:
Dac filter C
[image]

sbouhabib, 2017-03-01:
In general both approaches are in the schematics and PCB
@hartytp the simulation for FIlter_B is still "theoretical"  will you go for the BFCN-3085A+ or the other approach? or both :)

hartytp, 2017-03-01:
@sbouhabib Thanks for doing this.

We will most likely choose to go for a discreet filter, with specifications still tbd.

For the prototype round, we'd like our hardware to be shipped without filters (i.e. components DNP/0R as appropriate to provide a direct RF pass through). We'll design our filters ourselves later, after giving it a bit more thought.

IMO, since this filter is specific to our work, it shouldn't be on the schematic (we'll post them on the Wiki if others are interested). I assume the same is true for NIST/filter_C?

To aid us in filter design, it'd be great if you could provide an AWR or similar project, which is setup with relevant board/component parasitics (choose a sensible series of LC components).

sbouhabib, 2017-03-01:
the discrete filter schematics has no values just 0R or DNPs i.e. if it is done according to the standard BOM then it is a just a feed through with no filtering.

ill prep the general AWR project

- [ ] DAC Filter Reference Design

hartytp, 2017-03-01:
Great, thanks.

sbouhabib, 2017-04-01:
Uploaded "theoretical" reference design, for substituting with ones own components, We could prepare one which also incorporates the PCB but maybe after the prototype is checked.

closed 2017-04-01 by sbouhabib