sinara-hw/sinara #112: BP_RTM/PCB_baikal schematic to git

jbqubit, 2016-12-07:
This schematic codifies several discussions including #78.

jbqubit, 2016-12-08:
@sbouhabib Please generate rough schematic for tomorrow's discussion and post here. A hand-drawn schematic should suffice.

sbouhabib, 2016-12-09:
[Baikal_rough_sch.pdf](https://github.com/m-labs/sinara/files/642357/Baikal_rough_sch.pdf)

jbqubit, 2016-12-09:
From today's conversation.
- [ ] use differential signaling for routing of clock from SMA inputs and outputs to/from 954.
- [ ] independent RF shields with removable covers for low- high-frequency distribution chains
- [ ] additional SMA output on PCB connected to 12th output of 954 for 100 MHz distribution; use is routing of 100 MHz clock to Metlino; see also #122; need to maintain clearance for 90-degree SMA 
- [ ] use nice Crystek 100 MHz TCXO as on-board reference ($20 budget)
- [ ] remove MMC
    - no use case at present for MMC on this PCB
    - [ ] use DIP switches to reconfigure clock distribution of 954's
- [ ] do we really need LTC chip for sin to square conversion on Sinara family @hartytp will make independent issue on this question
- [ ] ferite bead to decouple power

hartytp, 2016-12-10:
IIRC, for the RTMs we plan to feedback on the fan speed for each slot to provide active temperature stabilisation. Do we also plan to do this for the eRTM15? This might be a useful feature for minimising drifts in the clock phases.

jbqubit, 2016-12-12:
IIRC there are two fan speed servo's specified in uTCA.4. One common to all
AMC boards and one common to all RTM boards. Slot-to-slot air flow is
varied using static baffles.

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

hartytp, 2016-12-12:
@jbqubit thanks for clarifying that.

hartytp, 2017-01-27:
@sbouhabib AFAICT, the grounds for the SMA connectors on Baikal are currently entirely floating wrt circuit ground. Are we worried about that from an ESD point of view? IIRC, on Sayma we added high resistance (100k?) pulldowns between the SMA and circuit grounds to avoid this...

jbqubit, 2017-02-13:
@sbouhabib When will this be ready for review? Please update [status](https://github.com/m-labs/sinara/wiki/status).

hartytp, 2017-04-16:
@sbouhabib ping

hartytp, 2017-08-02:
@sbouhabib Thanks for updating the designs! That all looks good to me.

jbqubit, 2017-08-03:
@sbouhabib Indeed! Thank you. :)

hartytp, 2017-08-10:
Since this has been open for a while without anyone raising any issues with the current schematic, I'm assuming that all interested parties have looked over the schematic and signed off on it. Please re-opn if that's not the case.

closed 2017-08-10 by hartytp