sinara-hw/sinara #113: Compatibility with MicroTCA.4.1

WeiDaZhang, 2016-12-08:
It has been said on the "5th MicroTCA workshop for industry and research" currently held in DESY, Hamburg that "MicroTCA.4.1" has been adopted and will be published before this X'mas.
There is some "±VV for µRTMs adjustable 5-12V" feature mentioned here on the conference.
I'm just wondering, does our design compatible with the 4.1?

gkasprow, 2016-12-08:
I didn't look at this yet.
Add it as an issue.
My colleagues will attend the workshop so will get some materials about 4.1.

>
>

WeiDaZhang, 2016-12-08:
I'm at the workshop now.
Please let me know if something I can try to get for you.

jordens, 2016-12-08:
@WeiDaZhang good that you are going there. Feel free to talk about ARTIQ and Sinara and to increase the awareness/publicity.

One would hope that µTCA 4.1 is at least backwards compatible.

WeiDaZhang, 2016-12-08:
I'm afraid I heard the workshop too late to add any proceeding. But I'll try my test, and I've been explaining quantum computing to these accelerator people so far.

vdirksen, 2017-02-06:
Hi, MTCA.4.1 is backward compatible to MTCA.4. It adds the mechanical and electrical specification for a second optional µRTM backplane behind the AMC backplane, for the MCH-RTM including the µRTM backplane management extension, for a special Rear Powermodule for the µRTM backplane behind the

vdirksen, 2017-02-06:
AMC backplane. One µRTM backplane implementation is the Low Level RF backplane from Desy. This requested variable bipolar voltages. This is addressed by the Rear Power Module. Let me know, if this answers your questions.

hartytp, 2017-02-06:
@vdirksen Thanks for the clarification.

Closing this issue, as it seems resolved.

closed 2017-02-06 by hartytp