sinara-hw/sinara #115: layout review for Sayma and Metlino

jbqubit, 2016-12-08:
This review is slated for Friday 12/9. Some questions for @gkasprow in preparation for the review. By email you indicated that this review will be for Sayma_AMC, Sayma_RTM and Metlino_AMC.

- [ ] Which files should reviewers look at when evaluating layout?
- [ ] Can you generate 3D PDFs?
- [ ] How many layers are you planning for PCB? What is layer utilization? 
- [ ] Please show placement of RF shields. See discussion in context of analog mezzanine #8 for some thoughts (especially Oct 4).

gkasprow, 2016-12-09:
Schematic files: https://github.com/m-labs/sinara/tree/master/ARTIQ_EE - pdf and viewer (cce)
3D, step and viewer files:
https://github.com/m-labs/sinara/tree/master/ARTIQ_EE/PCB_Metlino/Output
https://github.com/m-labs/sinara/tree/master/ARTIQ_EE/PCB_Sayma_AMC/Output
https://github.com/m-labs/sinara/tree/master/ARTIQ_EE/PCB_Sayma_RTM/Output

I use 12 layers for Sayma AMC and Metlino.
L2, L4, L6,L9,L11 are ground planes
L1, L12 - signal layers, some diff lines that need low attenuation, mainly polygons and short traces (EMI mitigation)
L2,L5, L10 - high speed lines like DDR, LVDS
L7 - power plane
L8 - slow and not critical lines like I2C, SPI, LEDs

I use 8 layers for Sayma RTM
L2, L4, L7 - ground planes
L1, L8 - polygons, short traces and analog signals between ADC/DAC and connectors
L3 - high speed signals to mezzanines
L5 - power planes
L6 - non-critical signals like I2C, status, LED, etc

jordens, 2016-12-09:
@gkasprow you need to be much more careful with what you check into the repository. We are at gigabytes again. Remember that you can not delete files without feeling and causing lots of pain. And remember that there are [repository size limits](https://help.github.com/articles/what-is-my-disk-quota/) for good reason. I bet you could have reduced the repository size by a factor of ten.

gkasprow, 2016-12-09:
I tried but without these files i cannot open the design. I.e. work folder
is necessary. I didnt commit the 3D files which consume additional gigabytes

<https://github.com/m-labs/sinara/issues/115#issuecomment-266003674>, or mute
the thread

jordens, 2016-12-09:
What about these and similar:

```
42M     ARTIQ_EE/PCB_Metlino/Work/Layout_Temp/
38M     ARTIQ_EE/PCB_Sayma_AMC/Work/Layout_Temp/
49M     ARTIQ_EE/PCB_Sayma_RTM/Work/Layout_Temp/
```

And I see more than 200 MB of `.step` and 3D `.pdf`s. I am pretty sure generating those is a manual process for you and would suggest that you don't commit these generated files to git (same for the schematics pdf, the cce files, the gerbers, i.e. everything that is "generated") but that you do the following:

* commit (in git) the files you actually edit
* tag (in git) that commit with some systematic naming scheme
* make a release from that tag (on github)
* upload your generated files to that release

I did that for 0.1rc1, the first review (rc1) of the first prototype (0.1): https://github.com/m-labs/sinara/releases/tag/0.1rc1

gkasprow, 2016-12-09:
Joe wanted to have 3d PDF so I uloaded them.

I tried to remove Layout_Temp but the Expedition crashes.

I can put releases of project archives.

gkasprow, 2016-12-09:
OK, I got what you mean.

I will remove them from repo and place as release.

jbqubit, 2016-12-09:
@jordens Thank you for looking out after the repository size. However, please don't pollute this Issue with discussion of orthogonal issues (repository size). Create a new Issue for such matters.

jbqubit, 2016-12-09:
Following is easier for me to visualize.

# Metlino and Sayma_AMC Layers 
Layer | Function
----------|----------
L1 | signal layers, some diff lines that need low attenuation, mainly polygons and short traces (EMI mitigation)
L2 | ??
L3 | high speed signals to mezzanines
L4 | GND
L5| high speed lines like DDR, LVDS
L6 | ?
L7 | power plane
L8 | slow and not critical lines like I2C, SPI, LEDs
L9 | GND
L10 | high speed lines like DDR, LVDS
L11 | GND
L12 | signal layers, some diff lines that need low attenuation, mainly polygons and short traces (EMI mitigation)

- [ ] @gkasprow  Do you intend to specify L2 for both GND and " high speed lines like DDR, LVDS"? 
- [ ] @gkasprow Do you intend to specify L6 for both GND and "non-critical signals like I2C, status, LED, etc"

# Sayma_RTM Layers 
Layer | Function
----------|----------
L1 | polygons, short traces and analog signals between ADC/DAC and connectors
L2 | GND
L3 | high speed signals to mezzanines
L4 | GND
L5 | power
L6 | non-critical signals like I2C, status, LED, etc
L7 | GND
L8 | polygons, short traces and analog signals between ADC/DAC and connectors

jbqubit, 2016-12-09:
From today's hangout.

# air flow
Orient components to maximize air flow across PCB in vertical direction.
- [ ] clock mezzanine power/signal header rotate 90 deg
- [ ] cock mezzanine SMA's group SMAs into two vertical columns
- [ ] FPGA interface header, rotate 90 degrees and put in line with other components
- [ ] SATA headers, rotate 90 degrees and put in line with other components
- [ ] thermal loading test #124

# layout information  
- [ ] silkscreen labels on components
- [ ] silkscreen label on PCB -- create new Issue to discuss #123 

# other topics
- [ ] need better clarity on what parts to order for test systems #21

jbqubit, 2017-02-13:
@gkasprow Please post on this Issue when you'd like for the community to take a look at the layout of Sayma AMC and Sayma RTM. Based on [status](https://github.com/m-labs/sinara/wiki/status) looks like Metlino needs schematic review; please create a new Issue. Please see check boxes for Dec 9 outstanding question about layer usage.

gkasprow, 2017-02-13:
updated layer assignment. For RTM I had to use 10 layer board due to dense packing of components.

closed 2017-04-16 by hartytp