sinara-hw/sinara #116: Sayma/Metlino layout, clocking plan

jbqubit, 2016-12-08:
Sinara clocking plan is discussed on the wiki. https://github.com/m-labs/sinara/wiki/SinaraClocking

@jordens @sbourdeauducq Are you satisfied that Metlino and Sayma schematics are compatible with the Sinara clocking plan?

jbqubit, 2016-12-08:
Depends on #112.

jordens, 2016-12-09:
I checked feec424
Metlino MCH, looks good to me. Still going through Sayma AMC and RTM.

@sbourdeauducq could you verify the source of REC_CLOCK_P/N on Metlino and Sayma?

sbourdeauducq, 2016-12-09:
REC_CLOCK (from FPGA to Si chip) can be placed anywhere.

jordens, 2016-12-12:
Sayma_AMC and Sayma_RTM also LGTM.

jbqubit, 2016-12-12:
Thank you Robert.

jbqubit, 2016-12-12:
- [ ] Are you satisfied with the layout of baikal #112 with respect to the clocking plan?

jordens, 2016-12-14:
W.r.t. to the clocking plan, Baikal distributes the 100 MHz clock to the RF backplane and Metlino-AMC. AFAICT from Baikal_rough_sch.pdf it achieves that.

closed 2016-12-14 by jordens

jbqubit, 2016-12-14:
@jordens @sbourdeauducq Thank you for taking a close look.