sinara-hw/sinara #118: baikal test: RFB clock reflection due to unoccupied RTM slots

jbqubit, 2016-12-08:
First discussed in #78. 

Paraphrasing @sbouhabib
>one thing just occurred to me, the ADCLK954 has no option to turn outputs on or off, which means ,that either all outputs are on or all are off, if the whole crate is not populated, the rest are left floating which is not so nice. There may be reflections, mismatches, worse performance of ADCLK in general, higher cross-talk, unnecessary additional "noise" on the RF-backplane. 

@hartytp 
> When we have the prototype hardware, we can test whether this is actually a problem (no point complicating the circuit to fix this if it's not a problem). If the reflections do cause problems, my preferred solution would be to add an ADCLK925 (or similar) as an output buffer for each channel after the fanout buffers. These output buffers should provide sufficient isolation between the outputs.

sbourdeauducq, 2016-12-10:
@sbouhabib Why would the reflections bother the ADCLK954? We can also add termination resistors in unpopulated slots (no PCB respin, less configuration, conceptually simpler).

gkasprow, 2016-12-10:
As I said, we can use small MOSFETs with capacitor in parallel.
If you disconnect the pulldown resistor, no signal will be generated at the outuput.
Any idea why this solution would not work?

[image]

hartytp, 2016-12-10:
@gkasprow AFICT that would work fine. Would the MOSFETs be controlled from a microprocessor or using DIP switches? If we use a microprocessor, is it worth adding an RC filter on OUT_EN to ensure that noise doesn't couple onto the clock outputs?

jbqubit, 2016-12-12:
@sbourdeauducq asks "Why would the reflections bother the ADCLK954?"

The spec sheet doesn't detail output to output isolation. I think @gkasprow approach will work. Thought we agreed to use DIP switches for this type of configuration. Pads for termination resistors could be added as risk mitigation and to permit a differential test between termination vs MOSFET approach.

- [ ] add to schematic voltage supplied by OUT_EN

sbourdeauducq, 2016-12-12:
If it's DIP switches then no mosfets are needed.

sbourdeauducq, 2016-12-12:
But I'm skeptical we need any of this.

sbouhabib, 2016-12-21:
the RF backplane has pins for every slot called RF-conf which we can use as enables, this means that if a board is in the slot this output will be on using what @gkasprow suggested

sbouhabib, 2016-12-21:
the only thing left would be the 3 slots that get the CLK signals only, these cannot be controlled automatically, maybe we can leave them be, that would give us info on potential problems, while the "analog" signals throught the coax connectors are safe

jbqubit, 2016-12-21:
It's hard to talk about this without being able to see backplane schematics #135. It's new to me that RF-conf exist. I like the idea of using RF-conf by the way. :)

closed 2018-02-10 by hartytp