sinara-hw/sinara #119: LTC6957-1

hartytp, 2016-12-09:
Issue raised during design review conversation today: do we need/want to use LTC6957-1 for 100MHz sine-square conversion on Sayma RTM and eRTM15? 

We already use ADCLK9xx for mux/fanout, which should provide equal performance for the sine-square conversion when used with a differential input signal with decent power. Letting the ADCLK9xx do both jobs simplifies the design and removes an IC from the BOM. 

Any objections?

hartytp, 2016-12-12:
@gkasprow @sbouhabib Let's give people until the end of today to raise objections to this, and then make the change. This change will affect both Sayma RTM and the eRTM15.

jbqubit, 2016-12-12:
@dhslichter You may have an opinion on this.

jordens, 2016-12-12:
One advantage of the LTC6957-1 was isolation from unused but driven clocks. That does not seem to be relevant since they are either not used or can be suppressed at the source.
Otherwise the ADCLK9xx do seem significantly worse at squaring than the LTC6957-1: 300 fs RMS jitter vs. some 100 fs for a 100 MHz 10 dBm sine. But IMHO the ADCLK9xx datasheet also lacks important information, like a proper added phase noise plot making a comparison hard.

hartytp, 2016-12-12:
@jordens 

> One advantage of the LTC6957-1 was isolation from unused but driven clocks. That does not seem to be relevant since they are either not used or can be suppressed at the source.

True, the shut down input is a nice feature of the LTC6957-1. However, if we did need this, turning off the ADCLKxxx output bias using a FET (as suggested by @gkasprow) might be a nicer option, since fewer ICs in the clock chain is almost always better from a noise/drift perspective. 

> Otherwise the ADCLK9xx do seem significantly worse at squaring than the LTC6957-1: 300 fs RMS jitter vs. some 100 fs for a 100 MHz 10 dBm sine. But IMHO the ADCLK9xx datasheet also lacks important information, like a proper added phase noise plot making a comparison hard.

Agreed, the ADCLK9xx data sheet could be better. However, it does have an absolute phase noise plot, which allows one to put an upper-limit on the noise.

In any case, since we already have a fewADCLK9xx in our clock chain, the question isn't really how good its noise is compared with the LTC6957-1, but rather whether it's output noise is different for square waves and sinusoids at 100MHz. This isn't specified but I don't expect it to be a problem.

jordens, 2016-12-12:
It is different. That (output noise increase due to slow slew rate) is specified: [Figure 12](http://www.analog.com/media/en/technical-documentation/data-sheets/ADCLK948.pdf#page=9).

jordens, 2016-12-12:
I was more thinking of isolation from crosstalk due to driven but unused *inputs* on e.g. the ADCLK9xx

hartytp, 2016-12-12:
@jordens aah yes, I'd missed that figure. Well spotted! So,...

@100MHz, 1V/ns correspond to an input signal level of 1.125Vrms. For a 100Ohm signal, this corresponds to 11dBm. At 1V/ns, the chip's performance has degraded significantly, so we'd prefer the input to be a few times this. 5V/ns would be ideal, however, this corresponds to 25dBm, which is far outside the chips max operating range. 

Notice that any impedance mismatches/reflections can slow the rising edge down, making this worse. 

Given this, I'd suggest that we stick the using the LTC6957-1.

hartytp, 2017-12-01:
I'd like to reopen this thread for Sayma v2.0:

- As @jordens points out above, [this figure](http://www.analog.com/media/en/technical-documentation/data-sheets/ADCLK948.pdf#page=9) in the ADCLKxxx datasheet shows that these clock buffers do not quite hit their optimal phase noise performance when clocked at 100MHz from a 10dBm source. However, this *doesn't* show that they perform worse than the LTC6957-1 IC under these conditions. Thus, it doesn't establish that the LTC IC is actually helpful in Sayma.
- Weida took some data when testing our PLL mezzanine design which is relevant here. He took a 100MHz Wenzel oscillator then squared it up using one of these ICs and then use it as the reference for our ultra-low noise 2.4GHz PLL. Ignore all curves on this plot apart from the blue and red curves (other curves are more noisy due to a grounding issue introduced by the baluns). This shows no difference between the two ICs, suggesting that either is fine. More importantly, the data shows that squaring up a Wenzel oscillator with either of these two ICs provides phase noise performance that's far superior to the DAC's. As a result, either will do for this design.
- Currently, we have a LTC6597-1, followed by an ADCLK (the latter used for mux + fanout). Since we need the ADCLKxxx for mux/fanout, I would advocate removing the LTC6957-1 from Sayma 2. Simplifies things a bit, saves money and power and removes a potential source of noise + drify.

Any objections?

[image]

hartytp, 2017-12-04:
Also FWIW, the ADCLKxxx datasheet figure showing performance versus input rising edge rate is not necessarily relevant to our use case. This figure shows integrated jitter, which is really a measure of broad-band noise floor. Since we never plan to use the 100MHz directly, but rather as a reference for the PLL, the broad-band noise is less important than the close-in noise. It's less clear how the edge rate affects the close-in noise...

hartytp, 2017-12-07:
More data on this courtesy of @WeiDaZhang:

Phase noise of LTC and ADCLK ICs for sine-square conversion:

[image]

hartytp, 2017-12-07:
Comments:
- The LTC IC does give slightly better performance above a few kHz. Not sure if this difference disappears once the LTC output passes through the ADCLK used as a mux. The difference is more pronounced for lower power clock inputs (as expected).
- Noise performance of both ICs is better than the DAC's noise floow, so using the LTC IC will not provide a benefit on Sayma AFAICT.
- From the above data, both ICs are also sufficiently good that the difference between them is not enough to affect the performance of our clock mezzanine.

As a result, my recommendation is still to scrap the LTC ICs.

hartytp, 2017-12-07:
For those curious: 

[image]
AM spectra

hartytp, 2017-12-07:
And total noise power density:

[image]

hartytp, 2017-12-07:
- LTC ICs have worse AM noise. This is probably not an issue so long as following components have good AM to PM rejection.

gkasprow, 2018-08-10:
now we have such circuit:
[image]

closed 2018-08-10 by gkasprow

hartytp, 2018-08-11:
Hard to judge without seeing the rest of the circuit (what does that clock connect to), but looks fine as far as it goes.