sinara-hw/sinara #123: Sinara PCB silkscreen information

jbqubit, 2016-12-09:
What information should appear on PCB silkscreen on the Sinara components? @gkasprow @jordens @sbourdeauducq 
- copyright symbol and year; who holds copyright #16?
- link to ohwr.org
- ARTIQ logo
- WUT logo
- revision number

sbourdeauducq, 2016-12-10:
ARTIQ logo only, no other links or logos.

gkasprow, 2016-12-10:
We should have some board name, revision number which is somehow related with hardware version number (4 bits)

jordens, 2016-12-11:
* Name of the PCB/project: Sayma
* Version/revision (the git tag): 0.1
* license, copyright, year: Copyright 2016 WUT/M-Labs
* ARTIQ name and logo
* Manufacturer name or logo
* Serial number (or at least a batch number)
* Some flat white silkscreen area (7 mm by 20 mm) to write things onto (ID, configuration, inventory barcode)
* No link to ohwr.org

jbqubit, 2016-12-12:
CERN Open Hardware Licence v1.2

Serial numbers ought to be unique. Is somebody going to maintain a database of sequential ARTIQ hardware serial numbers?

jordens, 2016-12-12:
Why would you need a central database to make them unique?

jbqubit, 2016-12-12:
I'm pointing out that somebody needs to be responsible for assigning serials numbers. @gkasprow does WUT want to do this?

gkasprow, 2016-12-12:
Each board has a chip with unique number - MAC address. We can keep online
simple spreadsheet with SN and every user can write his board story. This
helps a lot to identify and manage issues.

sbourdeauducq, 2016-12-13:
We can simply put this on the Sinara GitHub wiki.

sbourdeauducq, 2016-12-13:
> Each board has a chip with unique number - MAC address.

What chip is that?

gkasprow, 2016-12-13:
AT24MAC402-XHM

24AA02E48T-I/OT

closed 2017-01-26 by jbqubit