sinara-hw/sinara #124: test of Sayma_RTM thermal loading

jbqubit, 2016-12-09:
WUT thermal analysis will be per-PCB not for full system. PCB_mezzanine_clock_template and PCB_mezzanine_analog_template will be populated with heating elements to permit testing of system performance under thermal load.
- resistors
- configure resistors using MOSFET switches toggled by I2C under MMC control

gkasprow, 2016-12-11:
Added dip switch and resistors.
For each power branch, 50% and 100% load can be selected.
MOSFETs are controlled by additional I2C extender. Manual operation is also possible.

jbqubit, 2016-12-12:
Thank you for making the modifications @gkasprow. 
- [ ] AFE_mezzanine_3D.pdf shows misplaced meander under SMA connectors. 

This is assigned to the "0.1 testing" milestone which will take place in Jan. Reopening.

gkasprow, 2016-12-12:
@jbqubit what do you mean by "meander" ?

jbqubit, 2016-12-12:
Grey bit under the SMAs in the following screen shot of  AFE_mezzanine_3D.pdf. 

[image]

gkasprow, 2016-12-12:
This is a silkscreen related with footprint.
It will be removed by PCB manufacturer.

jbqubit, 2016-12-13:
I see now. :/

-------
Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

closed 2017-02-16 by jbqubit