sinara-hw/sinara #125: PCB_mezzanine_clock_template schematic review

jbqubit, 2016-12-09:
- [ ] add components for thermal testing #124

gkasprow, 2016-12-12:
done, added same components as on AFE

jbqubit, 2016-12-12:
- [ ] rename ARTIQ_ALTIUM/PCB_mezzanine_clock to ARTIQ_ALTIUM/PCB_mezzanine_clock_template to distinguish from forthcoming Oxford board
- [ ] update based on #115 
- [ ] generate PDF of schematic

gkasprow, 2016-12-12:
Schematic
[PCB_mezzanine_clock_template.pdf](https://github.com/m-labs/sinara/files/647291/PCB_mezzanine_clock_template.pdf)

jbqubit, 2017-01-31:
Move discussion of schematic and layout from [here](https://github.com/m-labs/sinara/issues/99#issuecomment-276042586) back to this Issue. I probably closed this prematurely.

- add RF shield for thermal and geometric testing
- SMA taps for the RF would be helpful for future diagnostics if there's room

jbqubit, 2017-01-31:
- since this is the template for other analog mezzanine boards please label expected interface
    - voltage, current, bandwidth... copy from PCB_Sayma_RTM
- labels on PCB giving names to SMP's that match PCB_Sayma_RTM

hartytp, 2017-02-08:
Closing this issue as the boards have been sent for manufacture (create a new issue for comments on the manufactured boards).

closed 2017-02-08 by hartytp