sinara-hw/sinara #126: RF shielding of Sayma_RTM

jbqubit, 2016-12-09:
@gkasprow Please reiterate options we discussed in today's call. 

- [ ] RF shields for clock distribution components
    - use removable lids 
- [ ] what else should be shielded
- [ ] board-board isolation options

gkasprow, 2016-12-11:
Added 3 Wurth 30x30mm boxes with removable lid. All clocking circuits  are covered
[image]

hartytp, 2016-12-12:
@gkasprow Looks good, so long as it won't cause a problem with thermal performance.

jbqubit, 2016-12-12:
On Friday we also discussed metal shield fully covering the back side of each Sayma_RTM. This would provide board to board isolation. The value of this could be tested in 0.1 by comparing the performance of a rack with adjacent Sayma_RTM vs a rack with metal spacers interleaved between the Sayma_RTM. Is NAMC-Filler FS-RP1D made of metal? Can it serve as a good RF shield for tests?

gkasprow, 2016-12-12:
Piece of steel that has no connection to ground is poor shield.
So even if you manage to insert NAMC-Filler FS-RP1D (without panel) between AMCs, it won't help much in RF shielding.
I saw somewhere (I have to ask  Pentair) complete RF shields for RTMs. And IMHO only such solutions would help.

gkasprow, 2016-12-13:
I found covers from Schroff:
http://schroff.pentair.com/en/schroff/MicroTCA-/Module-cover-for-MTCA-4-front-and-RTM-modules
These are just mechanical protection cases, not RF cages.

gkasprow, 2016-12-20:
I got mechanical files from Schroff.
Such case will fit, but it requires some changes to the layout
Look here:
[Sayma_RTM_shield.pdf](https://github.com/m-labs/sinara/files/664944/Sayma_RTM_shield.pdf)
In this 3D view, both solution with cage and individual shields are shown so they intersect in some places
Similar cage exists for AMC side.

jbqubit, 2017-02-13:
> 3 Wurth 30x30mm boxes with removable lid. 

The Wurth boxes sound fine.  Closing Issue. Re-open if they pose thermal problems.

closed 2017-02-13 by jbqubit