sinara-hw/sinara #130: Simple Sayma AMC DIO FMC

dtcallcock, 2016-12-12:
In #99 I asked about digital IO for Sayma AMC.  It was pointed out that the [FMC DIO 32ch LVDS a](http://www.ohwr.org/projects/fmc-dio-32chlvdsa/wiki/Wiki) board has the same pinout as Metlino (VHDCI connector) so will work out with the Metlino breakout hardware being proposed.

I can imagine quite a few use cases though where one just wants one or two DIO (offboard rf switch, PID hold, cavity lock level detector etc...) and that solution would be overkill.  Looking through OHWR I spotted this simple 5-ch DIO card: [FMC DIO 5CH TTL A](http://www.ohwr.org/projects/fmc-dio-5chttla/wiki).  Would this be suitable and if so what is the approx cost of this board @gkasprow?  If not do you know of another already available FMC or would we just have to design one?

jbqubit, 2016-12-12:
@gkasprow Is there any room on Sayma_AMC front panel for a couple SMA connected to FGPA IO? Same 50 Ohm driver-receiver design as discussed in #99. I think this would satisfy @dtcallcock need.

gkasprow, 2016-12-12:
I will ask my colleague to provide you a quote. It is not very cheap AFAIR.

gkasprow, 2016-12-12:
@jbqubit  @dtcallcock 
We can easily fit 2 SMA connectors on Sayma AMC

dtcallcock, 2016-12-12:
Is there a downside to this apart from adding a few $ to the board?  I am pro.

gkasprow, 2016-12-12:
OK, I added this feature, but we should finally freeze the schematics otherwise won't be able to proceed with PCB routing.

closed 2016-12-12 by gkasprow