sinara-hw/sinara #131: expected timeline 0.1 -> 0.2 -> 1.0

jbqubit, 2016-12-15:
In #97 @hartytp  asks

> Assuming we don't find any major mistakes in the prototype hardware, when do you think a second hardware order will take place? This will affect how much prototype hardware we buy.

- [ ] @gkasprow Please predict date when the BOM and layout will be frozen.

- [ ] @sbourdeauducq Please comment on status of ARTIQ support for Sinara hardware.

# v0.1 
From date order is placed with board house... 
    + 2 weeks for printing, stuffing (assuming 100% BOM availability)
    + 1 week for WUT PCB checks using existing test tools (no ARTIQ dependence)
    + 1 week for international shipping
    + 3 weeks for testing, thrashing, bug busting, discussions by end users and M-Labs (@jordens visits JQI for a week)
    + 2 weeks schematic update, layout fixes, repeat SI analysis 

# if v0.2
The need for a second prototype round will be decided for each PCB. From date order is placed with board house... 
    + 2 weeks for printing, stuffing 
    + 1 week for WUT PCB checks
    + 1 week for international shipping
    + 2 weeks for testing, thrashing, bug busting, discussions
    + 1 weeks schematic update, layout fixes, repeat SI analysis

# v1.0 
Design locked in... purchases are placed with board house (eg technosystem.pl)
    + 4 weeks slack for procurement spool-up by prospective customers
    + 2 weeks for printing, stuffing
    + 1 week for WUT PCB checks
    + 1 week for international shipping

If v0.1 is frozen 1/1/2017 and you believe this timeline...
    - one prototype round (v1.1) on the way to delivery of v1.0 4/7/2017
    - two prototype rounds (v0.1 -> v0.2) on the way to delivery of v1.0  6/1/2017

- [ ] @gkasprow Please comment on how realistic this progression is in your experience. 

# deprecate prototype hardware?
Depending on the type and severity of errors in prototypes it may be expedient to deprecate v0.1/v0.2 once v1.0 becomes available. If the deltas are just loop-filter components and phase-noise motivated layout tweaks I guess extended support of prototype hardware by M-Labs isn't onerous. If substantial monkeying is needed to get the prototype hardware to function at all (eg lots of flying leads, hacks to compensate for cross talk induced bit errors) deprecation may be in order. But this is speculation at this point.

sbourdeauducq, 2016-12-16:
> + 1 week for WUT PCB checks using existing test tools (no ARTIQ dependence)

@gkasprow what is your plan for the test suite?
There is also the option of keeping this initial WUT testing to a minimum (power supplies OK, check that the FPGA answers to JTAG) for v0.1, until a proper test suite is developed and used for v0.2+.

gkasprow, 2016-12-16:
@sbourdeauducq It is good idea. Wojciech is still occupied with another project so at the moment he cannot fully switch to test suite development. The plan was to at least check the SDRAM and transceivers and we should manage to do so. We can leave full connectivity checking for v0.2 when complete test suite is ready.

gkasprow, 2016-12-16:
To mitigate the risk, we can assemble only 2 pieces of Metlino/Sayma AMC to check if no critical issues are present and within a few days assemble rest of them. Initial batch count was just 2 pieces but now it grown to higher volume.
In case of serious PCB bugs we won't loose precious components but only PCBs.

sbourdeauducq, 2016-12-16:
Right now there is mostly no ARTIQ support specifically for the new hardware. We have been busy with the PDQ, SAWG and DRTIO, and https://github.com/m-labs/artiq/issues/637 has been a frustrating time sink.

The major steps for something interesting (stand-alone Sayma) are:
* Kintex Ultrascale support in OpenOCD, done using KCU105.
* GTH port of the JESD204 core, @enjoy-digital will be working on it, can be done with the KCU105.
* DDR3, not started, needs hardware to test as KCU105 uses DDR4.
* support for the RTM FPGA, not started, preferably needs hardware but KCU105 can be quite easily hooked to KC705 for testing parts of it.

jbqubit, 2016-12-16:
> To mitigate the risk, we can assemble only 2 pieces of Metlino/Sayma AMC to check if no critical issues are present

I like this approach. 

>We can leave full connectivity checking for v0.2 when complete test suite is ready.

We shouldn't migrate from v0.1 until all component connectivity is confirmed. Exception is if there's a serious bug in FGPA, SDRAM, clocking or transceivers that precludes such a check. 

>Right now there is mostly no ARTIQ support specifically for the new hardware. We have been busy with the PDQ, SAWG and DRTIO, and m-labs/artiq#637 has been a frustrating time sink.

> Wojciech is still occupied with another project so at the moment he cannot fully switch to test suite development.
> Right now there is mostly no ARTIQ support specifically for the new hardware.

Both these answers are helpful in knowing the current state of development. More is needed to answer @hartytp question. @sbourdeauducq @gkasprow Please prognosticate about the timeline for development of tools to test 
- [ ] connectivity of all PCB components 
- [ ] data integrity at full data rate
- [ ] operation of all PCB components sufficient to permit start of analog testing

gkasprow, 2016-12-16:
Production test suite and verification of connectivity are two different
things.
We can check connectivity manually, line by line and in the meantime
develop the production test suite.

jbqubit, 2016-12-16:
From today's teleconference.

@gkasprow Confirms that BOMs are frozen for all the Sinara components slated for delivery with v0.1 #97. Does this mean 

@gkasprow says to adjust timeline as follows to be conservative
* printing, stuffing (assuming 100% BOM availability): 2 weeks -> 3 weeks
* WUT PCB checks using existing test tools: 1 week -> 2 weeks

Consensus is that the following is too vague; specific tests must be defined. "3 weeks for testing, thrashing, bug busting, discussions by end users and M-Labs". Discussion goes on here #117. 

M-Labs points out that most analog tests won't be practical until @sbourdeauducq "major steps for something interesting" are accomplished. @jordens predicts 6-8 weeks from today.  

 @jordens can just as well write gateware in Germany. Delay visit to JQI until analog tests are underway. 

----------------
Updated timeline in view of today's discussion. 

# v0.1 
Assume PCB order is placed with board house 1/1/2017. 
    + 3 weeks for printing, stuffing (assuming 100% BOM availability)
    + 2 weeks for WUT PCB checks using existing test tools (no ARTIQ dependence)
        - power electronics test, DDR test, GTX transceiver test
    + 1 week for international shipping
    + 3-6 weeks for testing, thrashing, bug busting, discussions #117
        - @jordens visits JQI near the end of this interval 
        - testing depends on M-Labs' "major steps for something interesting"; available between 2/1 and 2/15
    + 2 weeks schematic update, layout fixes 

# if v0.2
The need for a second prototype round will be decided for each PCB. From date order is placed with board house... 
    + 3 weeks for printing, stuffing 
    + 1 week for WUT PCB checks (now they're rehersed)
    + 1 week for international shipping
    + 2 weeks for testing, thrashing, bug busting, discussions
    + 1 weeks schematic update, layout fixes

# v1.0 
Design locked in... purchases are placed with board house (eg technosystem.pl)
    + 4 weeks slack for procurement spool-up by prospective customers
    + 3 weeks for printing, stuffing
    + 1 week for WUT PCB checks (now routine)
    + 1 week for international shipping

jordens, 2016-12-18:
My best guess was 3-10 weeks after the tests at WUT have completed.

jmizrahi, 2017-01-10:
Can I please get an update on where the timeline stands? Has the PCB order been placed? Are these PCBs being manufactured?

gkasprow, 2017-01-10:
Hi
The Sayma-AMC board routing is finished and the layout is being verified by
Michal. Please have a look at it.
After Michal runs the SI and PI analyses I will  do the cosmetics, run net
equalisation and generate production files.
Frankly speaking I underestimated amount of work with Sayma board - it took
much longer because at some point I had to reassign almost all FPGA
connections and start routing from the beginning. Now it fits nicely on 12
layers. It took one week longer than I planned. Another problem was with
Mentor license server at the University which was closed during Xmass
period.
The China will have its New Year and all factories are closed  for 2 weeks
so we will receive the boards at the beginning of February. German company,
ILFA has 8+ week delivery time.
In case of front panels the delivery of raw bezels is 4 weeks. They say
that can make custom cutouts in 5-6 weeks.
The Sayma-RTM routing is in progress. I estimate about 10 days to have it
done.
In case of Metlino I propose to make production run after we make Sayma
running. 90% of the board is identical to Sayma and once we make Sayma AMC
running, the Metlino will work as well.
But it is up to you how we proceed.
So far all critical components (FPGAs, connectors, SFP cables) arrived.

jbqubit, 2017-01-11:
>The Sayma-AMC board routing is finished

Excellent! 

>After Michal runs the SI and PI analyses I will  do the cosmetics, run net
>equalisation and generate production files.

How long is that expected to take? 

> China will have its New Year and all factories are closed  for 2 weeks
>so we will receive the boards at the beginning of February. German company,
>ILFA has 8+ week delivery time.

I thought technosystem.pl was slated to do the PCB printing and stuffing. Right? What boards are coming from China and ILFA? 

>The Sayma-RTM routing is in progress. I estimate about 10 days to have it
>done.

OK. 

> In case of Metlino I propose to make production run after we make Sayma
> running.

Let's decide on printing of Metlino once the routing, SI and PI analyses and net equalization is done. 

> So far all critical components (FPGAs, connectors, SFP cables) arrived.

That's great news. :)

gkasprow, 2017-01-11:
@jbqubit 
Technosystem is assembly house.
PCB printing is very complex process and only a few companies are able to cope with such technology.
So we subcontract it to NCAB, ILFA, Brandner and many other companies.
We should finish PCB verification in one week.

jmizrahi, 2017-01-11:
@gkasprow I don't understand, every PCB fab house I know of can print 12 layers (or much more). Is there something driving the complexity beyond the the layer count?
Can you tell me what you think the bottom line timeline is for the first prototype to be in-hand?

gkasprow, 2017-01-11:
We use special RF material capable of 12 GHz operation. So only a few
factories can do it.
Anyway, the boards were designed for certain materials so we would have to
keep the stuckup.
So let's assume we get the boards at the beginning of February. We will
need 3..4 weeks to make them running. There is a chance that we can have it
quicker but let's be pesimistic.
We will make 2 of them running, in the meantime we will assemble remaining
boards. We will need to test them but one week should be fine.
So realistic date is middle of March.

jmizrahi, 2017-01-11:
OK, thanks for the timeline.
What is the substrate?
The 3-4 weeks to make them running includes the time to populate the components?

gkasprow, 2017-01-11:
The substrate is ISOLA FR408HR
The component assembly takes about 2-3 days and Technosystem does it.
It requires P&P programme, SMT stensil, SMT assembly, THT selective wave soldering, automatic optical inspection.
Then the boards are delivered to WUT where we verify all functionalities like power supply, mezzanines insertion, ripple measurement and optimisation, all clock circuits, SDRAM operation, GTH performance, mechanics, etc.
It really takes a while and there are always some issues to solve.

gkasprow, 2017-01-22:
Sayma RTM routing is 80% done. It is really tough, I have to do most things manually due to board packing. Sayma AMC was trivial comparing with RTM. But I will manage it.
[image]

jbqubit, 2017-01-26:
@gkasprow @sbouhabib What's status of allaki? Please update the table below so we're all on the same page. Add columns 

# Dates when step is complete

Item | Schematic | Layout | PA | SIA | Net Eq | Sent to Production | PCBs at WUT | WUT Tests | Ship
------|-----------------|----------|------|------|-----------|--------------------------|---------------------|----------------|------
PCB_metlino | 1/1/2017 | | | | | | |
PCB_MCH_mordovia | 1/1/2017 | 1/11 | | | | | |
PCB_sayma_AMC | 1/1/2017 | ?  | | | | | |
PCB_sayma_RTM | 1/1/2017 | | | | | | |
PCB_mezzanine_analog_allaki | waiting | | | | | | |
PCB_mezzanine_clock_template | 1/1/2017 | 1/11 | | | | | |
PCB_mezzanine_analog_template | 1/1/2017 | 1/11 | | | | | |
PCB_baikal | 1/1/2017 | 1/11 | | | | | |

- __PA__ is Power Analysis
- __SIA__ is Signal Integrity Analysis
- __WUT Tests__ is WUT PCB checks using existing test tools (no ARTIQ dependence)
    - power electronics test
    - DDR test
    - GTX transceiver test
- __Ship__ shipped to physicists and m-labs

jbqubit, 2017-01-26:
Similar tracking for m-labs....
- support kintex ultrascale [#563](https://github.com/m-labs/artiq/issues/563)
     - [x] OpenOCD support for Kintex Ultrascale
     - [x]  Dual flash support
     - [ ]  JESD204 for Kintex Ultrascale (GTH)
     - [ ]  JESD204B synchronization and group-delay/latency alignment
     - [ ]  port of DRTIO to KintexUltrascale #562
     - [ ]  Kintex Ultrascale DDR3
     - [ ]  High-resolution IOSERDES TTLs for Kintex Ultrascale
- phaser (DAC gateware)
    - [x] jesd204b link to DAC on KC705
    - [x] jesd204b dds demo [ARTIQ #580](https://github.com/m-labs/artiq/issues/580)
    - [ ] two-tone demo with kc705+ad9154 [ARTIQ #561](https://github.com/m-labs/artiq/issues/561)
    - [ ] DRTIO [ARTIQ #563](https://github.com/m-labs/artiq/issues/562)
        - [ ] DMA for pulse sequences [ARTIQ #553](https://github.com/m-labs/artiq/issues/553) and [ARTIQ wiki](https://github.com/m-labs/artiq/wiki/DMA)
- ingester (ADC gateware) 
    - [ ] jesd204b input 
    - [ ] jesd204b support for AD9656 
    - [ ] basic servo (PID-like) [wiki](https://github.com/m-labs/sinara/wiki/Servo) 
- board support
    - [x] white paper on clocking [here](https://github.com/m-labs/sinara/wiki/SinaraClocking) 
    - PCB_metlino
        - [ ] XCKU040 FPGA
        - [ ] FT4232H Quad High Speed USB to Multipurpose UART/MPSSE IC
        - [ ] Si5324C Precision Clock Multiplier/Jitter Attenuator
        - [ ] ...
    - PCB_sayma_AMC
        - [ ] XCKU040 FPGA
        - [ ] FT4232H Quad High Speed USB to Multipurpose UART/MPSSE IC
        - [ ] Si5324C Precision Clock Multiplier/Jitter Attenuator
        - [ ] ...
    - PCB_sayma_RTM
        - [ ] USR_UART 
        - [ ] HMC7043_SPI_ 
        - [ ] HMC830_SPI_ 
        - [ ] ADCLK948BCPZ  SiGe 2:8 Clock Fanout Buffer
        - [ ] AD7194BCPZ 8-Channel, 4.8 kHz, Ultralow Noise, 24-Bit Sigma-Delta ADC
        - [ ] Si5324C Precision Clock Multiplier/Jitter Attenuator
        - monitoring/reset/LEDs (also available for WUT via IPMI)
            - [ ] MAX6642ATT90 temperature sensor
            - [ ] LM75AIMM/NOPB temperature sensor
            - [ ] PCF8574ADW 8-Bit I/O Expander for I2C-Bus
            - [ ] 24LC64-I/MS EEPROM
        - RTM FPGA
            - [ ] XC7A15T Artix-7 FPGA

    - PCB_mezzanine_analog_allaki

hartytp, 2017-02-03:
@gkasprow Any updates about progress on the Sayma RTM layout?

gkasprow, 2017-02-03:
We are working hard on it.

Had some troubles with signal integrity of 10Gbit links and too high voltage drops on planes to the mezzanine connectors so had to rearrange routing.

However 95% of connections are done.

Hope to finish in 1 week.

hartytp, 2017-02-03:
@gkasprow Great!

jbqubit, 2017-02-09:
@sbourdeauducq created nice table in wiki for tracking prototype [hardware status](https://github.com/m-labs/sinara/wiki/hardware-status). @gkasprow updated the wiki. Thanks guys.  I'm deleting the now obsolete tabular I posted 14 days ago in this Issue to prevent confusion. Let's continue to use this Issue for related discussions.

gkasprow, 2017-02-09:
well, I crated table with short summary for each board so others know which
board is which.
But in this table the summary disappeared

gkasprow, 2017-02-09:
sorry, it is here
https://github.com/m-labs/sinara/issues/155

jbqubit, 2017-02-21:
@gkasprow What's the status of PCB_Sayma_AMC and PCB_Sayma_RTM?

gkasprow, 2017-02-21:
All Power Integrity issues  in Sayma AMC were detected and fixed. Now we are fixing SI issues.
PCB manufacturer verified production files and sent list of things to fix. So we are very close.
With Sayma RTM status is similar, I'm just fixing PI and manufacturing issues.

jbqubit, 2017-02-21:
Great news! Thank you for the update.

gkasprow, 2017-02-21:
For ones curious how PI issues tracking looks like, here is a screenshot of 4V distribution domain.
[image]
[image]

gkasprow, 2017-03-16:
A set of boards just arrived at WUT
[image]

jbqubit, 2017-03-27:
@gkasprow Are PCB_Sayma_AMC, PCB_Sayma_RTM and PCB_mezzanine_ analog_allaki in proto production yet?

hartytp, 2017-03-30:
ping @gkasprow any news about these PCBs (particularly Sayma RTM)?

gkasprow, 2017-04-03:
Yes, they are in production. I didn't get confirmation date yet - PCB factory is preparing production masks.
It took so long because we had endless troubles with Xilinx vivado ibis model for GTH and DDR3. I tried with 3 versions of vivado and each generated different ibis file. So different that with one model I got nice results while with another they were not acceptable. Old Xilinx ISE didn't cause such troubles.
Finally I had to analyze ibis file (8MB of text) and learn how to modify it. I figured out for example that:
- sometimes, in completely random way the ibis file contains information about simulation mode- sometimes probes are placed at the die, sometimes at the pins which makes really big difference for pins with ODT.
- sometimes ibis file has missing lines with diff signal definitions. One has to add them manually, otherwise Hyperlynx generates errors or even crashes.
- sometimes differential lines are not defined so simulation tool treats them as single ended.
Now I have a short manual how to process ibis files with certain error types :)
Frankly speaking, this project was simulated and tested to such extend like I never did before. Many critical bugs were found which otherwise would not be found immediately during deugging but would cause issues later, i.e. with higher loading of FPGA or capacitor value depreciation with time.
I'm pretty sure that with this effort pays off and we will save a lot of time later.

hartytp, 2017-04-03:
Great! Thanks again for all the hard work.

Let us know when you're ready for us to place the order for our hardware.

hartytp, 2017-04-16:
Nothing left to do here, so closing.

closed 2017-04-16 by hartytp