sinara-hw/sinara #133: 1.0 formal documentation

jbqubit, 2016-12-16:
@filipswit is leading effort to write formal documentation of Sinara. When will the documentation effort start? For inspiration see [documentation](https://cdsweb.cern.ch/record/1977919/files/CERN-ACC-2014-0357.pdf) for [AFC](http://www.ohwr.org/projects/afc/wiki).

gkasprow, 2016-12-16:
He does schematic review right now. It is very time-consuming activity
because every single connectivity and component pin need to be verified. He
already found over 20 hardware errors in SInara AMC

filipswit, 2016-12-16:
@jbqubit  Thanks for inspiration documents, I think I'm familiar with them.  As @gkasprow  has written, these days I'm checking schematics. I will take me about 1-2 weeks. So January is good time to start.

jbqubit, 2016-12-18:
Thank you for the update @filipswit. :)

jbqubit, 2017-02-21:
@gkasprow said

> @filipswit started seriously working on Sayma and Metlino documentation. Filip just finished schemtic review of all boards and found several issues. He will use tex templates from AFCK.

That's great! @filipswit Re-open this Issue (or create new ones) if there are questions.

jbqubit, 2017-02-21:
In #129 @hartytp said

> What do people think about the idea of using the Wiki as the main source of documentation? This could have a few advantages:

>Easy for people to edit later on, for example to post measurement results or comment on hardware bugs we find

> Allows us to keep the entire Sinara ecosystem documented in a single place, rather than having a tex file for each component

>It's already a git repo, which makes version control easy

jmizrahi, 2017-02-22:
@hartytp I strongly disagree with this suggestion. A manual needs to be a well-structured, thought-out document that a new user can read through and get a clear picture of the piece of hardware in question, what different design decisions were made and why, or reference when questions come up. Once the hardware is built, it will be mostly static.
This is totally different from a free-for-all discussion forum (where one might "post measurement results or comment on hardware bugs"). It is also very different from the distributed, non-hierarchical node structure of a wiki, with fuzzy boundaries on where the manual begins and ends.
The manual can certainly be updated with things like lists of known bugs in the future (and should be!), but this should happen in the context of a structured document. The manual should live alongside the more fluid back-and-forth discussion you're talking about.

hartytp, 2017-02-22:
> A manual needs to be a well-structured, thought-out document that a new user can read through and get a clear picture of the piece of hardware in question, what different design decisions were made and why, or reference when questions come up. Once the hardware is built, it will be mostly static.

This should all apply to the Wiki as well: the bulk of the content should be written during the initial documentation stage, and it should only undergo minor modifications to keep it up to date, or additions when new hardware is added to the ecosystem. When I talk about "measurement results" I mean significant pieces of characterisation, like phase noise spectra that all users want to see, not random things that people have done. I don't think a well-managed Wiki is ever a "free-for-all discussion forum" -- that's what the mailing list is for!

Would it make you less unhappy if I changed my suggestion to: make the documentation a GitHub project written in markdown (rather than Latex)? In that case, it's easy to integrate content from the manual into the Wiki. The major benefit of this as I see it is that we can have a single website which provides documentation for the entirety of Sinara (Metlino, Sayma, Kasli, DACs, etc). This is really helpful for new users who are trying to understand what the hardware ecosystem looks like and how everything fits together (it's much harder to have a stack of independent manuals that don't cross-reference properly). 

NB we can restrict who has commit access to the Wiki so it's not a complete free-for all. Also, as the Wiki is a git hub project, we can also publish "releases" of it (e.g. after Sinara round 1), which freeze it -- or sections of it -- a certain points.

jmizrahi, 2017-02-22:
> The major benefit of this as I see it is that we can have a single website which provides documentation for the entirety of Sinara (Metlino, Sayma, Kasli, DACs, etc).

I fail to see how this cannot be done with a single PDF, if you really think they're that entangled. Who said that there has to be one manual per board?

BTW, I don't agree that they are that entangled at the level of hardware. Each of these boards can operate on its own without the others. 

The intended software and firmware makes them part of a larger ecosystem, but software documentation is clearly going to have to be elsewhere, i.e. part of the normal ARTIQ documentation.

hartytp, 2017-02-22:
> I fail to see how this cannot be done with a single PDF, if you really think they're that entangled. Who said that there has to be one manual per board?

It can all be put into a PDF in principle. But, the plan is to start writing the documentation for Sayma and Metlino now, while the hardware ecosystem is still evolving rapidly. So, the documentation will either have to cover only those boards, and not cover the wider ecosystem (and how they interact with it), or run the risk of being obsoleted rapidly.

Why the preference for tex PDF over markdown files anyway? Markdown files can be saved and opened locally in a browser if desired (rather than over the web). They can also be complied into a PDF if that's important to you. 

> The intended software and firmware makes them part of a larger ecosystem, but software documentation is clearly going to have to be elsewhere, i.e. part of the normal ARTIQ documentation.

The way this hardware is used in an experiment, and the other hardware it plugs into makes it part of the wider ecosystem.

hartytp, 2017-02-22:
> I fail to see how this cannot be done with a single PDF,

Also, if you use MD or similar then you can make the documentation easier to navigate by including sidebars etc.

jmizrahi, 2017-02-22:
> Why the preference for tex PDF over markdown files anyway? Markdown files can be saved and opened locally in a browser if desired (rather than over the web). They can also be complied into a PDF if that's important to you.

If all we're talking about is document formatting platforms, I really don't want to argue about it. I refer to you to the many excellent references on how TeX is a wonderful document formatting system, produces attractive and easy to read documents, and why PDF is a very useful format. I was more talking about the nature and structure of a manual, not how it's rendered.

> The way this hardware is used in an experiment, and the other hardware it plugs into makes it part of the wider ecosystem.

You are presuming a level of uniformity in how this hardware is used, which is both unfounded and unnecessary. How _you_ use this hardware in _your_ experiment, and what other hardware _you_ plug it into, is on you to document. It does not belong in a Sayma manual. Sayma (and Metlino etc) are designed extremely flexibly, and can be used in all sorts of ways. I would leave the manuals separate.

hartytp, 2017-02-22:
My thinking here is that we can write the documentation for Sayma/Metlino as pages in md, so that they can be viewed as standalone web pages or even compiled into a PDF if desired. However, as they're md, it's also trivial to import them into the Wiki github project. The same pages/files can then be used as part of a general Sinara documentation Wiki, which covers the entire ecosystem. 

****

> You are presuming a level of uniformity in how this hardware is used, which is both unfounded and unnecessary. 

Sure, there will be large differences in how we all use this hardware, but there is a lot of common ground too and it's helpful to make that clear. As things currently stand, the ecosystem will mainly be made of a few different main kinds of things (uTCA boards, Kasli + extensions, etc) which interface together in a small number of different ways (IDC/DRTIO/backplane). IMO, it's worth making that clear in the docs, rather than just documenting the functionality of each module independently without providing any context of why that functionality was provided, or what it is intended to be used for.

e.g. The Metlino VHDCI is designed to interface with the VHDCI carrier. So, it's nice to be able to follow a link from the Metlino docs directly to the VHDCI carrier docs. The VHDCI carrier is designed to interface with all of the extension modules (not just the TTL ones, but also the DACs, ADCs, camera link, etc) so it's useful to be able to directly link from the VHDCI carrier to a list of the available extension modules -- and it's useful for that list to grow as more modules become available. It's also useful to know that Kasli provides an alternate carrier for these same peripherals, and that it can interface with Metlino via DRTIO/fibre.  It's also nice to have a high-level introduction to all of this in the beginning of the docs.

jbqubit, 2017-02-22:
I agree with @hartytp that it's useful to include cross linking in the documentation. Since @filipswit and @gkasprow are writing the documentation I'm inclined to let them choose the language. 

Some observations. 

- [PDQ2 documentation](https://github.com/m-labs/pdq2/tree/master/doc) is written in [reStructured text](http://www.sphinx-doc.org/en/stable/rest.html), includes LaTeX inline, is rendered as html using [Sphinx](http://www.sphinx-doc.org/en/stable/) and is published on readthedocs.io ([link](http://pdq2.readthedocs.io/en/latest/architecture.html)). 
- [ARTIQ documentation](https://github.com/m-labs/artiq/tree/master/doc/manual) is also written in reStructured text, rendered Sphinx and published by [M-Labs](https://m-labs.hk/artiq/manual-release-2/). 
- Sphinx [supports](http://blog.readthedocs.com/adding-markdown-support/) Markdown. I personally prefer writing Markdown over reStructured text. 
- A wide range of [formats](https://github.com/m-labs/pdq2/blob/master/doc/Makefile) are supported by Sphinx including html, latex and PDF. 
- Sphinx provides natural support for cross linking with existing ARTIQ-related docs and source.

hartytp, 2017-02-22:
@jbqubit Works for me.

jasonamini, 2017-02-22:
@jbqubit I agree. Along those lines, I suggest @filipswit writes a full manual to be delivered with the hardware (e.g. a TeX file or PDF) as the "official" documentation. We can then reformat it to an editable user version that can be posted and edited by the community. 

Has anyone here used Pandoc (or similar) for converting between document markups? If so, does it work OK?

hartytp, 2017-02-22:
Okay, so to summarise what's been said (as I understand it), the plan is to write two forms of documentation:

1. Board-level documentation of Metlino + Sayma. This won't describe how they fit into the general Sinara ecosystem (inc Kasli etc). It will be "frozen" and not updated to keep up with changes in the Sinara ecosystem. It will take the form of a PDF.
2. Overall Sinara documentation. This has both a high-level overview of the entire ecosystem and board-level documentation for all Sinara-compatible hardware we build (Sayma and Metlino as well as future Kasli extensions, and relevant support hardware like the RF amp we've discussed). It will be kept up do date with changes in the ecosystem, and will be extensively cross-linked. I imagine this looking like a fleshed out version of the current Sinara Wiki.

WUT will write the board-level documentation for Sayma and Metlino as their "official" documentation effort. They will aim to do this in a way that makes it easy to integrate into the overall Sinara documentation.

Questions:

1. Who will be responsible for writing an initial version of the SInara documentation and merging the Metlino/Sayma docs into it?
2. Where will we host the Sinara documentation?

hartytp, 2017-02-22:
@jordens How does this fit in with your plans for documenting Kasli + the other hardware you're planning?

gkasprow, 2017-02-22:
Anyway, we will write dedicated manuals/datasheets for each board. They are quite universal and can also serve other purposes.

Our referable way is latex because we can nicely publish vector images but I’m open to other forms.

gkasprow, 2017-02-22:
I also think that board manuals should be separate documents with all details which are needed to fully explore their possibilities.

Such things like assembly options, diagnostics, testpoints are not very interesting for ARTIQ user. And for such document we need good tool that supports vector graphics, search functions and looks more like manual then webpage.

This is example of such documentation

http://creotech.pl/wp-content/uploads/2015/08/CTI-AFCK_datasheet.pdf

There should be another general document covering certain use cases that embeds interactive links to detailed documentation and here wiki would be better idea since the content will develop intensively while board manuals will be static.

gkasprow, 2017-02-22:
@hartytp it seems we came to the same conclusion:)

jordens, 2017-02-23:
It works well.

jordens, 2017-02-23:
@hartytp Right now I am just writing an overview about Kasli. I'll do so in reST. Bu I am much more concerned with the content that with the format.

hartytp, 2017-02-23:
> @jordens: right now I am just writing an overview about Kasli. 

Great! If you can cover everything we need to make Kasli compatible extensions (e.g. plans for a Kasli BP if there will be one, etc) in the process, then that will be a really useful resource.

> Bu I am much more concerned with the content that with the format.

As am I. Anything that integrates well into a nice, easily navigable webpage works well for me.

jbqubit, 2017-07-25:
@filipswit How's the documentation progressing? Please give an update.

filipswit, 2017-07-26:
Generally I have cores for all documents. Some pictures are just placeholders for appropriate pics. Signal table for Metlino is incorrect.
[SaymaRTM.pdf](https://github.com/m-labs/sinara/files/1176583/SaymaRTM.pdf)
[SaymaAMC.pdf](https://github.com/m-labs/sinara/files/1176584/SaymaAMC.pdf)
[Metlino.pdf](https://github.com/m-labs/sinara/files/1176585/Metlino.pdf)

Please say what else would you like to see in doc.

dhslichter, 2017-07-26:
It would be nice to have bus widths shown on the block diagram, thoughts from others?

jbqubit, 2017-07-26:
Thanks @filipswit! Some comments follow. 

# Sayma_AMC
- Xilinx Kintex UltraScale
    - [ ]  how many transceivers, transceiver type, transceiver maximum data rate for FPGA speed grade 

- MGT  
    - [ ] define MGT to be Multi-Gigabit Transceiver
    - [ ] new Figure devoted to routing of MGT resources

- Figure 4: General Block Scheme
    - [ ] both DRAMs are missing
    - [ ] MMC is missing

- FPGA bootstrapping
    - [ ] discussion of how this works
    - [ ] how to load FPGA .bit into flash (is there a .bit for bootstrapping FPGA passthrough to SPI flash?)
    - [ ] how to use MMC to load .bit into flash 

- MMC bootstrapping
    - [ ] discussion of how this works
    - [ ] how to load default .elf into flash
    - [ ] where to obtain default .elf
    - [ ] link to documentation on MMC software

- housekeeping signals
    - [ ] signals available to MMC/IPMI (eg temperature sensors, current, voltage, fan) and addresses
    - [ ] signals available to FPGA (eg temperature sensors, current, voltage, fan) and addresses

- FMC
    - group together FMC-related info into a single section
    - to what extent is implementation compliant with VITA FMC HPC?
    - clock and IO resources
    - VADJ voltage and maximum current draw
    - which IO banks are used on FPGA? 

- SFP
    - to what extent is this implementation compliant with SFP standards? 

- uTCA.4 compliance 
    - [ ] discussion of ways board is divergent from uTCA.4
    - [ ] how to disable uTCA features like power interlock checking

- USB-UART
    - [ ] how does this this chip present its multiple UART interfaces to PC? 
    - [ ] switch routing addresses
    - [ ] USB driver installation (if any)

- power
    - [ ] list power source options -- card edge vs stand-alone connector (connector name?)
    - [ ] voltage and maximum power consumption
    - [ ] is active cooling required for board? 

- JTAG
    - [ ] interface options (UART vs IDC header)
    - [ ] addressing 

- Signal Tables
    - [ ] column headers missing 
    - [ ] define FP1, FP2 and P2P

# Sayma RTM

- [ ] Get additional narrative about Project Description from the wiki. 
- [ ] Cite high speed ADC and DAC ICs by name and summarize high-level specifications -- that's what this board is about! 

- Figure 4 Block Scheme
    - [ ] SATA connectors
    - [ ] JTAG connector

- AFE mezzanine
    - [ ] short narrative discussing purpose of AFE mezzanines
    - [ ] AFE mezzanine is Analog Front End mezzanine
    - [ ] group together all AFE connectivity in single new section
    - [ ] list resources available to each AFE
    - [ ] list expectations for AFE modules
        - [ ] maximum power consumption
        - [ ] impedance, directionality, voltage ranges, maximum frequency
        - [ ] what to do to avoid damaging ADCs (fast and slow) on Sayma_RTM PCB
    - [ ] standoff length and screw size 

- clock distribution mezzanine
    - [ ] short narrative discussing purpose of clock mezzanine
    - [ ] see all above for AFE mezzanine

- housekeeping signals
    - [ ] signals available to AMC-side MMC/IPMI (eg temperature sensors, current, voltage, fan) and addresses
    - [ ] signals available to RTM-side FPGA (eg temperature sensors, current, voltage, fan) and addresses

- clock distribution
    - [ ] include details from Sinara wiki on clock configuration options -- there are many! 
    - [ ] use nomenclature for clock names used in wiki and Issues
    - [ ] RF back plane interface

- FPGA bootstrapping
    - [ ] discussion of how this works
    - [ ] not necessarily ARTIQ-specific 

- Figure 5
    - [ ] use AFE mezzanine not "mezzanine connector" 
    - [ ] clock mezzanine is missing 

- power
    - [ ] list power source options 
    - [ ] voltage and maximum power consumption
    - [ ] is active cooling required for board? 

# Metlino
- [ ] short discussion of special role of MHC slot in uTCA system
- [ ] short summary of the star topology of uTCA backplane along with cartoon schematic
- [ ] diagram showing how this board stacks with others for minimal MHC operation
https://github.com/m-labs/sinara/issues/21#issuecomment-267459815

See comments for Sayma_AMC. Most apply here.

filipswit, 2017-07-28:
@jbqubit, @dhslichter  Many Thanks for your comments!! It is much easier to write sth. with such a good feedback.

jbqubit, 2017-11-22:
Now that I have an Sayma_AMC to play with I can see additional documentation needs. 

- power configuration
    - power on delays and time constants for each IC
    - for each potential generated by the Exar chips
        - voltage, current, PID parameters, observed voltage noise
    - how to configure Exar chips
        - PCB header vs SPI to MMC
        - what version of Exar Power Architect?
        - location of [configuration](https://github.com/m-labs/sinara/tree/master/EXAR_config) files 
        - need to establish version control for power configuration files -- official release?

- JTAG 
    - which FTDI chip is used? 
    - jtag clock speed
    - SCANSTA112
        - Say what the SCANSTA112 chip does generically. 
        - Say how the chip is used in the context of Sayma_AMC -- sitcher mode, right?
        - What is power-on selection? 
        - What are all the JTAG busses in the system? What are their SCANSTA addresses?
        - For known ICs with JTAG interfaces list idcode, IR length, human readable description 
    - diagram showing how JTAG chain is extended by addition of RTM, FMC, etc. 
      contrast with KC705 approach Figure 1-8 
      https://www.xilinx.com/support/documentation/boards_and_kits/kc705/ug810_KC705_Eval_Bd.pdf

- Front panel
    - what are all the LEDs? what do their different states indicate? uTCA leds 
    - What does LPC1776 do when I press PB3 SW? Does it pull FPGA_RESET? 
    - front panel USB port exposes four serial port... what are they connected to?
    - names for SFP connectors
    - SMA connectors

- list of headers on PCB
    - JTAG multiple 
    - Cortex programmer plug, i.e. LPCXpresso (MMC_TMS...) 

- SW1
    - say its there even if its not yet used
	- FPGA_M0
	- FPGA_M1
	- FPGA_M2  

- I2C mux
    - mapping between address and devices
    - discuss how this is shared between FPGA and MCU
    - power on selection

- RGMII
    - discuss how this is shared between FPGA and MCU
    - power on selection
    - mention LD19 

- interface between AMC FPGA and MCU uP
    - discuss how this is supposed to work 
    - I2C 
    - SPI flash
    - RGMII

- Details of factory acceptance testing
    - power tests: stimulus, what to look for, passing values, related source code 
    - communication tests: stimulus, what to look for, passing values, related source code

jbqubit, 2017-11-24:
@filipswit What's the status of the documentation? The hardware has shipped.

gkasprow, 2017-11-28:
Expect documentation update from @filipswit <https://github.com/filipswit>
witnin 2..3 days.
He is working hard on it.

filipswit, 2017-11-29:
@jbqubit Here you have some update of documentation. I will add here every few days some newer version.

[SaymaAMC.pdf](https://github.com/m-labs/sinara/files/1515961/SaymaAMC.pdf)

jbqubit, 2017-11-30:
It would be best to hold off on posting until you've generated a version you consider ready for feedback. In particular, please implement (or reject) all the suggestions I made for the previous version before reposting. That makes it easier for all the contributors to comment/discuss the same document. 

Feedback on what you posted yesterday (Issue Date November 29, 2017).

p. 4 -- lots of the acronyms don't make sense without a short introduction to uTCA.4
p. 4 -- transceiver type, maximum data rate
p. 4 -- spell out what MMC is an abbreviation for, give brief synopsis of properties of LPC1776
p. 7 -- back view of PCB connectors
p. 7 -- consider [illustrating ](https://www.ifixit.com/Teardown/iPhone+X+Teardown/98975#s182886) the location and description of the different IC's in the way that iFixIt does
p. 13 -- LED call outs look great! 
p. 18 -- excerpts from schematics are helpful at explaining; more verbose would help

**Suggestion**: Ask a friend who knows nothing about Sayma to read your documentation and ask questions. This will be a type of feedback that is invaluable as it is from the perspective of a person who will use the documentation to understand what the board does.

gkasprow, 2017-11-30:
@filipswit you can add introduction to MTCA - just copy part of my [article](https://www.google.pl/url?sa=t&rct=j&q=&esrc=s&source=web&cd=6&ved=0ahUKEwiqoc2DiufXAhXDbxQKHfNNCQwQFghDMAU&url=https%3A%2F%2Fyadda.icm.edu.pl%2Fbaztech%2Felement%2Fbwmeta1.element.baztech-00845d8e-149f-4fb0-bbbd-e92a29c80032%2Fc%2FKasprowicz_G_vol62_no3.pdf&usg=AOvVaw3g7GoDjNNt-1gu27HCklYY) about it.

filipswit, 2017-12-06:
> p. 4 -- lots of the acronyms don't make sense without a short introduction to uTCA.4

Would you like an introduction or a glossary?

gkasprow, 2017-12-06:
Do both.

jbqubit, 2017-12-07:
@filipswit How's the documentation coming? Your docs are helpful for interfacing with M-Labs too. :) https://github.com/m-labs/artiq/issues/854#issuecomment-349905800

jbqubit, 2017-12-11:
@filipswit ping

filipswit, 2017-12-11:
This is what I've done so far. 
> p. 4 -- lots of the acronyms don't make sense without a short introduction to uTCA.4

There is a glossary, please say which one should I add yet.
Introduction will come later.
> p. 4 -- spell out what MMC is an abbreviation for, give brief synopsis of properties of LPC1776

abbreviation is in glossary. Properties will be later.
>p. 7 -- back view of PCB connectors

will take a photo this week
>p. 7 -- consider illustrating the location and description of the different IC's in the way that iFixIt does

p.17 - p18 Did you expect sth like this? 
>p. 13 -- LED call outs look great!

Thanks!
>p. 18 -- excerpts from schematics are helpful at explaining; more verbose would help

Text in progress

jbqubit, 2017-12-12:
Thanks @filipswit -- not enough changes to warrant a re-read. Please respond to existing comments before sending for review. See eg https://github.com/m-labs/sinara/issues/133#issuecomment-346499637 Also please post both source and PDF next round.

jbqubit, 2017-12-14:
Need to add MMC documentation.
- Where is source code?  https://github.com/m-labs/sinara/tree/master/SAYMA_firmware
- How to compile?
- How to flash?
- What are configuration steps done by MMC.

jbqubit, 2017-12-14:
@filipswit When you've implemented documentation changes in this Issue please close it. For next documentation review please create a new Issue for that revision. This is  Issue is getting too long.

jbqubit, 2018-01-11:
@filipswit How's the documentation coming? Please feel free to check items off (above) as you implement them.

jbqubit, 2018-01-12:
@filipswit It's getting critical that the documentation be finalized for Sayma v1. The PLL and Ethernet Issues will be soon resolved. Docs are one of the last parts of the WUT contract.

jbqubit, 2018-01-17:
@filipswit @gkasprow I'm not confident that details such as the following are being added to the Sayma v1.0 documentation. 

- [ ] JTAG always-on configuration cf https://github.com/m-labs/sinara/issues/463#issuecomment-358046337
- [ ] document which ICs have Ethernet access for Sayma 1.0. https://github.com/m-labs/artiq/issues/854

filipswit, 2018-02-04:
I added most of the Issues, some of them may still nedds some corrections. But I think is good moment to get some new remarks;)

Actualy, I dont't know what you expect about this "document which ICs have Ethernet access for Sayma 1.0. m-labs/artiq#854" - for now just added placeholder.

The  source code is already in git.
[SaymaAMC.pdf](https://github.com/m-labs/sinara/files/1693420/SaymaAMC.pdf)

jbqubit, 2018-02-06:
Awesome @filipswit! Looking forward to jumping in and reading. :)

jbqubit, 2018-02-07:
For reference, this version of the documentation provided by @filipswit has commit id https://github.com/m-labs/sinara/commit/e745783298d77aebc46f4e7b66e0e10ec732c286. 

This is a big job @filipswit! Thanks for tackling it. Some comments are below. 

# Introduction to ARTIQ Sinara 

Start with a section outlining the ARTIQ Sinara ecosystem. Language from wiki seems fine. 
https://github.com/m-labs/sinara/wiki/images/overview.png

---

Sinara is an open-source hardware ecosystem originally designed for use in quantum physics experiments running the ARTIQ control software. It is licensed under CERN OHL v1.2.

Control electronics used in many trapped-ion and other quantum physics experiments suffers from a number of problems. In general, an ad-hoc solution is hastily put together in-house without enough consideration about good design, reproducibility, testing and documentation. This makes those systems unreliable, fragile, and difficult to use and maintain. It also duplicates work in different laboratories. In addition, the performance and features of the existing systems (e.g. regarding pulse shaping abilities) is becoming insufficient for some experiments.

To alleviate those problems, Sinara aims to be:
- high-quality
- simple to use and "turn-key"
- reproducible and open
- flexible and modular
- well tested
- well supported by the ARTIQ control software

Sinara is currently developed by a collaboration including M-Labs, Warsaw University of Technology (WUT), US Army Research Laboratory (ARL), the University of Oxford, the University of Maryland and NIST. The majority of the hardware was designed by WUT. The work was funded by ARL, Duke University, the University of Oxford, and the University of Freiburg.

Following the ARTIQ model, an experiment consists of a __Core Device__ (master) -- typically either a Metlino or Kasli -- controlling multiple slave devices in real time using ARTIQ's distributed real-time IO (DRTIO) protocol. DRTIO provides both gigabit communication links and time distribution over copper cable or optical fibres. It synchronises all device clocks, ensuring they have deterministic phase relationships, and enables nanosecond timing resolution for input and output events across all devices in the experiment. More detailed information about communication between devices and time distribution inside Sinara can be found here.

Sinara uses two main form factors for hardware requiring real-time control: microTCA (uTCA) and Eurocard Extension Modules (EEM). Non real-time hardware is typically connected to the host PC using Ethernet.

MicroTCA (uTCA) is Sinara's preferred form factor for high performance hardware with high-speed data converters requiring deterministic phase control, such as the Sayma Smart Arbitrary Waveform Generator (SAWG). Information about uTCA hardware, including a list of parts needed to build a Sinara uTCA crate can be found here.

EEMs provide a lower cost, simpler platform than uTCA for hardware that requires real-time control, but not bandwidth or complexity of uTCA hardware.

Extension modules connect to a carrier, such as Kasli or the VHDCI carrier, which provides power and DRTIO. They are designed to be mounted either in stand-alone enclosures, or in a rack with a carrier, and connect to the carrier via ribbon cable. More details about the extension module standard can be found here.

uTCA hardware interfaces with the extension modules either directly, using a VHDCI carrier, or indirectly, using a Kasli DRTIO slave.

-------------

# 1 Overview

Rename section to "uTCA.4  Overview".

"We make use of the most recent extension to the uTCA standard, uTCA.4" is not accurate. True we use uTCA.4. But latest is uTCA.4.1 is the [latest](https://www.picmg.org/resources/news/member-press-releases/n-a-t-announces-new-microtca-4-1-low-level-radio-frequency-llrf-backplane-for-high-energy-physics-research-and-telecom-applications/). 

Figure 1: There's an extra board in AMC 7 that's fine to have in the picture but confusing if the boards are not labeled. Add colored overlay labeling MCH, Sayma AMCs, power module. 

New Figure:  add photo of Sayma_AMC attached to Sayma_RTM sitting on bench so how they're connected is clear. 

# 2 uTCA.4 RF Backplane
Just lump this together with pervious section. Yes, please add data sheet. 

# 3 uTCA in Sinara

Delete this section since it's covered in the new Sinara overview. 

# 4 uTCA parts and suppliers

Here's what I ordered in the past.

- NAT AC 600D, qty 1
- NAT MCH-Basic v3.5, mid-size front panel, qty 1 
- NAT Native-R5, qty 1

# 5 Schematic / Layout Viewer
Add additional content. 
- ARTIQ_EE folder is for designs made with the Mentor Graphics Xpedition Enterprise CAD tool.
- ARTIQ_ALTIUM folder is for designs made with Altium Designer CAD tool.

# 6 Sayma
[... 233 lines cut ...]

# 19 Housekeeping

Narrative discussing uTCA housekeeping concept where MMC collects info and passes on to MCH via IPMI. Discuss that OpenMMC will run on MMC chip and that an OpenXXX will run on MCH and collect AMC device housekeeping information. 

Discuss how I2C busses are shared between MMC and FPGA. Do this here or refer to earlier section where it's discussed. 

# 19.2 Safety interlocks
temperature and current safety interlocks. Is this enforced by MMC or by Exar chips? How does a user know if an interlock is active? 

# 21 Factory acceptance tests

This is being written by Creotech, right? Please ask that they commit their test jigs and code to repository. 

# A Appendix

Give this Appendix a name.  

# Gateware configurations

https://github.com/m-labs/sinara/wiki/Sayma-gateware#data-and-sample-rates

# Missing
- OpenMMC interaction. 
- Ethernet sharing between MMC and FPGA.

jbqubit, 2018-02-19:
@filipswit Please refactor in light of my comments. I'd like to do review of AMC and RTM docs at the same time.

jbqubit, 2018-02-27:
@filipswit Please update so we can get this Issue closed and get WUT paid. A student from Duke University will be visiting UMD this weekend who is a Sayma user. I will ask him to read your documentation so you can get feedback from a user with fresh eyes.

filipswit, 2018-02-27:
I think by tomorrow evening will be able to put newer version on repo.

jbqubit, 2018-02-27:
Splendid. Please aim for RTM too and I'll give them a read together.

filipswit, 2018-03-01:
I have already pushed on repo Sayma AMC, from some reason i couldn't commit it via git so I pushed it via files upload - It is with default message;/

[SaymaAMC.pdf](https://github.com/m-labs/sinara/files/1770717/SaymaAMC.pdf)

jbqubit, 2018-03-01:
Thank you! Will review this weekend.

jbqubit, 2018-03-01:
@filipswit @gkasprow  The documentation for MMC is broken. 

> The source code is written in C and can be found on github.
Source code: https://github.com/m-labs/sinara/tree/master/SAYMA_firmware.
pre-compiled binary: https://github.com/m-labs/mmc-firmware/releases
>OpenMMC Project:https://github.com/lnls-dig/openMMC

There are three different repositories referenced. Which one is right? There's no pre-compiled binary at the specified location. Please let me know which to use since I'd like to update the MMC firmware on my board.

jbqubit, 2018-03-01:
More on MMC flashing... 
- For pre-compiled binary do you mean [this](https://github.com/m-labs/mmc-firmware/blob/master/releases/openMMC%20v1.0.axf) .axf file? FlashMagic expects .hex or .bin. 
- The instructions don't specify for instance "Interface" type. 
- Rhetorical question: how do I know if I'm correctly connecting to MMC uP? Should I attempt to Read Signature first? 
- Please rewrite FlashMagic instructions while actually flashing a board. 

Also not mentioned is how the serial ports are used by Sayma. 
```
/dev/ttyUSB0 JTAG
/dev/ttyUSB1 :: usually the MiSoC console 
/dev/ttyUSB2 
/dev/ttyUSB3 :: MMC console  115200, 8, n, 1 
```

jbqubit, 2018-03-04:
Empirically, programming using FlashMagic requires baud rate of 57600 and selection of Interface "None (ISP)". Communicating with MMC serial port requires baud rate of 115200.

closed 2018-10-25 by jbqubit