sinara-hw/sinara #135: uTCA.4 backplane schematics

jbqubit, 2016-12-21:
Please add schematics for backplanes to the repository. This will make it easier to understand dependencies like activation of clock circuits as in #118. 
- [ ] PCB_BP_AMC_7slot
- [ ] PCB_BP_AMC_12slot
- [ ] PCB_BP_RTM_12slot
- [ ] PCB_BP_RTM_7slot

jbqubit, 2016-12-21:
Can NAT or Schroff supply these for the specific part numbers we've specified for 0.1?

jbqubit, 2017-02-13:
@gkasprow Is this info available? We need schematics of some sort so end users and testers know how to use the backplane.

jbqubit, 2017-12-11:
@gkasprow Is this information available? If yes please attach link to this Issue and add to wiki. If not let's just close.

gkasprow, 2017-12-11:
It's not available to my knowledge.

closed 2017-12-13 by jbqubit