sinara-hw/sinara #139: m-labs Sinara timeline, checklist

jbqubit, 2017-01-26:
Similar to #131 tracking delivery timeline for Sinara hardware the intent of this Issue is to layout the steps needed for ARTIQ support of the Sinara hardware. I'm sure I'm missing items so please suggest additions. Spawn new Issues for things that require discussion. 

# phaser 
DAC gateware
- [x] jesd204b link to DAC on KC705
- [x] jesd204b dds demo [ARTIQ #580](https://github.com/m-labs/artiq/issues/580)
- [ ] two-tone demo with kc705+ad9154 [ARTIQ #561](https://github.com/m-labs/artiq/issues/561)
- [ ] DRTIO [ARTIQ #563](https://github.com/m-labs/artiq/issues/562)
    - [ ] DMA for pulse sequences [ARTIQ #553](https://github.com/m-labs/artiq/issues/553) and [ARTIQ wiki](https://github.com/m-labs/artiq/wiki/DMA)
 
# ingester (ADC gateware) 
- [ ] jesd204b input 
- [ ] jesd204b support for AD9656 
- [ ] basic servo (PID-like) [wiki](https://github.com/m-labs/sinara/wiki/Servo) 

# Sinara board support
White paper on clocking is [here](https://github.com/m-labs/sinara/wiki/SinaraClocking). ICs common to several boards in the Sinara family follow. 
- monitoring/reset/LEDs (also available for WUT via IPMI)
    - [ ] I2C_MUX
    - [ ] LM75AIMM/NOPB I2C temperature monitors
    - [ ] MAX6642ATT90 temperature sensor
- [ ] Si5324 Precision Clock Multiplier/Jitter Attenuator
- [ ] MAX24287ETK+ Ethernet interface
- [ ] FT4232H Quad High Speed USB to Multipurpose UART/MPSSE IC
- [ ] FMC support for CERN test jig used by WUT 

## Kintex Ultrascale related
This is XCKU040 FPGA. Primary tracking is [#563](https://github.com/m-labs/artiq/issues/563).
- [x] OpenOCD support for Kintex Ultrascale
- [x]  Dual flash support
- [ ]  JESD204 for Kintex Ultrascale (GTH)
- [ ]  JESD204B synchronization and group-delay/latency alignment
- [ ]  port of DRTIO to KintexUltrascale #562
- [ ]  Kintex Ultrascale DDR3
- [ ]  High-resolution IOSERDES TTLs for Kintex Ultrascale

## PCB_metlino
- LPC17762984 used for IPMI (ignored by m-labs)
- [ ] VHDCI support exposes IO that mimics Kasli peripheral interface
- [ ] RTM support exposes IO, clk in ARTIQ for future uses

## PCB_sayma_AMC
- LPC17762984 used for IPMI (ignored by m-labs)

## PCB_sayma_RTM
- [x] AD9154 DAC
- [ ] AD9656 ADC -- not yet funded
- clock distribution (summarized [here](https://github.com/m-labs/sinara/issues/75))
    - [ ] HMC7043 SPI  14-Output Fanout Buffer for JESD204B
    - [ ] HMC830 SPI fractional-N PLL
- [ ] AD7194BCPZ analog mezzanine monitoring/calibration 8-Channel, 4.8 kHz, 24-Bit ADC
- monitoring/reset/LEDs (also available for WUT via IPMI)
    - [ ] PCF8574ADW 8-Bit I/O Expander for I2C-Bus
    - [ ] 24LC64-I/MS EEPROM
- Sayma RTM/AMC FPGA link
    - [ ] XC7A15T Artix-7 FPGA
    - [ ] link support for ARTIQ SPI PHY 
        - timing requirements of remote SPI: match ARTIQ SPI PHY (what is that spec?) 
    - [ ] link support for ARTIQ TTL PHY
        - timing requirements of remote TTL: match ARTIQ TTL PHY (1 ns resolution)

## PCB_mezzanine_analog_allaki_allaki
Following is specific to Allaki and seems nearly trivial. Intent is to build infrastructure to support configuration of a variety of analog mezzanines. Assume different analog mezzanines may be installed in the same Sayma_RTM motherboard. IO is administrered by Sayma RTM/AMC FPGA link. 
- HMC542BLP4E variable attenuator via TTL PHY
- HMC849ALP4CE RF switch via TTL PHY
- AD8363ACPZ-WP readout via ADC on PCB_sayma_RTM

jbqubit, 2017-02-16:
Support needed for AD7194BCPZ analog mezzanine monitoring/calibration ADC established in Issue #141.

jbqubit, 2017-02-22:
@jordens @sbourdeauducq Please propose a road map for support of Sayma v0.1 hardware with emphasis on the LogiQ use case. Please include testing, delegation and responsibilities.

jbqubit, 2017-07-17:
M-Labs hasn't made their own roadmap or checklist for Sayma support in the ARTIQ repository. I'd like to keep this open so I can roughly track what the steps are.

sbourdeauducq, 2017-07-17:
Take our quote SINARA2 and the earlier UMD one - it's in there. Why do you want it in the ARTIQ repository?

jbqubit, 2017-07-17:
A subset of the Sayma-dependencies is funded by ARL/UMD with the balance funded by Oxford. Other groups with an interest in Sayma don't have access to the contracts that fund the gateware/software work so are unaware about what's forthcoming. My previous efforts to introspect M-Labs development plans and make lists (eg above) were not well received. Please externalize your development trajectory and priorities for the benefit of the community. A simple check list worked reasonably well for eg Sayma hardware testing #224 and got some traction with DRTIO https://github.com/m-labs/artiq/issues/562. 

> Why do you want it in the ARTIQ repository?

@jordens said in #3 "Ack. Gateware and firmware can be tracked in artiq." As a bridge between Sayma and ARTIQ consider using the ARTIQ repo sayma tag https://github.com/m-labs/artiq/labels/area%3Asayma.

hartytp, 2017-07-17:
My preferred way of handling this is to create an ARTIQ issue for each item that's funded. cf issues like [687](https://github.com/m-labs/artiq/issues/687). I'll post issues for other items we've funded (e.g. Novogorny Urukul) servo soon.

sbourdeauducq, 2017-07-17:
Yes, one issue per item sounds good. Many already exist.

jbqubit, 2017-07-17:
@hartytp I agree. One Issue per item is a good approach. @sbourdeauducq please do this for the UMD/ARL contracts. And choose a unifying tag and/or milestone.

jbqubit, 2017-07-19:
@sbourdeauducq @jordens How do you want to handle this for UMD/ARL contracts?  I'm happy to create Issues as @hartytp did but I run the risk of dictating your workflow. Please comment.

hartytp, 2017-07-24:
AFAICT, this has been done now.

closed 2017-07-24 by hartytp

jbqubit, 2017-07-24:
Yes. Thank you @sbourdeauducq and @jordens. 

4.0 mile stones -- Sayma baseline support
https://github.com/m-labs/artiq/milestone/11

5.0 mile stones -- extended Sayma support 
https://github.com/m-labs/artiq/milestone/14