sinara-hw/sinara #140: gateware for Sayma AMC-RTM FPGA link

jbqubit, 2017-01-26:
In #9 we discussed IO multiplex mechanisms for interface between Sayma_AMC FPGA with Sayma_RTM_analog_mezzanine. A decision needs to be made as to the desired performance. Following is based on my review of #9. 

# Sayma RTM/AMC FPGA link

## IO performance

__16 IO pins__ per analog mezzanine
- latency: <1 ns RMS jitter
- directionality: bi-directional
- burst bandwidth: parallel update of all 16 IO at 100 MHz for up to 10 us (per mezzanine)
- continuous bandwidth: parallel update of all 16 IO at 10 MHz (per mezzanine)
- SPI
    - option to use any IO for SPI
    - default: gateware for 2 SPI channels on highest-numbered IO pins
- discussed in Issue starting [here](https://github.com/m-labs/sinara/issues/9#issuecomment-252083235)]

__2 pins for I2C__ per analog mezzanine
- hardware has dedicated I2C interface
- continuous bandwidth: 1 MHz data rate (per mezzanine)

## RTM FPGA Configuration
- XC7A15T Artix-7 FPGA .bit configuration by either of [two hardware-supported options](https://github.com/m-labs/sinara/issues/9#issuecomment-252353431))
    - config from Kintex IO pins (primary approach)
    - Sayma_AMC Flash, daisy chain Artix as slave Kintex Ultrascale as master (backup approach)

## Implementation
Possible implementations 

1) @sbourdeauducq "We can design deterministic receivers in FPGAs that do not require a clock, but having the clock makes a CDR unnecessary (so we don't need the more expensive FPGAs)."

2) Use DRTIO

gkasprow, 2017-01-26:
Please notice that I upgraded I2C scheme.
Now one can configure any combination of I2C devices to be connected to MMC or FPGA.
It is possible to select one I2C device at the time or any combination of them.
The upgrade is applied to all boards.

sbourdeauducq, 2017-01-27:
First of all DRTIO or not is independent from the type of transceiver used (GTX with CDR or IOSERDES with dedicated clock signal).

Then, the AMC-RTM FPGA link, or at least part of it, has to be up and running before DRTIO. This is because the clock chips that are required for DRTIO must be configured over the link. For this, we will use a Wishbone-over-serial bridge that will give the AMC CPU access to the SPI core on the RTM side and let it do transactions on the converter SPI bus.

If required, we could use one AMC-RTM LVDS pair for Wishbone, and the second that we have could be clocked synchronously to DRTIO for fixed-latency/time-critical transactions. Since we have only one Si5324, [this scheme](http://www.xilinx.com/support/documentation/application_notes/xapp589-VCXO.pdf) can be used on the Wishbone pair where phase noise (and fixed latency, since this scheme precludes GTP TX buffer bypass) is less important.

jbqubit, 2017-01-27:
>"we will use a Wishbone-over-serial bridge that will give the AMC CPU" 

By "AMC CPU" do you mean the AMC soft-core uP in the FPGA? I didn't think m-labs was using the AMC-side hardcore uP (LPC1776, Cortex-M3). 

Once bootstrap over has completed (using RTM_FPGA_LVDS1,2) communication with the RTM-FPGA will be via DRTIO using transceivers (RTM_FPGA_GTP_Rx0,1), right?  

Is the IO performance plausible?

sbourdeauducq, 2017-01-28:
Softcore uP. Everything over GTP. DRTIO is possible if desired, but needs support for switches and different data rates in the same system.

jbqubit, 2017-02-01:
Is is reasonable to start with a simple approach and upgrade in the future to DRTIO when [DRTIO switch](#619) becomes available? That is is #619 a hard dependency?

sbourdeauducq, 2017-02-02:
Start with a simple approach for doing what exactly?

jbqubit, 2017-02-02:
Initial implementation using one AMC-RTM LVDS pair to communicate with Wishbone bus RTM. SPI, I2C and TTL PHY's attach to the Wishbone bus for control of  RTM peripherals. I can't well gauge the relative simplicity of this approach. I believe some of the implications vs full DRTIO switch are the following. 
- RTM resource access would not be synchronous with DRTIO. 
- Bandwidth is limited by wishbone. How much? 
Are there others relative to IO Performance mentioned in opening post? 

In the future, use second LVDS pair, DRTIO switch for fixed-latency/time-critical transactions.

sbourdeauducq, 2017-02-03:
Again, this isn't only about simplicity or performance, the Wishbone link (or something equivalent) is made necessary by the clock chips on the RTM side which need to be running before DRTIO can work at all.

jbqubit, 2017-02-09:
OK. Let's proceed with the simple approach for v1.0 of Sinara hardware support. The implication of this is that real-time signals will not be available on the RTM side until the [DRTIO Switch](https://github.com/m-labs/artiq/issues/619) is developed.

jbqubit, 2017-02-09:
@sbourdeauducq Do you have enough information to generate a quote for this feature?

jbqubit, 2017-02-13:
Mirroring discussion in #148, I propose the following gateware configuration options for the analog mezzanine IO. 

- let c1 be 4 of SERDES_TTL_PHY (or 4 of TTL_PHY if DRTIO Switch is unavailable)
- let c2 be 1 of SPI_PHY (uses 4 IO lines)

Each analog mezzanine has 16 general purpose IO. Proposed configuration is 
- 4 groups of 4 IO: AM0, AM1, AM2, AM3
- AM0, AM1, AM2, AM3 each independently configurable as c1 or c2

Additional IO for each analog mezzanine: I2C.

jbqubit, 2017-02-16:
@dhslichter @dtcallcock @sbourdeauducq Any comments on my proposal from three days ago? If this is satisfactory I'll update leading post.

dhslichter, 2017-02-17:
These two configuration options, and being able to choose configurations in blocks of 4 IO as described, sounds good to me.  I2C is a separate entity.

jbqubit, 2017-02-21:
I updated the leading post to this Issue with a distillation of what I believe is the consensus. I'm closing the Issue as there don't seem to be unresolved questions. Please reopen if I'm mistaken.

closed 2017-02-21 by jbqubit