sinara-hw/sinara #141: features for analog mezzanine monitoring/calibration DAC

jbqubit, 2017-01-26:
What ARTIQ support is desired for the AD7194BCPZ  16-Channel, 4.8 kHz, 24-Bit ADC? It's channels are shared among the analog mezzanines. 

# Features
- setting/getting IC register contents by RTIO
    - register only support for digital filtering options
- support for single-ended (16) or differential (8) data acquisition (data coding differs) 
- default: continuous-conversion mode (does not power-down automatically after read)
- synchronization: no need for synchronization of acquisition with DRTIO clock
- high level interfaces to 
    - trigger ADC calibration using ADR440BRZ voltage reference on PCB; m-labs supplied subroutine handles calibration according to instructions in Data Sheet
    - set gain (1, 8, 16, 32, 64, 128)
    - read ADC channel

jbqubit, 2017-02-13:
Anybody have comments on this? @dhslichter @dtcallcock ? 

@sbourdeauducq Does this adequately specify ARTIQ support for this IC? If so I'll close.

dhslichter, 2017-02-13:
This looks good to me.

sbourdeauducq, 2017-02-16:
This should not be a issue tracker, and especially not one for the hardware. I think the mailing list is a better place to brain storm and discuss specifications; the latter can be located in a wiki (the ARTIQ one is appropriate in this case). Again, issues are for single actionable items.

"setting/getting IC register contents by RTIO" and "no need for synchronization of acquisition with DRTIO clock" are contradictory.

Anyway, we can definitely develop such a low-level interface for this ADC. Is it enough?

jbqubit, 2017-02-16:
>  issues are for single actionable items.
Issue title changed to reflect actionable task. "specify analog mezzanine monitoring/calibration ADC functionality"

> "setting/getting IC register contents by RTIO"
ACK. RTIO is not needed for this slow IC. This post predates the discussion in #140. I've updated the leading post on this Issue. 

> we can definitely develop such a low-level interface for this ADC.  Is it enough?
Yes. I'm closing Issue as this looks resolved.

closed 2017-02-16 by jbqubit