sinara-hw/sinara #142: Slow DAC features/specs

hartytp, 2017-01-27:
Breakout issue to discuss the Kasli/VHDCI slow DAC extension board in more detail.

The following is a combination of the info currently on the Wiki and summary of our initial thoughts on what we'd like the hardware to be able to to. All numbers are pretty rough atm:

- **Channel count**: >=30, ideally ~48
- **DAC count**: use a few lower channel-count DACs rather than a single 40ch DAC to increase update rate a bit?
- **update rate**: ~10channels simultaneously  at >=100kSPS (assume SPI running at 50MHz?)
- **Output**: +-10V (ideally +-12V using +-15V supply rails -- are these generated on the extension board using switch modes, or supplied by the carrier?)
- **Resolution**: 16 bits
- **Filters/bandwidth**: 2-pole active filter, followed by 1-pole RC. ~100kHz cut-off
- **Noise/drift**: Integrated noise/drifts (+-1C lab temperature stability) <a few LSB. Noise density above 1MHz < TBD nVrtHz 
- **Connectors**: Use cable assembly with individual screening for each channel?

jordens, 2017-01-27:
I would not focus on high update rate and not try to tweak the design for that too much. The scheduling of the individual updates over SPI is really difficult if you want to do streaming. Allocating the SPI bandwidth so that something is guaranteed is complicated.
I am fine with "fewer channels". But 32 would be my goal. You can always use more boards and leave the unused channels alone to increase the sample rate.
Yes. Connectors are a challenge. They should be easy to assemble, commonly available, mechanically and electrically good. Don't know what's en vogue nowadays. VHDCI again?
@hartytp feel free to attach labels to these issues (and make labels where missing). That simplifies searching for issues a lot.

hartytp, 2017-01-27:
> I would not focus on high update rate and not try to tweak the design for that too much. 

Agreed, optimising the bandwidth definitely shouldn't be a priority for this design. But, using a few DAC ICs rather than a single one has a minimal impact on the cost/complexity of the board and gives the user a bit more flexibility.

> The scheduling of the individual updates over SPI is really difficult if you want to do streaming. Allocating the SPI bandwidth so that something is guaranteed is complicated.

Could you elaborate on that a little, please?

jordens, 2017-01-27:
Yes. I am fine with splitting the extension so that there are as many SPI buses (plus LDAC) as we can fit on the IDC connector and 32 channels in total. But with 8 signals we can only do two SPI buses. And then we don't even have LDAC yet.

Well it's a shared bus. You can not set a channel at some arbitrary time because there might already be traffic for other channels at or around that time. You always have to treat them all as a block of channels if you want to pump lots of data. And you will be dealing with contention all the time.

hartytp, 2017-01-27:
- Are we happy to use a shared SPI clock for all DACs? (I am)
- Do we want to connect the DACs MISO (readback) lines to the IDC? (Otherwise, they can just go to a test header for debugging).
- I assume we don't need CS lines for the individual DACs.

In which case, we need a single shared clock line + 2 lines/DAC (MOSI + LDAC), allowing 3 DACs per board. 3 * 16channels = 48 channels, which seems like a good number to me.

jordens, 2017-01-27:
Sharing SPI clock breaks our SPI master. That's currently a no-go for us.
Trying to save pins by demoting MISO do "debugging" is a terrible idea.

hartytp, 2017-01-27:
Okay, if you feel that strongly about it then we can stick with a single DAC IC.

> Trying to save pins by demoting MISO to "debugging" is a terrible idea.

Out of curiosity, what do you plan to use the DAC's MISO for?

jordens, 2017-01-27:
Verifying that the chip is there, verifying that writes work, reading overtemperature indicator. Sure it's not much but I don't think saving a pin or two is worth it since even the occasional debugging would require building new bitstreams/runtime, and manual intervention inside the hardware.
SYNC is CS. We could share the two LDAC and the two MISO. Then two DACs fit on one IDC.

dtcallcock, 2017-01-27:
Or just use 2xIDC for 3x16ch DAC.  That still lets you run 192ch off a single Kasli.

> Yes. Connectors are a challenge. They should be easy to assemble, commonly available, mechanically and electrically good. Don't know what's en vogue nowadays. VHDCI again?

Individual screening will tricky without going to a custom assembly (like NI does for a lot of its products).  One simple option would be HDMI - that gets you 4 individually screened pairs in a cheap, easily available cable.  The 6 connectors needed would fit comfortably on the panel (also mini connectors are available).  Reasonable breakouts already exist: https://www.adafruit.com/products/3120

dtcallcock, 2017-01-27:
>Output: +-10V (ideally +-12V using +-15V supply rails -- are these generated on the extension board using switch modes, or supplied by the carrier?)

The carrier only provides +12V.

gkasprow, 2017-01-27:
> The carrier only provides +12V.
>
>
> yes, because most extension boards require single voltage.
On DAC ones we will add simple DC/DC converter that would produce +/-15V

gkasprow, 2017-01-27:
VHDCI are difficult to solder. Usual DB or high density DB are much more common.

hartytp, 2017-01-28:
@jordens 
> Yes. Connectors are a challenge. 

Every time we build this kind of thing, we end up going for a standard D-sub and saying "next time we should really find a better solution".

> They should be easy to assemble, commonly available, mechanically and electrically good

I'd add to that list that pre-made connectors should be available commercially with a "sensible" lead time/cost (I don't want to waste time because one of us cross-wired 2 channels for a 200 electrode ion trap when they hand assembled the cabling).

@dtcallcock 
>  One simple option would be HDMI - that gets you 4 individually screened pairs in a cheap, easily available cable. 

From an electrical perspective, HDMI is pretty nice. But, for routing say 200 channels to a trap, the cabling gets to be a bit of a mess (although, not unworkable). Also, since our signals will be single-ended, routing them as screened pairs is not ideal.

> Individual screening will tricky without going to a custom assembly (like NI does for a lot of its products). 

AFAICT, none of the cables that NI offers as stock would be suitable for this design, which is a shame. I've never looked into getting custom cable assemblies made, so no feeling for what the costs would be. Might not be too bad if we placed a bulk order for everyone who's interested?

> Usual DB or high density DB are much more common.

And, high-quality screened (but not individually screened) Dsub cables are widely available, which is a plus.

Given the relatively low speed of this design, individual screening is probably not required to keep channel-channel cross-talk to an acceptable level for cable runs of a few m (we should make sure the cables are driven from a nice low impedance though).

So, unless anyone (@gkasprow @sbouhabib) can suggest a better alternative, let's stick with DB.

For 48 DAC channels, I'd suggest either going for 2xD25 (1 GND per cable) or, if we want a bit more grounding, for 2xD37. Any preferences?

hartytp, 2017-01-28:
> Verifying that the chip is there, verifying that writes work, reading overtemperature indicator. Sure it's not much but I don't think saving a pin or two is worth it since even the occasional debugging would require building new bitstreams/runtime, and manual intervention inside the hardware.
SYNC is CS. We could share the two LDAC and the two MISO. Then two DACs fit on one IDC.

Playing devil's advocate, while I agree that it's nice to be able to read DAC registers:
- The issue isn't "saving a pin or two", it's a choice between having independent LDACs or having a (shared) MISO 
- One can verify that the board is present + powered by reading from the EEPROM via I2C, so I'm not fussed if one can't also verify that the individual DACs are present (there is still plenty that can be wrong with the board even if the DACs are responding via SPI)
- verifying that writes work is something that should only need to be done when the initial firmware is written. 
- I can count on one hand the number of times I've ever read the overtemperature indicator etc for a DAC. IME, when the DAC outputs start playing up it's generally not due to something that can be debugged by checking registers (e.g. likely to be related to an output OpAmp etc).

I would have marginally preferred to keep the LDAC lines for the two DACs separate to give the user a bit more flexibility in how the DACs are used. However, I don't feel strongly about this, as separate LDAC lines aren't required for any of the uses I have in mind (and, in any case, these boards should be cheap enough that a user can always use more boards to get more flexibility).

@dtcallcock 
> Or just use 2xIDC for 3x16ch DAC. That still lets you run 192ch off a single Kasli.

That's definitely a possibility. But, it seems nicer to keep it to 1 IDC per extension board if possible: fewer cables to make; and, makes setting up the software configuration easier.

hartytp, 2017-01-28:
Edited top-level post to include comments so far.

jordens, 2017-01-28:
Verifying that the SPI communication with the DACs works is something we would want to do in CI all the time. There will be changes to the SPI core and the drivers. There will be changes to the other layers involed in getting data to the DACs. Continuously being able to verify that these changes don't break stuff and being able to automatically identify the faulty commit is something I value very much. In all cases I can remember where convenient readback has been omitted it turned out to be the wrong decision.

hartytp, 2017-01-28:
@jordens You raise a good point -- I'm not used to thinking in terms of CI and long-term maintainability for a project of this scale. Shared LDAC and MISO it is.

gkasprow, 2017-01-28:
We can order such cables from here 

http://www.m2cables.us.com/M2-D-Sub-9-Pin-to-SMA-Fanout-Custom-Assembly-Call-p/m2cable-assembly-212.htm

or we could stick with standard DSUB-BNC SVGA cables. Just google VGA BNC

They are low cost and high performance (200MHz) and they have standard pinout. The only drawback is 5 channels. 

And they have color marking.

On a front panel we can easily fit several DB15HD connectors

You can have them for 5$ per 2meter cable.

 

Another option are CCTV cables where DSUB1HD fans out to 8 BNC, this one looks pretty standard one

http://www.dx.com/p/vga-15pin-male-to-8-bnc-female-connectors-cable-for-cctv-blue-25cm-cable-90310#.WIzzNn80968

http://www.dx.com/p/vga-15pin-male-to-8-bnc-female-connectors-cable-for-cctv-blue-white-32cm-cable-110062#.WIzzPH80968

http://www.tradekorea.com/product/detail/P310544/VGA-to-8-bnc-cable,VGA-15Pin-Male-Breakout-Cable-to-8-BNC-Female-Cable.html

http://topcctvcable.com/index.aspx?menuid=12&type=productinfo&lanmuid=48&infoid=148&language=en

 

and costs only 4$ per piece

They are used in CCTV recorders

 

we can order higher quantity from Alibaba and distribute them among interested parties.

gkasprow, 2017-01-28:
In terms of Kasli and DACs we can use single ended lines for DAC signals. 
But in case of VHDCI breakout we have much longer pairs with strong coupling between them so single-ended signalling can be used only for slow signals, after lowpass filtering and Shmidt trigger buffer.
So each LVDS can drive 8 LVDS pairs of 16 LVTTL signals in Kasli easily.
For VHDCI breakout we can also use LVTTL but only for signals that do not need high performance, i.e. DAC selection, resets, status line providing that we still meet setup and hold time constraints. 
MISO, MOSI, CLK need to be driven by LVDS transceivers.

hartytp, 2017-01-28:
@gkasprow 
> MISO, MOSI, CLK need to be driven by LVDS transceivers.

Agreed, it makes sense for these signals to be LVDS.

Thanks for the suggestions for cables to use with the (single-ended analog) DAC outputs. However, all the cables you suggested are coaxial, with SMA/BNC on one end. Given the space constraints we have in the lab, and the fact that an experiment might require 200+ DAC channels, I think we really need something with a higher density.

In the past, I've hand-assembled cables using individually screened twisted pairs (for single-ended signals, one of the pair is GND) and DB connectors on either end. As @dtcallcock points out, this is what [NI do for their DAQ cards](http://www.ni.com/datasheet/pdf/en/ds-467). IMO ideally, we'd find a solution where people can use a standard DB cable if they don't care about cross-talk, but also specify a compatible higher quality cable for more sensitive applications...

gkasprow, 2017-01-28:
In such case one can use 3M MDR (mini delta ribbon) cables.
The cables can be soldered directly to the contacts or one can use IDC version where 2 ribbons are attached simultaneously. One can use twisted pair ribbon cables as well. Special tool is required, don't try to do it at home:)
There are shielded versions of such twisted and not twisted ribbon cables from 3M.
I used MDR50 due to space constrains, but much more popular is MDR68 used in SCSI-2. One can buy such cables in various lengths.
There are popular SCSI cables with VHDCI on one end and VHDCI on another end.
National is using VHDCI68 cables but with different diff pair assignment, but for slow analog signals it won't make any difference
So we can keep NI signal assignments like here, page 16
http://www.ni.com/datasheet/pdf/en/ds-153
or page 19
http://www.ni.com/pdf/manuals/375140a.pdf
In this way one could use National terminal boxes
http://sine.ni.com/nips/cds/view/p/lang/pl/nid/10175
It's better to keep any suitable standard than no standard at all.
Just to make it difficult to mix the connections, for analog signals let's use MDR68, for digital LVDS - VHDCI. In this way there are several options for DAC outputs:
- direct fanout with individual wires connected to MDR68 contacts
- custom made cable with IDC MDR68
- SCSI cable with 2 MDR68 connectors and custom terminal box 
- SCSI-VHDCI cable or shielded NI cable with National terminal box

hartytp, 2017-01-29:
@gkasprow Thanks for posting that. I haven't used MDR before, but they look to be exactly what I was after.

Looking on Digi-Key/Mouser, the 50pin connectors seem to be the most common. So, what about going for 40ch per DAC board (assuming you'd be happy putting that many DAC channels + active filters etc on a single board) with 2 x MDR50 connectors, each supplying 20 DAC channels? Pin out per [3M 14X50-SZWB-XXX-0NC series cables](http://multimedia.3m.com/mws/media/297587O/3mtm-mdr-cable-assy-050-hispeed-video-data-trans-ts2103.pdf)? Alternately, we could put 48 DAC channels down a single 141A0-SZ5B-XXX-0DC MDR100 cable, if people would prefer that (assuming the cabling doesn't get too bulky/inflexible with that many conductors)? Or, if you have a good source of MDR68 cables, I'd be happy to use them instead.

Digi-key sells high-quality cables with individual screened twisted pairs for ~300-200USD per cable depending on length + conductor count. There are a few of the 50-way assemblies in stock, otherwise lead-time is 9 weeks. 

For those looking for a cheaper alternative, IDCs are widely available http://uk.rs-online.com/web/p/scsi-connectors/1633336/ (unless @gkasprow you think that these aren't easy to hand assemble without special tooling?) Or MDR50 SCSI cables can be ordered in bulk (100) from Ali for ~10USD each ( @gkasprow do you know a supplier for cheap MDR SCSI cables in smaller quantities?).

On the 3M 14X50-SZWB-XXX-0NC we could consider using some of the unshielded pairs for +-15V to power any low-power active filters etc on the Rx end. For the 141A0-SZ5B-XXX-0DC we could re-purpose some of the 4 auxiliary GND conductors for this (assuming we don't mind breaking the standard slightly).

gkasprow, 2017-01-29:
It is possible to assemble ribbon cables manually but you need non-standard ribbon cables (thin ones)
With standard ones it's extremely hard to  clamp them using vice.

gkasprow, 2017-01-29:
The SCSI-2 and MDR68 connectors have the same footprint so one can assemble
one or another on the PCB.

hartytp, 2017-01-30:
@gkasprow 

1. What pin-out were you thinking of using with the MDR68 connectors? e.g. how many DAC channels/GNDs were you thinking of connecting to the cable?
 
2. It would be nice to keep the option open of using cabling which provides screening of individual channels. For MDR50/MDR100 I could find cables like [141A0-SZ5B-XXX-0DC](http://multimedia.3m.com/mws/media/297587O/3mtm-mdr-cable-assy-050-hispeed-video-data-trans-ts2103.pdf) as standard items from Digi-Key/Mouser. But, I couldn't find an equivalent for MDR68. Did I miss one? Otherwise, is there any reason not to go for MDR50/MDR100, as they both seem to be widely available and supported?

hartytp, 2017-01-30:
> The SCSI-2 and MDR68 connectors have the same footprint so one can assemble one or another on the PCB.

To check I understand the terminology: by "SCSI-2 connector", do you mean MDR50? So, are you suggesting that we design the board to work with either MDR50 or MDR68 connectors? (so in the MDR68 case, one either gets slightly more DAC channels, or we just ground the unused conductors?)

gkasprow, 2017-01-30:
I meant only SCSI cables that has connectors which share footprint with MDR68. 

They do not have shields, just twisted pairs.

Let’s go for MDR50/MDR100.

MDR100 would be hard to fit on 3U panel and are not handy.

Greg

hartytp, 2017-01-30:
Okay, so how about we go for:

- 2xMDR50 per extension board
- MDR50 pin out per 14X50-SZWB-XXX-0NC
- 20 DAC channels per cable using the 20 twisted pairs, with 1 conductor for each pair being GND
- 3 unscreened twisted pairs can be used to supply +-15V, +5V for low-power circuitry (active filters etc) on the Rx end (assuming there is room for this in our power budget)

That gives us 40channels total per extension board, so we can go for:
1. 1x40 channel DAC
2. 2x32 channel DACs (wasting 24 DAC channels)
3. 1x32 channel DAC & 1x8 channel DAC

IMO, we should go for options 1 or 2, since DAC channels are cheap and asymmetry is nasty. I'd generally prefer option 2, but don't feel overwhelmingly strongly about it.

Thoughts? Objections?

hartytp, 2017-01-30:
Thoughts on low frequency (<1kHz) noise and thermal drift...

Physicists goal is for a 1MHz trap frequency to be stable to a few Hz. This requires DAC noise/drifts to be <10ppm RMS (1LSB RMS for 16 bits), so:

- Will need a decent reference
- Will need to be careful over temp-co of gain-set resistors in active filters
- May need screening to prevent rectification of RFI
- Assume lab air temp is +-1K, still need to be careful over thermal management for this board. As it could dissipate a few W, we may well need forced air cooling.

jordens, 2017-01-30:
@hartytp Why not then just do one 40 channel DAC per extension, and have those who need faster channels grab two extension cards and waste the channels there? No adapter board needed. And it makes the design less dense, power hungry and cutting fewer corners (SPI pin sharing etc.). It seems weird that because of the connector choice we are now trying to balance wasting resources. I'd much rather have a less densely populated PCB right now to have some room for all the additional bells and whistles that will be needed as opposed to squeezing every possible use case in there.

hartytp, 2017-01-30:
@jordens Okay.

2 DACs is a "nice to have", but I agree that it doesn't make as much sense any more, given our wiring choice.

> No adapter board needed.

Which adapter board?

gkasprow, 2017-01-30:
SCSI-2 use 68-pin D-type connector like this

http://www.digikey.com/product-detail/en/te-connectivity-amp-connectors/5787190-7/A33515-ND/1121382

which has the same footprint like MDR68

http://www.digikey.com/product-detail/en/3m/N10268-5242PC/N10268-5242PC-ND/1091938

But they are not compatible.

 

So we can have 2 options – either install MDR68 connector and use custom cables or install D 68 SCSI-style connectors and use standard, low cost SCSI cables which are twisted pairs like VHDCI. 

There are standard SCSI cables that have VHDCI on one side and D-68 on the other side.

We can also install MDR50 and use 3M cables

hartytp, 2017-01-30:
@gkasprow Thanks for clarifying. Top post edited to summarise these points.

gkasprow, 2017-01-30:
@hartytp 
HD68 (SCSI2) and MDR68 have exactly same footprint and panel cut-out. They both have 68 pins but are not compatible. 
MDR50 are smaller and have 50 pins

jordens, 2017-01-30:
@hartytp 
>> No adapter board needed.

> Which adapter board?

My fault. I was thinking wrong.

hartytp, 2017-01-30:
> HD68 (SCSI2) and MDR68 have exactly same footprint and panel cut-out. They both have 68 pins but are not compatible. MDR50 are smaller and have 50 pins

Yes, sorry, I misread your post (end of the day...). 

> So we can have 2 options – either install MDR68 connector and use custom cables

How much would this kind of custom cable cost, and who would be a supplier?

gkasprow, 2017-01-30:
There are also SCSI cables with MDR50 and MDR68 connectors, called high density or micro- Centronics connector.
But they are not very popular.
The most popular ones are HD50 and HD68 SCSI cables.
So it makes a lot of sense to use MDR50 connectors.
We can then use high performance 3M  MDR50 cables mentioned before (68pin versions are outdated) or 50pin SCSI cables with HD-50 connectors.
I use MDR50 connectors in these projects
http://www.ohwr.org/projects/fmc-adc100m14b16cha-dac2m14b4cha/wiki
http://www.ohwr.org/projects/fmc-adc100m14b16cha-dac2m14b4cha/wiki

hartytp, 2017-01-31:
@gkasprow Okay, thanks for clarifying further.

Summary as I understand it:

- either MDR50/HD50 or HD68 (SCSI) would be acceptable choices
- In general, HD68 would be nicer to use than MDR50 because:
  - They are very common, and bulk-screened cables are readily available
  - They give a higher density than MDR50 and 34 pairs is a convenient number -- e.g. we could use 32 pairs for a 32ch DAC, and still have two pairs left for +-15V. This increased density is likely to be nice for experiments requiring 200+ DAC channels
- The only downside of HD68 I can see is that cables with individually screened pairs aren't readily available for HD68, which could lead to cross-talk issues. 

I don't have well-calibrated intuitions for how bad the cross-talk would be in, say, a 5M SCSI cable driving high-impedance loads single ended with 100kHz sinusoids. My guess is that it's not going to be a problem, but unless we have some data on this, I'd like to keep the individually screened cables open as a "plan B". I'll have a think about this, and maybe make some measurements in a week or two when I have time...

gkasprow, 2017-01-31:
We can always use differential drivers that can be used either as single-eneded or differential.
So they can give +/- 20Vpp in diff mode or +/-10V in single ended mode.
In this way we can greatly reduce crosstalks.
Another option is to use twisted pair with one line grounded on both sides.
Yet another idea is to drive diff line in single ended mode and add common mode choke on both ends. In this way for DC we use only one line and ground, but the higher frequency is, the more differential transmission becomes. The CMC on remote side makes signal single-ended again. This is simple and reliable solution but requires grounding of the remote side.

hartytp, 2017-01-31:
> Another option is to use twisted pair with one line grounded on both sides.

Twisted pair with one ground per DAC channel sounds like a good idea.

I'm happy to connect the grounds on both sides of the cable  (ie to connect the Kasli/DAC ground to the Rx/ion trap ground). My reasoning here is that, so long as we keep the Kasli + DAC boards reasonably cheap, ground loops are avoided by driving different grounds from independent Kaslis/DACs.

> We can always use differential drivers that can be used either as single-eneded or differential.

Differential drivers are a good approach wrt signal integrity, but can be a bit inconvenient as they require active circuitry on the Rx (ion trap) end for differential to single-ended conversion. Space around the ion trap is generally at a premium, so we normally use only a small passive filter/break out PCB to connect the cables to the ion trap. However, if we produce a sufficiently compact receiver board, this could be okay.

To avoid unnecessarily overcomplicating the design, I'd prefer to avoid using differential drivers unless required. A rough target for cross-talk would be:
- <-60dB worst-case cross-talk between pairs
- 1MHz sinusoid
- 5m cable length
- One conductor of each pair grounded at both ends of cable
- Source impedance <<1k, load impedance >~1kOhm

If standard SCSI cabling exceeds that*, then this is all a non-issue anyway IMO.

* I assume it does -- comfortably -- but don't have any data to back that up...

gkasprow, 2017-01-31:
We use standard SCSI cables for analog signals up to 100MHz but we send them in differential form and use CMCs.

We can add common mode chokes. IF you don’t want it, just short negative to ground. If you want – just install another CMC on the end.

CMC price is low <1$  <https://github.com/notifications/beacon/AEH-vt3stoZzrtgZcroo1Cz1_rMFqeUEks5rX1tIgaJpZM4LvzLa.gif> 

http://pl.farnell.com/bourns/srf0905-652y/common-mode-choke-6500uh-300ma/dp/1791514

it has 6.5mH for single ended signals, 0.39uH for diff ones and works well up to 2MHz.

Can be easily used for common mode filtering on both sides.

hartytp, 2017-02-01:
Okay, let's go for SCSI cables and CMCs.

jordens, 2017-02-01:
For the DAC output filtering, I would not like to see much active filtering in on the PCB. A good passive filter to clean up the ~MHz band: absolutely. But active filters are not that good at rejecting above MHz and can themselves easily dominate the noise there. Also the filtering requirements tend to be very different and custom. I am fine with an active filter that can be integrated for free into the gain stage that gets us to +-10V but I would not add extra active components. We do have a frontend power supply on the output connector so that people can apply their own active filtering, right?

gkasprow, 2017-02-01:
@jordens 
 CMC is completely passive EMC filtering device, invisible for signal.
Later on, user can add additional filter.
The most popular CMC you can fin on every DC, HDMI or USB cable.
http://www.fair-rite.com/wpapers_anotes/ferrite-cores-for-low-frequency-emi-cable-suppression-2/
we will use discrete version of it.

jordens, 2017-02-01:
@gkasprow Yes. The EMC filtering on the cable is all fine and a good idea. I was more criticizing the per-channel active lowpass that Tom had suggested to put on the board between DAC and connector/cable.

hartytp, 2017-02-01:
@jordens 

> I am fine with an active filter that can be integrated for free into the gain stage that gets us to +-10V but I would not add extra active components. 

Agreed. My assumption is that we should get active filtering more or less for free using buffer/gain amps that will be required anyway. If the amps are there then we might as well at least have pads so that they can be used as active filters if one wants.

> A good passive filter to clean up the ~MHz band: absolutely. But active filters are not that good at rejecting above MHz and can themselves easily dominate the noise there. 

Yes, so we'll want pads for a passive filter after the active one.

----

I'm happy to remove the explicit reference to active filters from the specification if you'd prefer. It's probably better to keep this higher-level anyway: we'll specify the noise we want, including LDAC/CLK/etc spurs. If that can be done passively then I'm happy.

hartytp, 2017-02-01:
**Broad-band noise**

- Preliminary specification: <1nVrtHz for all frequencies >~1MHz
- Includes both white noise from OpAmps etc and spurs from SMPSs, LDAC, DAC_CLK etc.
- Based on heating rate calculation for Sandia HOA2 (calculations by @cjbe):
  - Heating rate previously measured to be 30quanta/s on a 2MHz radial mode, expect 200quanta/s for 43Ca+ on a 1MHz axial mode. 
  - Aim for heating rate due to DAC noise on electrodes to be an order of magnitude below this
  - Applying 1nVrtHz of uncorrelated white noise to all electrodes gives 40quanta/s heating rate. Thus, want a max noise of ~0.7nVrtHz
  - Results for correlated noise depend on assumptions about the relative amplitudes and phases of the coupled noise, but the above gives a reasonable target to aim for.

hartytp, 2017-02-01:
I've updated the top post to take into account all feedback so far, and filled in the remaining question marks.

This "specification" is now complete in all aspects that I care about. If anyone else has additions/objections/changes, please post them. Otherwise, I'll close this issue and update the Wiki...

hartytp, 2017-02-03:
One final thought about this:

- to make this design as useful as possible, we should also provide a convenient way of breaking the HD68 out to something lower density. There are lots of ways of doing this, and I really don't mind which we go for, so long as it doesn't rely on obscure, hard to obtain connectors/cables.
- One option is to break the HD68 out to 4xDE-15 (aka DB15HD/VGA) connectors, each carrying 8 DAC channels (7 GNDs). The readily available CCTV adapters that @gkasprow mentioned above can then be used for further breakout to BNC if desired (NB VGA to 5xBNC is more standard than VGA to 8xBNC, but less convenient).
- Various ways of providing the HD68 to DE-15 breakout (I'm not aware of conveniently available adapters/cables to do this):
  1. Provide a separate adapter/breakout board + suitable enclosure to mount it in
  2. Can we fit 4xDE-15 connectors on the same 3U front panel as the HD68 (so each signal connects to both) without overcrowding it too much? DE-15 could be IDC connectors, which connect to 100 mil header on the PCB via ribbon cable.

gkasprow, 2017-02-03:
The easiest seem to be simple plastic box with 2 panels (easy to laser-cut) and PCB where HD68, BNCs, VGA, DSUBs and some CMC can be installed. Just specify what you need on the other side.

gkasprow, 2017-02-03:
Such PCB or many of them can be also attached to 1U panel. Only mounting fixture need to be foreseen.

hartytp, 2017-02-03:
> The easiest seem to be simple plastic box with 2 panels (easy to laser-cut) and PCB where HD68, BNCs, VGA, DSUBs and some CMC can be installed. Just specify what you need on the other side.

PCB with 1xDB68 as input and then 32x(CMC +  BNC) for output sounds good to me.

> Only mounting fixture need to be foreseen.

Ideally something can can be mounted in a 19'' rack, but can also be used conveniently on a desk.

gkasprow, 2017-02-03:
Ok, just add these requirements to wiki

dtcallcock, 2017-02-03:
Another option would be to also make a lower density version of the DAC extension with the 8 channel version of the DAC chip (AD5362) and BNC outputs.

hartytp, 2017-02-03:
> Another option would be to also make a lower density version of the DAC extension with the 8 channel version of the DAC chip (AD5362) and BNC outputs.

I did consider that. The reason I initially decided against it was that it would mean yet another piece of hardware to build, test & maintain. 

One possible way of mitigating that concern would be:
- use the same PCB for both "high-density" and "low-density" versions of the board, including the same 32 channel DAC in both cases (IMO, the cost difference between 8 & 32ch DACs isn't sufficient to warrant a different PCB design)
- for the low-density version, leave all buffers, filters etc DNF for 24 channels.
- Put both 68-way pin-header on the PCB and pads for 8xBNC on the PCB.
- For the low-density version, populate the BNCs, and use the same FP (front panel) as for the ADC board
- For the high-density version, use a FP with IDC HD68, which connects to the PCB via ribbon cable.
- Add a jumper/pull-down resistor so that Kasli/Artiq knows which population option we're using.

hartytp, 2017-02-03:
Or, in a similar vein:

- Keep the HD68 footprint on the PCB
- Also add 4xIDC, each containing 8 DAC channels to the PCB
- Make a simple (passive, straight-through) IDC to 8xBNC breakout board and accompanying 3U front panel
- Then, one has two options:
  a. Populate HD68 + all output channels (filters, buffers, etc) etc
  b. Do not populate HD68. Populate nx8 output channels and connect n IDCs up to breakout boards
- IDCs can have an extra "discovery" line, that is pulled low by the breakout board. That way, Artiq knows what the current population option is. 
- In this scheme, the breakout boards should be designed to mount directly to the front panel. Ideally, if the HD68 is not connected, a breakout board should mount to the same FP as the DAC (i.e. try not to waste rack space).

How does that sound? Friday evening BS?

gkasprow, 2017-02-03:
We can have both HD68 and IDCs. It is not gigabit signal so no issues will
arrive.
So the user would have choice to use some channels directly via local BNC
adapter and remote HD68

hartytp, 2017-02-03:
> We can have both HD68 and IDCs. It is not gigabit signal so no issues will arrive.

I'm definitely not worried about signal integrity for this!

The concern is more about usability:
- If we only want to use, say 8 channels, then we might want to reduce cost by not populating unused amps/filters/etc (although, if cost is sufficiently low, this isn't an issue). But, it could cause confusion for the users if a HD68 is present, but some pins are NC
- it could be confusing for a user if the same DAC channel can be accessed using both the HD68 and BNC -- particularly if the BNC is in a separate 3U panel, and so has the appearance of being a separate piece of hardware to unfamiliar users! This hardware has to be used by new physics graduate students, who aren't familiar with either Kasli or basic electronics. So, we want to keep things intuitive to use as far as possible.

So, my thinking was that when the BNC breakout is used, it might be better not to fit the HD68 (obviously, the footprint remains in place). This means we would need 2 different 3U front panels: 1xHD68, and 8xBNC

dtcallcock, 2017-02-03:
I much prefer the idea of adding IDC headers for BNC breakout to putting the HD68 on a ribbon cable.

As for "discovery", isn't that what the EEPROM is for?  That would hold the information on whether the HD68 was populated, and if not, how many buffer channels were populated when the board was stuffed.  I don't think the hardware needs a way of detecting how many straight-through IDC-BNC adaptors are in place any more than it needs to be able to auto-detect how many BNCs are plugged in.

hartytp, 2017-02-03:
> I much prefer the idea of adding IDC headers for BNC breakout to putting the HD68 on a ribbon cable.

Me too. 

> I don't think the hardware needs a way of detecting how many straight-through IDC-BNC adaptors are in place any more than it needs to be able to auto-detect how many BNCs are plugged in.

True.

jordens, 2017-02-03:
I'd also go for having footprints for both IDC and HD68, always populate everything. Then you can choose which frontpanel and connector you want: either HD68 with a very narrow panel or a big panel with your choice of panel mount BNC/SMA connectors and (shielded or unshielded) pigtails that connect them into an IDC connector.

[13 more replies]

closed 2017-02-06 by hartytp