sinara-hw/sinara #143: allaki: AD8363 RF power input pin

jbqubit, 2017-01-27:
AD8363 has two alternate input paths for RF power INLO and INHI. Spec sheet says.

> Using INHI (Pin 14), users can obtain the best dynamic range at frequencies up to 2.6 GHz. Above 2.6 GHz, it is recommended that INLO (Pin 15) be used.

For the target frequency range of Allaki input should be INLO.

dtcallcock, 2017-02-08:
@jbqubit I thought the target frequency of Allaki was the first Nyquist zone?  If so, INHI is correct (also correct for second Nyquist zone).

jbqubit, 2017-02-13:
>If so, INHI is correct (also correct for second Nyquist zone).

Yes. I see. The current allaki_mezzanine schematic is indeed correct. The labeling of the pins in the product documentation is counter-intuitive. Closing issue. 

>INHI This is the RF input pin for frequencies up to and including 2.6 GHz. The RF input signal is normally
ac-coupled to this pin through a coupling capacitor. See Figure 34

>INLO This is the RF input pin for frequencies above 2.6 GHz. The RF input signal is normally ac-coupled
to this pin through a coupling capacitor. See Figure 34

closed 2017-02-13 by jbqubit