sinara-hw/sinara #144: PCB_mezzanine_analog_allaki_allaki: ARTIQ board support

jbqubit, 2017-01-27:
PCB_mezzanine_analog_allaki_allaki has several IC components that are interactive or configurable. 
- HMC542BLP4E variable attenuator (set via TTL)
- HMC849ALP4CE RF switch (set via TTL)
- AD8363ACPZ-WP RF power meter (readout via ADC on PCB_sayma_RTM)
- AD8253 programmable gain op amp (gain set via TTL)

We should all get on the same page about what level of support will be build-in to ARTIQ. 

# configuration
I assuming a couple things. 

1) The AMC-RTM FPGA link is in place. Something along the lines of #140.  

2) At a high level the users interacts with the analog mezzanine IO as either TTL PHY or SPI PHY. 

3) Calibration ADC is implemented as in #141. 

4) We don't care about detail of temperature trimming of AD8253 (see below).

# configuration
If (1-3) are in place that's enough in my view to use ARTIQ TTL PHY's for active configuration of the PCB. Other details of operation are left up to the end user (eg calibration). 

# AD8363 temperature compensation

AD8363 has two analog inputs (TCM1 and TCM2) used to compensate for
temperature-dependent effects in power-sensing. AD8363  has an
internally generated VREF which can be divided down and used for TCM1
and TCM2. 

As currently wired up in the schematic it is optimized for RF frequency of 3.8 GHz and large temperature swings.  
- TCM2 driven at 2.3 V x 10/(13+10) = 1.0 V
- TCM1 driven at 2.3 V x 1/(1+3) = 0.575 V

If operated near 25C TCM1 and TCM2 don't matter much regardless the input frequency.

dtcallcock, 2017-02-04:
> TCM1 driven at 2.3 V x 1/(1+3) = 0.575 V

You need to take into account the 5k input impedance of TCM1 (TCM2 is a negligible 500k).  Also, the potential divider is 1k and 3.24k, not 1k and 3k.  This gives TCM1 of 2.3 V x 0.833/(0.833+3.24) = 0.47 V.  This is optimised for 100MHz.

jbqubit, 2017-02-07:
@dtcallcock Thanks for pointing that out. I was misreading the schematic. 

@gkasprow In the schematic for AD8363 the resistors read  "1k5" and "1k4". AFAICT this is odd nomenclature. Does your schematic package support default formatting as "1.5k" and "1.4k"?

hartytp, 2017-02-08:
>  In the schematic for AD8363 the resistors read "1k5" and "1k4". AFAICT this is odd nomenclature. Does your schematic package support default formatting as "1.5k" and "1.4k"?

@jbqubit "1k5" is pretty standard (and good) notation, designed to avoid mistakes due to not seeing a decimal point on printed schematics. AFAICT it's used pretty consistently across the schematics in this project (at least for resistors), so why would you want to change it?

gkasprow, 2017-02-08:
This is CERN library database that provides such formatting. It seems that it is European style:)

jbqubit, 2017-02-08:
​If that's the standard for CERN I'm fine with it. Now I know. :)

Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

jbqubit, 2017-02-13:
> If (1-3) are in place that's enough in my view to use ARTIQ TTL PHY's for active configuration of the PCB. 

@sbourdeauducq Do you concur? If you concur I'll close this Issue and focus on 1-3.

sbourdeauducq, 2017-02-13:
Contrary to what you say, the HMC542BLP4E is set via SPI not TTL, but we also have a gateware PHY for SPI, as you know. The AD8253 interface could also use some optimization instead of taking three generic TTLs, but that's not mandatory.

Anyway, generally speaking, this should work as far as the gateware is concerned, but someone has to look up where each pin is connected, dig into datasheets, write nice Python functions that wiggle them, write device database entries, test everything, go through a few rounds of yak shaving, write documentation, etc.

jbqubit, 2017-02-13:
>the HMC542BLP4E is set via SPI not TTL
Thanks for catching that. I've updated the leading post. 

> write nice Python functions that wiggle them, etc
OK. #139 updated. Closing Issue.

dhslichter, 2017-02-13:
Don't forget that these boards will need I2C support for the identifying EEPROM as well as the temperature sensor.

jbqubit, 2017-02-13:
@dhslichter Thanks for the reminder. I2C is included in #140 but I forgot these IC's on Allaki. Now added to leading post on this Issue. 

>- 24AA02E48T-I/OT EEPROM (readout via I2C)
>- ADT7420 temperature sensor (readout via I2C)

jbqubit, 2017-02-13:
Proposed analog mezzanine IO configuration for the RTM-side FPGA is in Issue #140.

sbourdeauducq, 2017-02-14:
> Not sure to what extent the existing I2C protocol for KC705 can be ported over for this purpose

The current I2C protocol code can be used as-is, but the I2C protocol itself isn't as much as an issue as what to do with the contents of the EEPROMs.

jbqubit, 2017-02-15:
Proposed baseline functionality for ARTIQ interface with allaki. 
- HMC542BLP4E variable attenuator (set via SPI)
    - get and set 6 bit attenuator value via SPI
- HMC849ALP4CE RF switch (set via TTL)
    - trivial TTL toggle
- AD8363ACPZ-WP RF power meter (readout via ADC on PCB_sayma_RTM)
    - analog readout using slow ADC on PCB_Sayma_RTM
- AD8253 programmable gain op amp (gain set via TTL)
    - trivial TTL toggle of two gain bits 
- 24AA02E48T-I/OT EEPROM (readout via I2C)
    - no support from m-labs for v1.0 of board support 
- ADT7420 temperature sensor (readout via I2C)
    - one-shot 16-bit temperature readout: automatic shutdown to low-power state after 240 ms readout cycle
    - no support for fault queue, set points, software reset

dhslichter, 2017-02-15:
ADT7420 -- to clarify, read/write to configuration register, read status and temperature registers.  We also want to have the software reset (very simple to implement).  No support for set point or ID registers.

jbqubit, 2017-02-16:
@dhslichter I updated the leading post to this Issue inclusive of your comment. 

@sbourdeauducq If you concur that this is adequately specified please close Issue.

jbqubit, 2017-02-21:
I'm closing this Issue.

closed 2017-02-21 by jbqubit