sinara-hw/sinara #145: Kasli/3U/PCB_BNC schematic & layout review

jbqubit, 2017-01-31:
Design concept is on the [wiki](https://github.com/m-labs/sinara/wiki/Kasli). Move discussion of schematic and layout from [here](https://github.com/m-labs/sinara/issues/99#issuecomment-276042586) to this Issue. 

Changes: 
- please make a key in the corner of the schematic for the different IC functions for quick reference
    - 24AA02E48T is EEPROM
    - TPS62175DQC is voltage regulator
    - SI8651BB is 150 MBPS isolator
    - SN74BCT25245 is 25-ohm bus transceiver
- R2, R4, ... should be 50 Ohm

Comments: 
- power supply caps look good for large current draw of SN74BCT25245 
- LEDs are a nice :) 
- ground plane isolation looks good

gkasprow, 2017-01-31:
@jbqubit 
If you want to have 50Ohm termination on each input, there should be switchable termination.
The BCT series chip won't be able to drive both 50Ohm termination to GND and 50Ohm load.
I will add then MOSFET and 47Ohm termination to GND when input mode is selected

jbqubit, 2017-01-31:
The most common signal source/sink in most lab is 50 Ohm coaxial. 

A Port can source and sink much more current than Port B. Why isn't Port A outward facing?

gkasprow, 2017-01-31:
Indeed, the ports should be reversed.
Port A can deliver still TTL high level (2.1V) at 80mA of output current.
It means that load resistance cannot be lower 26.5Ohm.
So if we use 50R termination and 50Ohm load then we are in forbidden region
of TTL inputs.
The original circuit was borrowed from a module where termination was
switched and that's why it worked.
So we will switch ports and add switchable termination.

dtcallcock, 2017-01-31:
Not every digital signal in the lab one may wish to sample is capable of driving 50Ohm.  Could the switching of the termination be separate to the direction?  There are spare outputs on the PCF8574 to do this but IC1 & IC4 would have to be switched to SI8650 (5 inputs on VDD2 side) to give an extra isolated line for this control. There could be some logic that prevents both 'output' and 'terminated' being selected at the same time.  

Alternatively a jumper could be added to disable the termination.

dtcallcock, 2017-01-31:
Also, could a series termination resistor be provided?  This would be populated with 0R (or jumpered) for driving 50Ohm loads and 47R for absorbing reflections when driving high impedance loads over a long 50Ohm coax cable (the usual scenario in the lab).

gkasprow, 2017-02-01:
For termination control, I can add low cost optocoupler.

Do you want to have it individually or per 4 channels?

gkasprow, 2017-02-01:
Series termination works only if cable output is opened.

If you have some capacitive load it won’t really work well.

The driver has already 25Ohm output resistance so will dump reflections to some extend.

hartytp, 2017-02-06:
@gkasprow 

Re 3U_BNC_PCB: this board looks really nice, good job!

The only thing that concerns me a little is that there is no visual indication (e.g. front panel LED) or software read-back of the termination state. 

I can imagine this leading to a lot of confusion in the lab: an experiment will often have several graduate students working on it, who may not know much about electronics. It would be easy for two BNCs to be switched around by someone who isn't aware of the termination settings, causing confusion.

There are several options for fixing this. One possibility would be to make the terminations switch in groups of 4, and then use tri-colour LEDs on the front panel rather than the current bi-colour ones (green=out, orange=in high-Z, red=in 50R). It looks like software readback might be possible using IC28/IC11 B5/A5 (AFAICT DIR1_RD and DIR2_RD are not currently connected to anything). Software control of the termination would also be nice, but it doesn't look like there is a convenient way of doing this without adding an extra isolator, so it's probably not worth it.

What do you think?

gkasprow, 2017-02-06:
The board is in production now, so we will add this feature in next release.

We can have MOSFETs switching termination resistors and LEDs on the front panel. In this way one would know which BNC is terminated while connecting the cable. Termination readout would have any practical application?

Of course, we can add switchable termination in group of 4 with visual signalization. This would need additional isolators.

hartytp, 2017-02-06:
> The board is in production now, so we will add this feature in next release.

> We can have MOSFETs switching termination resistors and LEDs on the front panel. In this way one would know which BNC is terminated while connecting the cable. 

That sounds good.

> Termination readout would have any practical application?

No, it's not really necessary.

gkasprow, 2017-02-06:
OK, so please add it as an issue to fix:)

.  <https://github.com/notifications/beacon/AEH-vnc3-xYTu4PzqEy33PkHeQlxlqPcks5rZz-6gaJpZM4Ly9LK.gif>

hartytp, 2017-02-06:
Closing this issue, since the boards have been sent for manufacture.

closed 2017-02-06 by hartytp