sinara-hw/sinara #146: Kasli/3U/PCB_VHDCI_breakout schematic & layout review

jbqubit, 2017-01-31:
Design concept is on the [wiki](https://github.com/m-labs/sinara/wiki/Kasli). Move discussion of schematic and layout from [here](https://github.com/m-labs/sinara/issues/99#issuecomment-276042586) to this Issue. 

Comments: 
- please make a list of IC names and functions in corner of schematic for easier reading
- is I2C power-on reset OK with the line pulled to 3.3V all the time?
- label on PCB next to each ribbon header for LVDS1, ... ,LVDS8 
- label on PCB giving names to VHDCI connectors eg VHDCI_A and VHDCI_B (add same labels to PCB_Metlino, and Metlino front panel)
- label on front panel indicating 
    - VHDCI_A drives LVDS1...LVDS3
    - VHDCI_B drives LVDS4...VLDS8
- indicate on schematic maximum current for each ribbon from P12V0
- indicate on schematic current expected from DC jack

dtcallcock, 2017-01-31:
Should we define a backplane connector pinout (see #129) and apply it or wait for a future revision?

gkasprow, 2017-01-31:
We first have to plan FPGA io utilization

We have 5 identical banks.

One bank would be occupied by SDRAM

Then 3 banks are occupied by SFP control, clocks, I2C, config pins

So we have only one bank left with 24 LVDS pairs. And these we can route to the  backplane.

What backplane topology do you want to have?

What kind of boards you want to plug.

DIN connector is not suitable for gigabit signal transcef, it works quite well up to 200..300MHz with LVDS signals and point to point configuration.

We can also run MLVDS signals over it but then one have to limit speed to 100MHz. The benefit is that we can run all 24 pairs across the backplane + some dedicated I2C ones.

gkasprow, 2017-01-31:
Oh, I didn’t upload the recent version where most of these things are implemented….

Michal will update all of them and let you know.

hartytp, 2017-02-08:
Closing this issue as the boards have been sent for manufacture (create a new issue for comments on the manufactured boards).

closed 2017-02-08 by hartytp