sinara-hw/sinara #149: SFP/SATA/FAT_PIPE usage

jordens, 2017-02-01:
(from #129)
For Metlino, Sayma, and Kasli, we need to do some radical distillation on designated use cases of the SFP cages, the SATA connectors, the FAT_PIPE links, and the FMCs. Proposal:

* Metlino
  * SFP123: all DRTIO downstream 
  * FABRICD: all DRTIO downstream
  * FABRICE:
  * FMC
    * DP0123: FMC 4xSFP): DRTIO downstream
    * LA: FMC DIO32: TTL IO
* Sayma:
  * SFP1: DRTIO downstream
  * SFP2: DRTIO downstream
  * SATA1: DRTIO upstream (different bitstream probably)
  * SATA2 (swapped): DRTIO downstream
  * FAT_PIPE1 (FABRICD): DRTIO upstream
  * FAT_PIPE2 (FABRICE):
  * FMC:
    LA: FMC DIO32: TTL IO
* Kasli:
  * SFP1: Ethernet in stand-alone, DRTIO upstream otherwise
  * SFP2: DRTIO downstream
  * SATA1: DRTIO downstream
  * X:

jbqubit, 2017-02-06:
Thank you for starting this conversation. 

## Metlino
- Fabric E transceivers are available for future applications. 

## Sayma 
- Fabric E transceivers are available for future applications. 

## Kasli 
> SATA1: DRTIO upstream (different bitstream probably)
SATA2 (swapped): DRTIO downstream

What do you mean by the parentheticals? 

Following discussion in #148. 
- Let c1 be 8 of TTL_PHY
- Let c2 be 4 of TTL_PHY and 1 of SPI_PHY
- Let EXTn be 8 LVDS pairs
- Gateware supports EXT0... EXT7 individually configurable as c1 or c2 

I'm a fan of Kasli GTP4 for SFP3. 

> EXT8 (backplane Kasli): ?

What do you mean by the parenthetical? If there's extra backplane lines seems like using them for EXT4...EXT7 makes sense.

jordens, 2017-03-29:
> SATA1: DRTIO upstream (different bitstream probably)

We currently have exactly one DRTIO upstream on a DRTIO satellite.

> SATA2 (swapped): DRTIO downstream

As discussed elsewhere, that connector has TX/RX swapped from the usual computer perspective.

> EXT8 (backplane Kasli): ?

If Kasli can support more than eight EEMs, they would be wired to the backplane. Described in the wiki.

jordens, 2017-03-29:
entered into the wiki

closed 2017-03-29 by jordens