sinara-hw/sinara #150: High efficiency AOM amp

hartytp, 2017-02-02:
We're planning to design + build a multi-channel AOM amplifier based on [NPA10003QA](https://cdn.macom.com/datasheets/NPA1003QA.pdf). The target here is Doppler cooling AOMs etc, which don't require great linearity or noise, but where high-efficiency is desirable to minimise the lab heat load (e.g. most MCL 2W amps take ~100W). 

Rough target is:

- 19 inch rack module (2U?)
- ~10 channels
- Mains powered
- Target power efficiency close to 50%
- G=30dB
- P1db > 33dBm
- Cross-talk < -60dB (-80dB preferred)

If anyone would potentially be interested in this, then post here and we can discuss further. Otherwise, I'll close this issue and keep this in-house.

jordens, 2017-02-02:
Nice! Specs look good. Add a bit of monitoring (T,I) if you can. And I wouldn't be afraid to go 3U. If you get above 30% wall plug efficiency at 1W I'd be content.

hartytp, 2017-02-02:
@jordens
>   Add a bit of monitoring (T,I) if you can

Sure. Any preference as to interface standard/connector to use for this?

> And I wouldn't be afraid to go 3U

Yes, that will probably be required once we start thinking properly about heat sinks and fans, but we'll see...

gkasprow, 2017-02-02:
I forwarded this to two groups at WUT who do such stuff. I'll let you know
once they reply.

klickverbot, 2017-02-02:
> Sure. Any preference as to interface standard/connector to use for this?

Ethernet? ;)

hartytp, 2017-02-02:
Can do, but then we have to put a microprocessor on, which seems a bit overkill IMO...

gkasprow, 2017-02-02:
One can buy simple Ethernet to I2C or serial bridge and connect temp sensors either with serial or I2C interface that would do all the job

hartytp, 2017-02-02:
@gkasprow Sounds good. Do you have a part number for that?

gkasprow, 2017-02-02:
For such purposes I use Lantronix embedded servers (Xport)
https://www.lantronix.com/products/xport/
there are lower cost Tibbo devices
And you just hook one or several TMP107 sensors.
Once you configure secure virtual serial port or connect to certain port
over Telet, you get sensors readout + a few IOs without writing single line
of code.
One can use it for I2C control with this chip

>
>

gkasprow, 2017-02-02:
The server cost is 44$

dtcallcock, 2017-02-02:
I think we'd be interested, especially if it doesn't sound like a jet aircraft taking off.  However quite a few AOs we have are 4W.  A quick look at the datasheet suggests the amp can do that if you're prepared to give it a bit more voltage and sacrifice a bit of efficiency.

Also, if you're adding an ADC anyway, how about monitoring the output power and VSWR?  I spend a lot of time switching power meters in and out of these setups when debugging.  

Finally, adding a shutdown somewhere would allow for over-current/temp/power/VSWR protection.  At that point though a microcontroller would probably be needed.

Please also add the brightest blue LED they make to the front panel.

jordens, 2017-02-02:
@dtcallcock Enlighten me: are you in short supply of blue light in the lab, do you want to use it as a blue flashlight, or why else do you want the brightest blue LED?
https://www.amazon.com/LOHAS%C2%AE-Color-Power-Energy-Saving/dp/B00JR2NOAK

dtcallcock, 2017-02-02:
We had a masters student who did this years ago.  It projected a bright blue disk on the lab wall.  Fortunately 50W versions weren't available back then!

Looking at the datasheet I notice the max no-damage VSWR specced is 10:1.  If active protection is not implemented do you have a plan for ensuring this isn't exceeded?  A 1dB pad on the output?

jbqubit, 2017-02-02:
Thank you for posting on this @hartytp. I'm interested in this as well. 

- Consider using Kasli as the controller. The vision for Kasli is a capable, inexpensive controller that can be economically applied even in simple applications. 

- In a lab environment the load is sometimes poorly behaved. For example, 100% reflection as if sweeping across the lineshape of a resonant cavity (EOM). Or, cable disconnection (open load). A design that accounts for this is helpful.

hartytp, 2017-02-02:
Okay, glad there's interest in getting this made.

> I forwarded this to two groups at WUT who do such stuff. I'll let you know
once they reply.

Thanks @gkasprow, I'll be interested to hear what they have to say -- particularly, how much it would cost to get WUT to do the design.

To set expectations correctly though: while I'd ideally like to get this made professionally, we haven't currently got funding set aside for it. The current plan is for one of the physics/engineering postdocs working on our project to do the design. This means we are limited in how many bells and whistles we can realistically hope to include. Depending on the cost of a professional design, among other factors, that might be able to change. If anyone else is interested in/able to provide more funding for this then let me know!

> For such purposes I use Lantronix embedded servers (Xport)

Nice! I hadn't seen those before. Seems like a better option than I2C or SPI.

@dtcallcock 

> if it doesn't sound like a jet aircraft taking off

"not audible above the lab AC" is one of the design goals for this...

> A quick look at the datasheet suggests the amp can do that if you're prepared to give it a bit more voltage and sacrifice a bit of efficiency.

Yes, the plan would be to run the IC of Vds=28V, Idq=100mA for maximum P1dB. PAE is still fine when doing this IMO.

> Also, if you're adding an ADC anyway, how about monitoring the output power and VSWR? 

That's a nice idea. I'll add that to the wish-list.

> Looking at the datasheet I notice the max no-damage VSWR specced is 10:1. If active protection is not implemented do you have a plan for ensuring this isn't exceeded? A 1dB pad on the output?

Yes, that's a bit of a problem with this kind of PA. My plan was to add pads for a T-attenuator (0dB by default), which can be used to stabilise it. 10:1 VSWR isn't bad though, and in most cases cable + device return loss will be enough to ensure stability. Active protection is a bit of a pain to implement effectively, and probably beyond the scope of our current ambitions.

I did also look at other amps/transistors which could be made unconditionally stable. However, I didn't find anything that could deliver the same performance without a lot more work.

> Finally, adding a shutdown somewhere would allow for over-current/temp/power/VSWR protection. At that point though a microcontroller would probably be needed.

I'll have a think about it. But, without some extra funding, that may be a bit unrealistic for the time being. The current plan is to keep this design as simple as possible and accept that it won't be a fully ruggedised solution. 

> Please also add the brightest blue LED they make to the front panel.

Most of the bright blue LEDs in our lab still have tape over them that you put there ;)

@jbqubit 

> Consider using Kasli as the controller. The vision for Kasli is a capable, inexpensive controller that can be economically applied even in simple applications

Our current plan is to keep this design more or less purely analog, so Kasli isn't needed -- if we only have a few IC temp/current sensors, then the ICs that Greg suggested are more appropriate than Kasli.

Also, Kasli takes up a fair amount of front/back panel space, which I'd prefer to reserve for other things -- no need to cram in the SMAs in order to fit in SFPs IMO.

hartytp, 2017-02-02:
> In a lab environment the load is sometimes poorly behaved. For example, 100% reflection as if sweeping across the lineshape of a resonant cavity (EOM). Or, cable disconnection (open load). A design that accounts for this is helpful.

> Looking at the datasheet I notice the max no-damage VSWR specced is 10:1. If active protection is not implemented do you have a plan for ensuring this isn't exceeded? A 1dB pad on the output?

One more thing: while stability/no-damage is only guaranteed for VSWR up to 10:1 for this IC, IME the region of the smith chart that can actually lead to damage is generally quite small for this kind of PA. So, damage due to an off-resonant EOM, etc. is pretty unlikely once cable losses etc are taken into account.

So, if we have the resources to produce a design that can ensure no damage for all loads, then great! let's go for it. But, if not, then I think that a design that works well under reasonable loads is still very useful...

sbourdeauducq, 2017-02-03:
@hartytp I think the mailing list would be a better place for this discussion.

gkasprow, 2017-02-03:
Hi

My college Tom Filipek (RF expert) can do this project.

He is involved in another project but in one month will be free to start.

I attached his email in CC.

Greg

hartytp, 2017-02-03:
@sbourdeauducq 

> @hartytp I think the mailing list would be a better place for this discussion.

I'd prefer to keep the discussion here for now unless you really feel strongly about it? IMO, the amp board is a relevant piece of support hardware for the Sayma/Kasli DDS boards being developed here. So, this is a natural place to have the discussion. Also, my impression is that the followers of this Git Hub project are a good representation of the sub-set of the Artiq list readers who are interested in hardware development.

sbourdeauducq, 2017-02-03:
Well, GitHub issues are more appropriate for actionable items than for discussions or asking questions around.

hartytp, 2017-02-03:
@sbourdeauducq I know this isn't really what the issue tracker is for. But, it does work very well, so can we go on abusing the system a while longer ;)

sbourdeauducq, 2017-02-03:
I'm pretty sure the mailing list would work just as well, e.g. all the participants in this issue are subscribers.

hartytp, 2017-02-03:
@sbourdeauducq Okay...

jbqubit, 2017-02-03:
I advocate for using the Issue system as we are in this Issue. Downsides of mailing list AFAICT
- distribution includes people who don't care about Sinara-Kasli related hardware
- @mention, #issue auto-linking isn't available
- no wiki for pulling together ideas
- inclusion of inline images is unpleasant
- single-page viewing of conversation thread is not supported
- Markdown isn't supported

hartytp, 2017-02-03:
@sbourdeauducq I can re-cast this in issue format if that helps: nebulous discussion thread => issue: specify this hardware, close when specification agreed.

hartytp, 2017-02-17:
Top post updated on the basis of these suggestions. Closing this issue, please open if you have anything else to add.

One question though (@jordens or anyone else who has a strong opinion): do we want to keep designs (specification) for this in the Sinara repository (Wiki). Or, do you want to move into a new project?

closed 2017-02-17 by hartytp

jordens, 2017-02-17:
@hartytp that's up to you. Feel free to take charge and to use the wiki. I like it because it is a git repository that I can clone and work on using my own editor and offline as well. If the design ends up using libraries/templates/files from Kasli/Sinara, doing so in this repository is also fine.

hartytp, 2017-02-18:
@jordens okay, done.

A bit off-topic, but: at the same time, I did a bit of reorganising on the Wiki and added some higher level documentation. My motivation for this was that I've talked to people in other groups about Sinara recently, and there seems to be quite a bit of confusion about what's being built and how it all fits together. In particular, people are confused about which parts offer deterministic latency, and which bits don't. e.g. Kasli TTL extension is deterministic, but the Kasli DDS extension does not have deterministic phase control. 

I know you had some nice stuff on this in the Artiq hardware white paper you wrote, but that's now getting pretty seriously out of date. As we're using the Wiki pretty heavily at the moment, it seems worth making sure it has a decent summary of the project. 

Anyway, I wrote that all pretty quickly, so it's a bit BS and probably contains a few mistakes. But, hopefully it's a useful starting point for proper documentation. Please feel free to alter/correct if you have time.

jordens, 2017-02-18:
@hartytp Excellent. Very much appreciated.
I have added a few lines about determinism. But this is -- like so many other things -- not an entirely resolved question.
Yes. `artiq-hardware` should be merged into this wiki IMHO.

hartytp, 2017-02-18:
Okay, great. Well, it still needs a fair bit of work, but at least it's a start...

jordens, 2017-02-19:
I do expect those who were most vocal in the different design decisions, contributed most to the status, and thereby shaped the scattered or unwritten specification to step forward here.
In several areas I/we don't have a good idea how it all fits together.

hartytp, 2017-02-20:
@jordens ACK.  I've done a bit more work on this. Hopefully, the Wiki is now sensibly structured and provides a rough high-level overview of how everything is planned to fit together...

jordens, 2017-02-20:
@hartytp Thanks.

jbqubit, 2017-02-21:
@hartytp Thank you for doing this. The new organization is a big improvement.

gkasprow, 2017-02-21:
@jbqubi, my professor encourages me to write habilitation on ARTIQ in a context of MTCA application. Originally I planed to focus on time synchronization ad RF distribution in MTCA but this seems to be better idea.
For US guys, habilitation it is something like second PhD, which is needed to became a professor :)
So the idea is to write a book that gives introduction to MTCA and several new features and how all of them can be used to make quantum physicists happy. Such book would be great manual for our system, at least for hardware part.

hartytp, 2017-04-07:
Funded by Oxford. Details on the wiki.

jbqubit, 2017-04-07:
Splendid! Is the back end based on Kasli? -Joe

Joe Britton
Sensors and Electron Devices
Army Research Lab
2800 Powder Mill Rd
Adelphi, MD 20783
301-394-3130
joseph.w.britton5.civ@mail.mil

hartytp, 2017-04-07:
Nope. It will be a microprocessor providing a simple SCPI-type interface over ethernet.

gkasprow, 2017-04-07:
Kasli would be an overkill for this purpose. Simple 5$ ARM microcontroller does the job nicely.

hartytp, 2017-04-07:
@jbqubit NB the power monitoring and other remote control features in this design are explicitly **not realtime**, so this will not be a DRTIO slave.