sinara-hw/sinara #152: SMP bullets

gkasprow, 2017-02-02:
I did short summary of available SMP bullets.
To fit between Sayma RTM and mezzanines, their length must be 6.4 or 6.45mm
Here is short list of suitable ones
Radiall R222 705 000
Rosenberger 19K101-K00L5
TE 1056700-1
Molex 73415-4295 / 73415-4293
Amphenol 1290-4008
Telegarthener J01392A0001 

Take into account that for every Sayma RTM we need 48 connectors.
We may not use LO distribution but it still gives 40 which multiplied by 9$ gives 360$ for Radiall ones.
I will try to find some source of lower cost SMPs at quantity of let's say 1000pcs.

gkasprow, 2017-02-08:
I got offers for SMPs. The price is much lower than in Digikey or Farnell.
We have to choose one of the offers below:
1. Radiall R222705000W,  4,3$/pc, LT 4 weeks [ spec ](http://www.farnell.com/datasheets/82178.pdf?_ga=1.220608670.785074793.1431025719)
2. JYEBAO  AD-P8P8-6.45, 3,95$/pc, LT 6-8 weeks,[ spec  ](http://www.jyebao.com.tw/files%20download2013/232.pdf)
3. Some replacement (spec below) specified for 26.5GHz, 3.6$/pc, LT 3 weeks
[image]

jbqubit, 2017-02-16:
@gkasprow How was this resolved? Let us know and close Issue.

gkasprow, 2017-02-16:
I just wanted to know if you have any opinion on these connectors.
If not, I will order one with shortest LT or lowest price.

gkasprow, 2017-02-16:
OK, closing

closed 2017-02-16 by gkasprow