sinara-hw/sinara #153: Kasli: Slow ADC features/specs

hartytp, 2017-02-03:
Proposal for Kasli 3U_ADC:

- **General**: base this on an IC like [ADAS3022](http://www.analog.com/en/products/analog-to-digital-converters/ad-converters/adas3022.html#product-overview). Not aiming for anything particularly ambitious in terms of noise/drift/speed/channel count. "Cheap and simple" is the goal here...
- **Use-case**: data logging, slow feedback loops, etc.
- **Channel-count**: 8 channels
- **Connectors**: insulated BNCs, front panel based on 3U_BNC module
- **Resoultion**: 16-bit
- **Sample rate**: >100kSPS for a single channel (1MSPS would be nice)
- **Input range**: +-10V max, ideally with several gain/range settings
- **Front-end**: basic common-mode + differential-mode filters to minimise RFI issues in a lab environment. Filter cut-off chosen to avoid compromising ADC bandwidth too much. Suitable protection diodes (eg so 10V can be connected to the input when the board is not powered)
- **Grounding**: 10k between input grounds and circuit ground. Either keep grounds for all inputs isolated from each other (preferred) or tie input grounds together, and add suitable common mode chokes.
- **Input termination**: 10k to input grounds. This is a low speed board, so 50R termination not required.

As always, feel free to post comments etc.

closed 2017-02-17 by hartytp

jordens, 2017-02-17:
We would be interested in having a Kasli extension for a general purpose, multi-sensor, multi-type temperature measurement IC, like [LTC2984](http://cds.linear.com/docs/en/datasheet/2984fb.pdf) (20 channels, 24 bit ADC, supports B/E/J/K/N/S/R/T Thermocouples, 2/3/4 wire RTDs, Thermistors, Diodes). We might also develop it.
The analog side is pretty different from that general purpose ADC, but if we can keep in mind the potential similarities on the digital side in both projects, that could be helpful. Just throwing that info out here.

gkasprow, 2017-02-17:
I see one potential problem with cold junction compensation since it should
happen on the place where thermocouple material touches connector.
So either the 3U panel should have standard thermocouple connector or we
should build dedicated box with temperature sensor in it that extends 3U
board connector to several thermocouple connectors.

jordens, 2017-02-17:
@gkasprow Absolutely. I am also pretty sure that we can't build a device that supports all types of temperature sensor on each channel. We'd probably restrict that and build it in a version for K-type TCs and another for RTDs.
For TCs, it might even make sense to put the IC into the dedicated external box you describe and use the SPI/RJ45 Kasli extension.

gkasprow, 2017-02-17:
@jordens The IC doesn't have to be outside of the box. There must be just dedicated cold junction sensor.
We can also use 3U LVDS board to connect SPI over LVDS providing that 4 lines are sufficient.
In this way little box with RTD sockets and IC could be connected at any distance using ordinary shielded twisted pair cable. In this way to single Kasli/VHDCI breakout we could connect up to 16x LTC2984, each serving i.e. 16 sensors and distribute them around the lab. This would satisfy probably all needs and let us use standard thermocouples with standard connectors.
Such sensor box could be supplied by 5V USB wall wart since we didn't implement power over LVDS.  But this is yet to be considered :)

hartytp, 2017-02-18:
@jordens ack. Ideally, I'd like to make sure we can use this design as a template for other extensions.

hartytp, 2017-02-20:
@jordens Have you thought about the practicalities of how one might use this extension module for slow laser noise-eater feedback loops? e.g. presumably this will work the same way as the digital photodiode noise-eater you mentioned before.

I'm assuming one would do the feedback on the core device using integer maths. How slow do you think this would be? (e.g. how long to do a few updates to, say, 10 channels)?

jordens, 2017-02-20:
@hartytp For us it is not enteriely decided whether we'd go for the digital photodiodes or help/invest into this multichannel ADC.
We'd probably shoot for loop bandwidths sub-Hz and any of the following approaches could work:

* do it with integer math on the core device (run a couple dozen PI cycles in integer math every few experiments)
* do it with moninj from the host, completely without kernels.
* do it with the experiment kernels broadcasting measurements as datasets, running the PI loop on the host, and then the kernels applying the updates again.

hartytp, 2017-02-20:
Thanks for the info.

> do it with integer math on the core device (run a couple dozen PI cycles in integer math every few experiments)

How long would this take per PI cycle? I assume it'd be limited by the latency on DRTIO SPI read/writes, right?

> For us it is not enteriely decided whether we'd go for the digital photodiodes or help/invest into this multichannel ADC.

That's good to know. When do you think you'll make a decision about this (is the funding confirmed yet?)? Any extra funding for this would be a real help for us, but I'd like to get the work started...

jordens, 2017-02-21:
Yes. I'd expect it to be similar or better than the SPI xfer time/bandwidth.
Funding is not strictly confirmed. And officially the project would start in April/May. But we'll definitely try to get solidify the specification before that.

hartytp, 2017-02-21:
@jordens So, to be clear: you expect a single PI cycle for feedback on the core device (sample on a 1MSPS ADC and reprogram a DDS etc over DRTIO) to take ~1us. Correct?

hartytp, 2017-02-21:
> Funding is not strictly confirmed. And officially the project would start in April/May. But we'll definitely try to get solidify the specification before that.

Okay, good to know. We hope to get things moving a bit faster than that (next couple of weeks). But, if you want to be involved in funding/helping to define the specifications (which would be welcomed) then be in touch asap.

jordens, 2017-02-23:
I'd expect each (ADC, PI, DAC) to take on the order of 1 us.