sinara-hw/sinara #155: Status of the boards

gkasprow, 2017-02-06:
Update of the boards status:

|No | Board name | Board status/activity |  Execution date  | Next activity |
| ---| ------------- | ------------ | -------------- | -------------- |
| 1. | Sayma AMC | PI&SI verification |   Feb 2017 | proto production |
| 2. | Sayma RTM | layout 98% done, PI verification |  Feb 2017 | proto production |
| 3. | Metlino | Schematics done, placement done, routing started |  March 2017 | review |
| 4. | Kasli | founding in preparation |   | review |
| 5. | PCB_3U_BNC | proto production | Feb 2017 | delivery |
| 6. | PCB_3U_DAC | founding in preparation |  | review |
| 7. | PCB_3U_RJ45 | proto production | Feb 2017 | delivery |
| 8. | PCB_3U_SMA | started, schematics 90% done| ?? | review |
| 9. | PCB_3U_VHDCI_breakout | proto production | Feb 2017 | delivery |
| 10. | PCB_3U_ADC |  founding in preparation |  | review |
| 11. | PCB_3U_DDS |   founding in preparation |  | review |
| 12. | PCB_I2C_switch | founding in preparation |  | review |
| 13. | PCB_AMC_SATA | proto production | Feb 2017 | delivery |
| 14. | PCB_eRTM15_Baikal | schematics 80% done | ?? | review |
| 15. | PCB_MCH_mordovia | proto production | Feb 2017 | delivery |
| 16. | PCB_mezzanine_analog_allaki | layout in progress | ?? |
| 17. | PCB_mezzanine_analog_template | proto production | Feb 2017 | delivery |
| 18. | PCB_mezzanine_clock | not started | ?? | review |
| 19. | PCB_mezzanine_clock_template | proto production | Feb 2017 | delivery |
| 20. | PCB_RTM_loopback | proto production | Feb 2017 | delivery |

sbourdeauducq, 2017-02-07:
Moved to wiki: https://github.com/m-labs/sinara/wiki/Status

closed 2017-02-07 by sbourdeauducq

jmizrahi, 2017-02-08:
@gkasprow Could you write a list of the description / role of each PCB somewhere (or add it to that table)? Now that we are talking about 20 separate PCBs, I am having trouble keeping track of what all of them are.

gkasprow, 2017-02-08:
@jmizrahi done

jmizrahi, 2017-02-08:
@gkasprow Thanks!