sinara-hw/sinara #157: SATA pinout unification

gkasprow, 2017-02-08:
We need to unify SATA pinouts
SATA pinouts is defined from computer side (at least on connector symbol)
2 A+, Tx
3 A-, Tx
5 B-, Rx
6 +, Rx

We have SATA connectors on:

1. Sayma AMC - fat pipe RX/TX4, RX/TX5, Port0 from internal PHY
Now they are connected in follwoing way
J11, FP RX/TX4, Master SATA, A=Tx, B=Rx, TX is AC-coupled
J12, FP RX/TX5, Slave SATA, A=Rx, B=Tx, Tx is AC coupled
J10, Gbit PHY, Port0, A=Tx0, B=Rx0, TX and RX are AC coupled

2. Metlino:
J24, Gbit PHY, A=Tx0, B=Rx0, TX and RX are AC coupled

3. Sayma RTM
J70, Master SATA, A=Tx, B=Rx, Rx and Tx are AC coupled
J71, Slave SATA, A=Rx, B=Tx, Tx are AC coupled

4. Kasli
Computer pinout means A=Tx, B=Rx?

5. I will provide SFP to SATA adapters. 
Will 50cm be enough?
How should I wire them?

jordens, 2017-02-08:
All ACK from my side.
I'd wire the adapter to be plugged into a "Master SATA" connector.

jbqubit, 2017-02-08:
Looks good to me.

closed 2017-02-16 by jbqubit