sinara-hw/sinara #158: Sayma thermal simulation

gkasprow, 2017-02-09:
We did thermal simulations of Sayma AMC.
It seems that the most critical parta are SFPs.
I specified 1W per transceiver. What SFP models will be used?

[image]

jordens, 2017-02-09:
>= 6 Gb/s

Surprising that two Watts heats the board up that much, given that a many-dozen Watt FPGA doesn't.

michallgaska, 2017-02-09:
In this simulation FPGA has a heat sink.

michallgaska, 2017-02-09:
I did thermal simulations with [this](http://www.10gmodule.com/10g-850nm-300m-sfp-optical-transceiver-module.html) 10G modul. 

[image]

jbqubit, 2017-02-09:
@michallgaska What air temperature and air flow rate does your model assume? 

I'm using generic SFP that is compatible with ZyZEL SFP-BX-1310-10-D and SFP-BX-1490-10-D. From XyZEL [specification sheet](ftp://ftp2.zyxel.com/SFP-BX1310-10/datasheet/SFP-BX1310-10_3.pdf) supply voltage is 3.1-3.5V at up to 300 mA. Operating temperature up to 70C. MTBF is specified as 137,682 hours. 

I took one of mine apart -- see photo. The die cast body is a decent thermal conductor but there is no solid thermal link (eg thermal paste) to detector/emitter/PCB. Nor is there a solid thermal link between the die cast body and the flimsy external housing. Our application isn't much different from usual industrial application so I'm not too worried. It might just be that they run hot. 

I've never taken one of these apart to know if other manufacturers do a better job of thermal coupling. @sbourdeauducq do your SFP get hot when doing continuous tests on KC705? 

[image]

sbourdeauducq, 2017-02-10:
They do get hot, but stay under ~50C. The 10G ones seem to heat more idling than the 1G ones continuously transmitting. I still haven't got the 10G SFPs to work correctly, so I don't know what happens when they continuously transmit.

jbqubit, 2017-02-13:
@sbourdeauducq Please consider using Xilinx IP to test a 10G SFP at 10 Gb/s. It would be good to know if there's a need for additional heat-sinking on the PCB or within the SFP modules themselves (eg adding thermal paste). 

https://www.xilinx.com/support/documentation/application_notes/xapp1200-k7-xcvr-wiz-example-design.pdf

sbourdeauducq, 2017-02-13:
> Please consider using Xilinx IP to test a 10G SFP at 10 Gb/s.

Then what should I measure and what is this going to tell you that the SFP datasheet does not? And adding thermal paste inside SFPs is not an option.

gkasprow, 2017-02-13:
I have a couple of 10Gbit SFPs, we will test in with full speed once HW is
ready.

michallgaska, 2017-02-14:
@jbqubit In MicroTCA standard we have 20m3/h per AMC card. I can only specify air velocity so i considered airflow through the cross section area of a duct and i assumed 100 cm/s.

jbqubit, 2017-02-15:
@michallgaska Assuming the full cross sectional area of the duct sounds very optimistic. Especially with the mezzanine boards installed in PCB_Sayma_RTM. The 3D models are no longer in the repository so I can't check. Consider repeating modeling with reduced flow informed by available cross sectional area.

michallgaska, 2017-02-17:
@jbqubit Now I am working on Sayma RTM thermal simulation and i assume 10m3/hr per card. I have to take into account mezzanine cards somehow. Maybe we will have to add small heatsink to some ICs but it is not a problem.

jbqubit, 2017-02-21:
I just double checked flow for the uTCA chassis I'm using. Native R5. Flow is >> assumed by your model. Good. 
```
# AMC/RTM card cross section is about 1" x 6" = 3.8e-3 m**2
# NAT NATIVE R5 AMC fans 4 x 110 m**3/h, 9 slots gives 49 m**3/h/slot
# NAT NATIVE R5 RTM fans 2 x 110 m**3/h, 7 slots gives 31 m**3/h/slot
31/3.8e-3/3600 # is 226 cm/s
```
@michallgaska Since you created #162 for PCB_Sayma_RTM I changed the title of this to PCB_Sayma_AMC. 

Thermal analysis of PCB_Sayma_AMC looks good. 10 Gbps test noted on wiki [test page](https://github.com/m-labs/sinara/wiki/HardwareTestCharacterization).

closed 2017-02-21 by jbqubit