sinara-hw/sinara #159: I2C tree

gkasprow, 2017-02-13:
I added on Sayma/Metlino top entity simplified I2C connectivity with chip numbers and addresses. 
Is such drawing clear enough?
Sayma
[image]

Metlino
[image]

jbqubit, 2017-02-13:
Looks good... Some comments. 
- Metlino MUX Port 6/7 right-hand-side blue box should be "VHDCI" not "RTM ports"
- on PCB_Sayma_RTM each analog mezzanine has one I2C
    - allaki I2C chips #144

gkasprow, 2017-02-14:
@jbqubit what do you mean that "on PCB_Sayma_RTM each analog mezzanine has one I2C " ?
Do you mean one I2C interface

jordens, 2017-02-14:
@gkasprow IMHO the most useful representation would be textual, machine and human readable. In the end, it will need to be machine readable anyway.

Something like this (for Sayma):

```json
{
  "AMC_MUX": {"path": [], "addr": 0x71},
  "FMC_EEPROM": {"path": [0], "addr": 0x50},
  "SFP1_EEPROM": {"path": [2], "addr": 0x50},
  "SFP1_DDMII": {"path": [2], "addr": 0x51},
...
  "AMC_SI5324": {"path": [4], "addr": 0x68},
...
  "AMC_MAX6642": {"path": [6], "addr": 0x24},
...
  "RTM_MAX6642": {"path": [7], "addr": 0x24},
  "RTM_MUX": {"path": [7], "addr": 0x76},
  "AFE0_TCA9539_0": {"path": [7, 0], "addr": 0x74},
  "AFE0_TCA9539_1": {"path": [7, 0], "addr": 0x75},
  "AFE0_LM75": {"path": [7, 0], "addr": 0x48},
  "AFE0_24AA02E48": {"path": [7, 0], "addr": 0x50},
  "AFE1_TCA9539_0": {"path": [7, 1], "addr": 0x74},
...
  "RTM_SI5324": {"path": [7, 5], "addr": 0x68},
...
}
```

jbqubit, 2017-02-14:
> what do you mean that "on PCB_Sayma_RTM each analog mezzanine has one I2C "

In schematic for Mezzanine_connectors1 I2C_SDA is connected to MEZ_SDA4. I don't see the analog Mezzanine connectors' I2C in your map.

gkasprow, 2017-02-16:
@jbqubit you are right, I modified the schematic to keep the convention. 
The second table describes connections of mezzanines. I changed AFE to Analog mezzanine to keep naming consistent.

closed 2017-02-16 by gkasprow