sinara-hw/sinara #163: spaces in wiki page names

jbqubit, 2017-02-24:
@hartytp Please don't include spaces in names of wiki pages. The resulting URL's are difficult to cross reference. For example.

https://github.com/m-labs/sinara/wiki/Zotino%2032-channel%20DAC

Better is a single word or CamelCase. For example,

https://github.com/m-labs/sinara/wiki/Zotino

hartytp, 2017-02-25:
Okay, I'll do something about this next week when I have time.

hartytp, 2017-02-25:
Before we (I) spend too much time tidying up the Wiki, we should decide on our plans for system-level documentation of Sinara going forwards  -- there's no point doing this kind of cosmetic work on the Wiki if we're going to change platforms soon anyway, which will require larger reformatting.

To reiterate the discussion in #133: I don't care what platform/language we use, so long as we end up with a single site for web-docs for Sinara. This  should cover the entire ecosystem, from both high-level and board-level perspectives. It should be kept up to date and provide thorough cross-linking between the various projects making up Sinara.

Currently, I've been using the Wiki, WUT are planning to write in LaTex and @jordens is writing up Kasli in reStructured text. So, what's the plan for merging this all together into a single site? e.g. I'm assuming that the information @jordens is writing about Kasli (or, at least a large sub-set of it) should end up in the same place as the information currently on the Wiki. What's the plan for making that happen?

I'm closing this issue as I don't plan to resolve it until the larger question about documentation has been settled, and if we do decide to move the Wiki docs to a different platform/language then this issue becomes irrelevant anyway.

closed 2017-02-25 by hartytp

hartytp, 2017-03-04:
Done. I've tried to fix all internal links, but if you spot any I've missed please do fix them.