sinara-hw/sinara #165: allaki analog mezzanine: schematic review for analog input

jbqubit, 2017-03-02:
The following comments pertain to ADC_channel.SchDoc on PCB_mezzanine_analog_allaki.  Recall that discussion of specification for allaki analog input is #81. Note that existing Issue #79 is design review for analog output. 

Bugs and comments follow. 
- Help out customers who will use this board. 
    - [ ] State input impedance for SIG_IN and RET_IN for both input configurations. 
    - [ ] State input voltage range and bandwidth for each input range.
    - [ ] State bit configuration (A0, A1) for each range. 
    - [ ] Label to follow something like what @hartytp called for eg "G=1 (+-10V input range) f3dB=10MHz, G=10 (+-1V input range) f3dB=4MHz, G=100 (+-100mV input range) f3dB= 550kHz" 
- [ ] Missing 10 pF to ground on RET_IN... in #81 @hartytp recommended "Input SMA: ground capacitively coupled to circuit ground with ~10pF to avoid low-frequency ground loops" 
- [ ] Group together RFI filter components and draw box around them; call out differential and common mode bandwidth in schematic 
- Did WUT independently check the various off-the-cuff estimates made by @hartytp in #81? For example, "expected offset drifts assuming +-1K ambient temperature stability?" 
- [ ] "input Vpp" should be +/- 5 V, right? 
- [ ] Fix "power matching required." This is supposed to be 50 Ohm source, right? 
- [ ] VOCM is output common-mode voltage for ADA4927
    - on allaki: VOCM -> ADC_VCM -> pin 33 on B2B connector 
    - on Sayma_RTM: pin 33 of B2B connector goes to VCCN_6V0_A 
    - ADA4927 manual says "The VOCM divider comprising two 10 kΩ resistors at a voltage approximately equal to the midsupply point, [(+VS internal divider, the VOCM ) + (−VS)]/2. Because of this pin sources and sinks current, depending
on the externally applied voltage and associated source resistance." 
    - ADC_VCM needs a well-defined source impedance, right?
    - Without filtering there will be noise on VCCN_6V0_A which is also used as input for the 5V regulator on allaki
    - I don't see how this matches common-mode voltage and range of AD9656 as configured on PCB_Sayma_RTM. "2.5 Vpp" is insufficient.... define common-mode voltage, Vpp and output imedance.

dhslichter, 2017-03-03:
Here are some comments/questions, by no means an exhaustive list:

- it appears that R24 should be 50 || 56 ~ 26 ohms to present matched differential input impedances to the ADA4927, on the ADC schematic.
- Is L4 really necessary?
- what is meant by the grayed-out boxes on the FP_connectors page?
- Layer Mid4 appears empty, should be ground plane?
- For high speed signals it would be better to place R24 and R16 on the bottom side rather than via-ed to the top, seems like there should be room for this.  Not desperately urgent.
- R41, C4, C63 should ideally be on the bottom side of the board (there appears to be room for them) to reduce the chance for crosstalk from the DAC output channels
- values and connections for ACG capacitors on HMC542 are different from sample schematic on datasheet.  Needs discussion?
- trace routing between IC7A and IC4A (switch pickoff to attenuator before power detector) crosses under board and is close enough to differential traces of "B" ADC input to give rise to differential crosstalk.  Two possible mitigations:
    - move IC4A to before via (will reduce signal strength, and thus crosstalk, by 12 dB)
    - reroute differential traces after R40B/R42B down and away so they are farther away from the signal going to IC4A, less crosstalk to differential mode.  
- check heights and vertical clearances on L1A, L1B w/r/t entire RTM assembly/available dimensions -- these components are tall and may not be able to be fit on the board underside, especially if shielding is to be added on top.
- maintain 50 ohm trace gaps after RF switch (C49, R2, C50)
- maintain 50 ohm trace gaps in midsection of discrete-component filter zone (R48, R49, R50, R51)
- ATT_CTRL selected pins from J13 must comply with gateware specification for SPI and TTL configurable blocks from Sayma RTM (see #140).  Obviously this can be reconfigured in gateware as desired to match what is here, but we should just make sure there are no difficulties/surprises arising from these particular pin choices.  In particular, it would be nice if we used adjacent pins (e.g. ATT2_CTRL is all over the map) for shared functions (e.g. CLK/S_IN/LE, which are SCLK/MOSI/CS for SPI).
- what is the purpose of R18/R19?  Since we have IC4 and IC7 following it seems we should leave those out.

sbouhabib, 2017-03-03:
I copied all the comments to comment them myself

- Help out customers who will use this board. 
    - [ ] State input impedance for SIG_IN and RET_IN for both input configurations. 
    - [ ] State input voltage range and bandwidth for each input range.
    - [ ] State bit configuration (A0, A1) for each range. 
    - [ ] Label to follow something like what @hartytp called for eg "G=1 (+-10V input range) f3dB=10MHz, G=10 (+-1V input range) f3dB=4MHz, G=100 (+-100mV input range) f3dB= 550kHz" 
- [ x] Missing 10 pF to ground on RET_IN... in #81 @hartytp recommended "Input SMA: ground capacitively coupled to circuit ground with ~10pF to avoid low-frequency ground loops" 
They are there, directly on the SMA connector on the FP_Connectors schematic
- [x ] Group together RFI filter components and draw box around them; call out differential and common mode bandwidth in schematic; related is RFI filter simulation #104
done
- Did WUT independently check the various off-the-cuff estimates made by @hartytp in #81? For example, "expected offset drifts assuming +-1K ambient temperature stability?" 
at some point it was said that we/you are happy with the filter as it is, i checked the freq. cutoff and posted them in #104 
- [ ] "input Vpp" should be +/- 5 V, right? 
- [ ] Fix "power matching required." There's an Issue on this. #111
- [x] VOCM is output common-mode voltage for ADA4927
    - on allaki: VOCM -> ADC_VCM -> pin 33 on B2B connector 
    - on Sayma_RTM: pin 33 of B2B connector goes to VCCN_6V0_A
the connector is mirrored to the sayma version, the one i use is the one as in the template 
    - ADA4927 manual says "The VOCM divider comprising two 10 kΩ resistors at a voltage approximately equal to the midsupply point, [(+VS internal divider, the VOCM ) + (−VS)]/2. Because of this pin sources and sinks current, depending
on the externally applied voltage and associated source resistance." 
    - ADC_VCM needs a well-defined source impedance, right?
    - Without filtering there will be noise on VCCN_6V0_A which is also used as input for the 5V regulator on allaki
    - I don't see how this matches common-mode voltage and range of AD9656 as configured on PCB_Sayma_RTM. "2.5 Vpp" is insufficient.... define common-mode voltage, Vpp and output imedance. 

- [x] it appears that R24 should be 50 || 56 ~ 26 ohms to present matched differential input impedances to the ADA4927, on the ADC schematic.
not really, for a 50 Ohm source you have R16||443OHm which gives 49.7 OHM, in the case not using the in AD8235
- [x] Is L4 really necessary?
not really - removed
- [x] what is meant by the grayed-out boxes on the FP_connectors page?
these were for me, if i wanted to add caps near the connectors, i removed the unneeded ones
- [x] Layer Mid4 appears empty, should be ground plane?
it is not empty, mid 1 and 4 are both ground planes 
- [x] For high speed signals it would be better to place R24 and R16 on the bottom side rather than via-ed to the top, seems like there should be room for this.  Not desperately urgent.
- [x] R41, C4, C63 should ideally be on the bottom side of the board (there appears to be room for them) to reduce the chance for crosstalk from the DAC output channels
- [x] values and connections for ACG capacitors on HMC542 are different from sample schematic on datasheet.  Needs discussion?
they should be actually 10nF as in the same as the lowest frequency in the transmittion line, we changed to 10nf the AC caps, I missed those.
- [x] trace routing between IC7A and IC4A (switch pickoff to attenuator before power detector) crosses under board and is close enough to differential traces of "B" ADC input to give rise to differential crosstalk.  Two possible mitigations:
    - [ ] move IC4A to before via (will reduce signal strength, and thus crosstalk, by 12 dB)
    - [x] reroute differential traces after R40B/R42B down and away so they are farther away from the signal going to IC4A, less crosstalk to differential mode.  
- [x] check heights and vertical clearances on L1A, L1B w/r/t entire RTM assembly/available dimensions -- these components are tall and may not be able to be fit on the board underside, especially if shielding is to be added on top.
moved to other side
- [x] maintain 50 ohm trace gaps after RF switch (C49, R2, C50)
they were
- [x] maintain 50 ohm trace gaps in midsection of discrete-component filter zone (R48, R49, R50, R51)
they are
- [x] ATT_CTRL selected pins from J13 must comply with gateware specification for SPI and TTL configurable blocks from Sayma RTM (see #140).  Obviously this can be reconfigured in gateware as desired to match what is here, but we should just make sure there are no difficulties/surprises arising from these particular pin choices.  In particular, it would be nice if we used adjacent pins (e.g. ATT2_CTRL is all over the map) for shared functions (e.g. CLK/S_IN/LE, which are SCLK/MOSI/CS for SPI).
the issue here is routing
from the FPGA point of view it can be changed and these are not critical signals to have them affect timing issues, you just have to define this in the constraints file (which will have to be done anyway), while here we have only one layer for digital signals (so no crossing, some changes can be made though unnecessary)and routing-wise it will be difficult and unnecessary to group them together, and swaping them on the connector (FPGA) is common practice,
- [x] what is the purpose of R18/R19?  Since we have IC4 and IC7 following it seems we should leave those out.
we said that some people will not want the power detector mounted, so for them it is a 50OHm match, this can be done using IC4 and a resistor in place of  C56 and C55 pads, but there is no point in having IC4 put in if the power detector is unneded

gkasprow, 2017-03-03:
One more issue - use via in pad only when it is really necessary. Unmask it on another side to remove pockets.

dhslichter, 2017-03-03:
Another point with via in pad -- want to have epoxy-filled vias for reliability of soldering/manufacturability.  Boards like this, you don't want to be dealing with back solder wicking issues.  I know this makes it more expensive but at scale the cost difference is not giant.  We should discuss.

sbouhabib, 2017-03-03:
I was also for having vias filled up

gkasprow, 2017-03-03:
Ok, don’t forget to note it on gerber files

hartytp, 2017-04-16:
@sbouhabib  What's the progress on Allaki? Please let us know when all the changes in these issues have been implemented and we can perform a final review on the finished design.

hartytp, 2017-05-15:
@sbouhabib (or @gkasprow) ping. Is Allaki ready for final review?

jbqubit, 2017-05-19:
@sbouhabib As we discussed on 4/24 please review all the allaki-related Issues. If they're resolved close.

@hartytp I had an email interaction with @sbouhabib and @gkasprow on 4/25. Here's their responses. 

> Allaki was finished a very very  long time ago. It's a little bit different matter of
A. Waiting for the tests of the afe mezzanines
B. Waiting for formalities

> The PCB is finished since some time.
But we don't produce it on purpose.
The board is simple (but expensve) and can be printed and assembled within 3-4 weeks.
I want to test analog mezzanine template first with Sayma, then after we make sure it works we will produce set of Allaki.

My follow up was the following. 
> Please comment on review and make brief comment on outstanding Allaki Issues and check checkboxes. This will trigger interested parties to take a look. If everybody is happy we can close. :) 

#181 
#165
#111
#110
#79

hartytp, 2017-08-02:
@sbouhabib Thanks for doing that.

- Should R24 be ~25Ohms assuming the source impedance is 50Ohms?
- it would be nice to arrange things so that one can switch between the InAmp and direct ADC paths by *only* changing the jumpers, without having to change ADA4927-1 Rf/Rg resistors as well. What if we moved R56 to the other side or R8? Then, changing to the InAmp would be achieved by using a 25Ohm jumper rather than a 0Ohm jumper. (AFAICT, we can keep the other resistors at 332 Ohms while using the InAmp, right? Or, are you worried about power dissipation?)
- @jbqubit What is our default population option? Can we make the InAmp the default option, on the grounds that laser intensity stabilisation is likely to be the most common use for this ADC in the near future? Users who need the full bandwidth can always switch by changing the jumpers...

jbqubit, 2017-08-03:
> Can we make the InAmp the default option, on the grounds that laser intensity stabilisation is likely to be the most common use for this ADC in the near future?

Yes. Let's make that the default configuration.

sbouhabib, 2017-08-15:
The in Amp and the Values of the components for the Inamp are set as the default config.
@hartytp: the values for the buffer resistors were taken mainly from the analog tool for diff amps. I would prefer to keep those now and try to check if the same values can be used for both cases, if any components can be unified for both routed we will test and than decide.

hartytp, 2017-08-16:
@sbouhabib Sounds like a good plan to me.

PS Thanks for finishing up Allaki. I'm happy with the designs as they are...

gkasprow, 2017-08-16:
@sbouhabib <https://github.com/sbouhabib> please send gerbers  and  BOM to
Technosystem  for  verification

hartytp, 2017-09-03:
AFAICT, Allaki has been sent for manufacture, so closing this issue.

closed 2017-09-03 by hartytp